Koenkk/zigbee2mqtt #4442: Lost entity sensor.0x00158d0002f8xxx_click after upgrade Zigbee2mqtt

xMariusz, 2020-09-23:
## What happened

After upgrade Zigbee2mqtt 1.14.4.1 I lost entity sensor.0x00158d0002f8xxx_click, however I see entity like sensor.0x.0x00158d0002f8xxx_battery and  sensor.0x.0x00158d0002f8xxx_linkquality. Automations stopped working

## What did you expect to happen
When I press button, automation should turn the light

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531

OzGav, 2020-09-24:
Have you tried re-pairing the button?

xMariusz, 2020-09-24:
Yes. Issue concerns Xiaomi Aqara products buttons/sensors

OzGav, 2020-09-24:
Not sure. I have a number of these and they are working fine on 1.14.4.2 

I have seen in the past entities getting renamed. Have you had a good look to see if that has happened? Might be under the friendly name?

Koenkk, 2020-09-25:
Can you share your `configuration.yaml`?

xMariusz, 2020-09-25:
I solved my problem. My steps:

1. Remove zigbee device from HA (Configuration -> Devices)
[image]
2. Unplug coordinator
3. Plug coordinator
4. Paring device
5. Solved

closed 2020-09-25 by xMariusz