Koenkk/zigbee2mqtt #4443: CC26X2R1 -- devices stop responding

dlashua, 2020-09-23:
## What happened
After some time (5-10 hours) devices stop reporting state and devices become uncontrollable. MQTT still shows zigbee2mqtt connected. The log shows no errors. Restarting zigbee2mqtt fixes the issue. This has happened for about a week now, however, it wasn't always the case.

I have not added any new devices to my network. But I have removed some, blocked them from rejoining, and moved them to another zigbee2mqtt network (on a CC2531) to resolve an issue with some xiaomi sensors. Additionally, I've pulled the "dev" docker image at least once a day.

I have 77 devices on this network. 

## What did you expect to happen
Devices to continue being controllable and reporting state.

## How to reproduce it (minimal and precise)
Just wait... 

## Debug info
Zigbee2MQTT version: 1.14.4-dev (docker)
Adapter hardware: CC26X2R1
Adapter firmware version: 20200805

dlashua, 2020-09-23:
I just noticed it happened again. It's been 1 hour and 10 minutes since my last restart.

With such a short time span, the logs are bit easier to pour through. Aside from MQTT Publish messages and the startup messages, the only thing in the logs are lines like `Failed to execute routing table for 'master_closet_overhead_s'` which happen during a network scan.

dlashua, 2020-09-24:
This is also happening on my other Zigbee2MQTT network. This one runs the same version (1.14.4-dev) but on a Raspberry Pi 4 with a CC2531 running the latest non-source-routing firmware.

I think connectivity to my MQTT server is sometimes unavailable for a few seconds (which, obviously, needs to be fixed). But I believe zigbee2mqtt is designed to recover from this. Perhaps changes to the connection code are working correctly?

Koenkk, 2020-09-25:
Zigbee2MQTT already reconnects automatically.

Can you provide the herdsman debug logging from starting Zigbee2MQTT till the crash?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

dlashua, 2020-09-25:
Yes. I'll make these changes and send the log the next time it happens. Thanks!

dlashua, 2020-09-25:
I haven't had a chance to add herdsman logging yet, but, I have some more information.

My MQTT server runs on one machine. Zigbee2MQTT runs on another machine. These are connected by a switch.

I needed to upgrade the switch today, so I just did it, leaving all the devices plugged in. During this time, zigbee2mqtt would be unable to communicate with the MQTT server. However, once the upgrade was complete, and the switch working again, I checked the zigbee2mqtt logs. It hadn't complained at all. It still thought it was connected to MQTT, and was still publishing messages (or, trying to... the messages were not showing up in MQTT) with no errors at all in the log.

All of the devices showed as "unavailable" in Home Assistant, because the LWT kicked in to mark zigbee2mqtt as offline. As soon as I restarted zigbee2mqtt, the devices all came back online in Home Assistant and respond to commands as expected.

Koenkk, 2020-09-25:
Please provide the debug log of: stopping your MQTT broker and triggering a message from a device (e.g. press a button).

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

dlashua, 2020-09-25:
When I stop the broker, it works as expected. I see a disconnect message in the logs. Then I start the broker up again, and I see it reconnect.

However, if I just yank the network cable out of the router for the server my MQTT server runs on, the logs show nothing interesting at all.

Here's some logs:
https://controlc.com/eb7770fa

The Zigbee button I was pressing is called test_lk_button. You can see that I press it MANY times. In the log, I press it while the network cable is still attached and it works as expected. Then I pull the cable and press it some more. The logs look the same, but the button presses are not making it to MQTT (of course, because the server is disconnected). Home Assistant doesn't see the button presses either. Then I plug the cable back in. And it still doesn't see the button presses.

Koenkk, 2020-09-26:
Please provide the log of this with the following environment variable being set: `DEBUG=mqtt*` (logging will only be sent to stdout). In the log clearly annotate where you unplug/plug the network cable.

dlashua, 2020-09-29:
I'm working on getting you this data. I should be able to run this test this weekend.

Though, I'm starting to think maybe there's nothing zigbee2mqtt can do. If the MQTT server drops off the network without closing the connection, the underlying OS will likely continue trying for some time before informing the system that the connection has been closed. From the kernel's point of view, I imagine it doesn't look different from a host with high latency.

closed 2020-11-03 by dlashua