Koenkk/zigbee2mqtt #4445: CC2531 Router doesn't connect

leranp, 2020-09-23:
## What happened
I flashed router firmware on new CC2531 with antenna, but the device doesn't connect to my coordinator , it only blink the red led on/off every second, tried to flash couple of times , when flashing with USB debugging i can see in the serial terminal that the device send every couple of seconds the line: " PANID:1A63,Ch:11,Join:1 "
The coordinator is allowing join (i tried with other devices) , i tried to press on the S2 button for 5 seconds like in the guide, no luck
Coordinator is up to date , and the static PANID is off
Any suggestions?
## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Bartekn86, 2020-09-27:
CC2531 works properly, be sure that you have right firmware and correctly set usb/ama0 port ( ls dev )

VascoW, 2020-10-06:
I struggled similarly this morning. CC2531 seem to behave normal but nothing get him into pairing with my bridge.

Solution was rather simple: my bridge was full (at 22 devices). I had to remove one sensor to make room for the router. 

To remove one: don't fiddle too long with zigbee2mqtt. The sensor has to "sign out" 
1) set the permit to join to false
2) trigger inclusion on your sensor you want to remove
3) Sensor signed himself out of zigbee2mqtt and opens for new inclusions. Nothing happens and he is dormant. 
4) remove the battery/power to keep it "dead"
5) set permit to join to true
6) include router
7) include your sensor again, should sign up with the router (that doesn't work for me, somehow my device limit stayed identical, even yet router was added)

I trust it's worth a try if you have quite a few sensors already added (mine was 22, online states everything between 15 to 48 devices pending firmware).

What did not work:
- removing device via zigbee2mqtt node-red admin dashboard
- removing the config file in the configurations.yaml file
- reading top 10 google hits twice
- scratching my head

leranp, 2020-10-06:
> I struggled similarly this morning. CC2531 seem to behave normal but nothing get him into pairing with my bridge.
> 
> Solution was rather simple: my bridge was full (at 22 devices). I had to remove one sensor to make room for the router. 
> 
> To remove one: don't fiddle too long with zigbee2mqtt. The sensor has to "sign out" 
> 1) set the permit to join to false
> 2) trigger inclusion on your sensor you want to remove
> 3) Sensor signed himself out of zigbee2mqtt and opens for new inclusions. Nothing happens and he is dormant. 
> 4) remove the battery/power to keep it "dead"
> 5) set permit to join to true
> 6) include router
> 7) include your sensor again, should sign up with the router (that doesn't work for me, somehow my device limit stayed identical, even yet router was added)
> 
> I trust it's worth a try if you have quite a few sensors already added (mine was 22, online states everything between 15 to 48 devices pending firmware).
> 
> What did not work:
> - removing device via zigbee2mqtt node-red admin dashboard
> - removing the config file in the configurations.yaml file
> - pressing any of the 2 buttons  on CC2531 (the important one is the one closer to the USB, anyway no effect for me)
> - reading top 10 google hits twice
> - scratching my head
> 

Thank you for your answer,
I was only 6 devices connected, i will try to remove them all and connect only the router.

This is my serial log from the CC2531 router:
PANID:1A62,Ch:11,Join:1  ---------> join chaned to 1 only when the coordinator is join permit , when the join is not permitted the log is the same but now Join:0
I tried to look for something in the log of the coordinator but in debugging mode i can't see anything related to the router.

I hope that removing the devices and connect the router first will solved the problem

closed 2020-10-06 by leranp

bigretromike, 2020-12-21:
I have same issue. only 5 device connected, did removing all devices helped?

bigretromike, 2020-12-21:
@VascoW removing every device, restarting, allowing to join and still no luck. any suggestion with your experience ? 
@leranp any luck with this? what helped you with issue

OzGav, 2023-01-03:
I know this is closed but it took me forever to find the solution in my case so I want to share it. In my case my old CC2531 that I wanted to repurpose as a router had the same IEEE address as my new coordinator. I found this post which shows how to change the IEEE address https://github.com/Koenkk/Z-Stack-firmware/issues/72

mrneutron42, 2023-10-26:
@OzGav it sounds like you are saying that your router CC2531 USB stick had the same IEEE address as the cooridinator CC2531 USB stick??  That is really weird.  How would that happen?  Doesn't the IEEE address get programmed into the chip by TI?

OzGav, 2023-10-26:
Sorry can’t remember now but changing it solved the problem for me