Koenkk/zigbee2mqtt #4446: Failed to exec start script - Request better error message

plasticrake, 2020-09-23:
## What happened
Every few months zigbee2mqtt will crash on startup. The solution that fixes it for me is to unplug and replug the usb device. Rebooting the system does not help. However the log output is very cryptic. **It would be great if the error was caught and output a more helpful message.**

## What did you expect to happen
Startup without crash or **output a more helpful error message.**

## How to reproduce it (minimal and precise)
Can't reliably reproduce.

## Debug info
Zigbee2MQTT version: 1.14.4 (using docker)
Adapter hardware: CC2530
Adapter firmware version: CC2530_CC2591_SOURCE_ROUTING_20190619

```
6 info lifecycle zigbee2mqtt@1.14.4~start: zigbee2mqtt@1.14.4
7 verbose lifecycle zigbee2mqtt@1.14.4~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.14.4~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/app/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle zigbee2mqtt@1.14.4~start: CWD: /app
10 silly lifecycle zigbee2mqtt@1.14.4~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.14.4~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.14.4~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.14.4 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.14.4
15 verbose cwd /app
16 verbose Linux 4.19.0-10-amd64
17 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
18 verbose node v12.18.3
19 verbose npm  v6.14.6
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.14.4 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.14.4 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Koenkk, 2020-09-25:
This is the npm log, not the zigbee2mqtt log. You can find this log in stdout or the data folder.

xshocuspocusxd, 2020-10-01:
Try to change from this:

      - /dev/ttyAMA0:/dev/ttyACM0
      #- /dev/ttyACM0:/dev/ttyACM0

to this:

    devices:
      #- /dev/ttyAMA0:/dev/ttyACM0
      - /dev/ttyACM0:/dev/ttyACM0

plasticrake, 2020-11-01:
@Koenkk That was the stdout from docker. Also I do not believe anything was saved to the logs in my log. My log_level configuration is set to `warn` and I don't have any logs:

```
$ du -h
4.0K	./2020-09-08.19-52-51
4.0K	./2020-09-08.19-53-12
4.0K	./2020-10-06.09-06-53
4.0K	./2020-09-08.19-53-33
4.0K	./2020-09-08.19-51-48
4.0K	./2020-09-08.19-53-54
4.0K	./2020-10-22.07-20-59
4.0K	./2020-10-02.04-00-16
4.0K	./2020-09-08.19-52-09
4.0K	./2020-09-08.19-52-30
```

@xshocuspocusxd Was your response posted to the wrong issue? Not sure what this is referring to.

Koenkk, 2020-11-02:
@plasticrake what is shown when executing with `npm start`? I need some logging to debug this further.

plasticrake, 2020-11-11:
@Koenkk I'm not having the issue anymore, but if this pattern continues I should have it again in a few months. Unfortunately I need my zigbee network up so I just unplug and restart to fix it, can't really spend days troubleshooting with my network down.

I run this in docker, can you explain how I should be running this to better debug in the future? When the container starts it just gets in a restart loop with failing and retrying. How do I run `npm start` in this case?

closed 2020-12-20 by github-actions[bot]