Koenkk/zigbee2mqtt #4452: Renaming devices (just friendly name) breaks Home Assistant integration

ciotlosm, 2020-09-24:
## What happened
**Steps**
- Get Mosquitto MQTT and setup integration using auto discovery
- Get Z2m add-on and configure it to use core-mosquitto and pick zzh! with latest firmware
- Start everything up and make sure it's connected
- Start pairing process for your devices and watch them appear in your Lovelace HA UI
- Rename the devices using Home Assistant: Configuration - Entities
- Also rename the Devices 
- 🎉 Finally get the new Frontend for Z2m!
- What the network be unreadable - decide to rename friend names (not HA IDs!)
- Everything looks nice at this point - Even if you did double work to rename both in HA and Z2m - but you used UI only so bonus for showing to friends that are not that good with tech (where you can provide a starter kit)

**Problem**
- Turning my Cube did not trigger my usual animation - No problem, check logs, it's still there, battery good... hmm
- Using another button (battery powered) also doesn't do anything (!Panic)

Seems for some reason the device queues in Home Assistant for the entities are expecting the ID to be used, but the queues coming from Z2m are no using friendly names.

```
INFO: MQTT publish: topic 'zigbee2mqtt/Mansardă Cub/action', payload 'flip90'
```

## What did you expect to happen
I was hoping that the change won't break my integration. 

## How to reproduce it (minimal and precise)
- Pair devices 
- Rename them in Home Assistant to nice names
- Create automations based on the devices (using IDs) in Home Assistant
- Test automations - Works
- Rename devices Friendly Name only in Z2m 
- Watch the automations fail to trigger once you renamed them

## Debug info
Zigbee2MQTT version:  1.14.4-dev (bbc7c7d)
Adapter hardware: CC26X2R1 (zzh!)
Adapter firmware version:
```
"coordinator": {
        "meta": {
            "maintrel": 1,
            "majorrel": 2,
            "minorrel": 7,
            "product": 1,
            "revision": 20200805,
            "transportrev": 2
        },
        "type": "zStack3x0"
    }
```

nurikk, 2020-09-25:
Remnaming device in something like foo/bar/baz and click on Update home assistant ID also causing issue
[image]

Koenkk, 2020-09-25:
@ciotlosm did you set the "Update Home Assistant entity ID" in the UI to true? Because if so this is expected behaviour.

ciotlosm, 2020-09-25:
@Koenkk nope, I did not, I just renamed the Friendly Name.

Koenkk, 2020-09-25:
Fixed in latest dev (together with some other fixes). Can you confirm?

ciotlosm, 2020-09-26:
Doesn't seem to be fixed, mqtt topic still changes to friendly name instead of using id.  

```
Zigbee2MQTT:info  2020-09-26 08:20:05: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0x00158d0001a66d6c","homeassistant_rename":false,"to":"Button"},"status":"ok"}'
Zigbee2MQTT:info  2020-09-26 08:20:09: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"action":"single","battery":100,"click":"single","linkquality":76,"voltage":3085}'
Zigbee2MQTT:info  2020-09-26 08:20:09: MQTT publish: topic 'zigbee2mqtt/Button', payload '{"action":"","battery":100,"linkquality":76,"voltage":3085}'
```

```
Zigbee2MQTT:info  2020-09-26 08:18:33: Starting Zigbee2MQTT version 1.14.4-dev (commit #fced7ed)
```

Koenkk, 2020-09-26:
That is expected, but Zigbee2MQTT will now update the discovery information in Home Assistant.

ciotlosm, 2020-09-27:
Tested button actions (after rename) and it does seem to work. For me personally I would have better liked mqtt topics to be using the ID, as the ID is used in Home Assistant automations and Friendly Name can have white space or special characters that are made for display, not much for machine interpretation. But this is a personal view.

closed 2020-09-27 by ciotlosm

McFarTech, 2022-09-01:
Was this ever fixed? I am having an issue of devices showing as hidden in HA after they have been renamed.

pscloud-patrick, 2024-12-18:
I have this problem to. If i rename a device in Z2M, its broken in HA and I need to reconfigure it. And yes, I checked update in Homeassistant switch!