Koenkk/zigbee2mqtt #4455: Support of CC2538SF53 2450BM15A0002 CP2102

Pascal51882, 2020-09-25:
I found a version of the CC2538 + CC2592 with USB (looks like CC2531 only longer).
I am new to this topic and wonder if your firmware would work also on this?
I cant find a serial connector on the board for flashing.
Here a picture of it:
[image]
The information from the supplier:
CC2538, CC2592, USB Dongle ZigBee adapter
CC2592 RF Front End includes PA, LNA, switches, Balance - balun and RF matching, which will increase the power output 2.4GHz low-power RF solutions and increase its receiver sensitivity.
CC2538 + CC2592 combined as required distances (higher output power) and higher sensitivity of IEEE 802.15.4 application provides a powerful solution.
Powerful ARM Cortex ™ M3,512KB programmable Flash, 32KB RAM
USB 2.0 full-speed device (12Mbps)
The chip use CC2538SF53，2450BM15A0002，CP2102,

Thanks for the great work so far :)

Koenkk, 2020-09-25:
@kirovilya @reverieline do you know if the existing CC2538 would work on this?

kirovilya, 2020-09-27:
https://github.com/Koenkk/zigbee2mqtt/issues/1568

This is the first time I've seen such a stick. where is it sold?
You need to look at which pins the amplifier is connected to. If they are the same as ours, then the firmware will work.
Well, you need to look at how to connect the JTAG connector for firmware, look for contacts.

Pascal51882, 2020-09-27:
[image]
[image]
I am no quiet sure if its the same layout.
They listed to pictures. One of it with JTAG Pins but I can not see anything of this on the product pictures

Pascal51882, 2020-10-28:
If someone knows if it would work here is also a link:
https://german.alibaba.com/product-detail/cc2538-cc2592-usb-dongle-zigbee-adapter-60426940995.html?spm=a2700.galleryofferlist.normal_offer.d_image.29413272cr1nH7

Sadly you can only buy high amounts and shiping is really expensive on Alibaba

closed 2020-12-06 by github-actions[bot]