Koenkk/zigbee2mqtt #4456: TuYa TS0601 - Time synchronization problem

longos, 2020-09-25:
Every time I manually set the time on the **TS0601** valve, it soon changes to a different time.
The manual says that the time will be the same as the gate time after pairing. Perhaps some time stamp is missing in the MQTT frames and the clock in the valve is resetting?

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter firmware version: 20190608

helmuteke, 2020-09-30:
Today i noticed the same problem , but i've add on of my four devices to a conbee gateway but also this thermostat forgot the time

longos, 2020-10-01:
After updating the valve firmware to 1.1.5 it's time to start syncing ALMOST correct.
At the moment, the time is always saved with a 20-minute delay.

helmuteke, 2020-10-01:
For me is update firmware new. How can i do this easy , using home assistant

longos, 2020-10-01:
> For me is update firmware new. How can i do this easy , using home assistant

You can only do this through the Tuya Gateway and their app downloaded from the Google Play Store.

helmuteke, 2020-10-01:
hmm , i don't have a Tuya gateway . . . no workaround ?

longos, 2020-10-01:
> hmm , i don't have a Tuya gateway . . . no workaround ?

No

helmuteke, 2020-10-09:
So , i’ve update yesterday to 1.1.5 but the problem with time sync is the same. Difference of more then a hour

marcin77, 2020-10-19:
Latest firmware and difference few hours.

longos, 2020-10-28:
The same on my.

closed 2020-12-05 by github-actions[bot]

dcboy87, 2020-12-11:
Is there any solution known? it seems, z2m can't update firmware of this device, even if we could get the firmware file

flrahe, 2020-12-24:
I seems to me that there is a time sync when pairing / reconnecting. Then the time is correct (i.e. when starting after removing the batteries). When I installed my first valve I was surprised that the time was set accurate to the minute. But after one or two days it's off by twenty minutes. That's too large. Can't we do force such a time sync? Maybe by an command. I think there must be something, since it happens during pairing / reconnecting...

florian-asche, 2021-01-09:
It should be fairly easy to add such function to the firmware. Does anyone have contact to the developers?

k3mpaxl, 2021-01-11:
The solution for these time sync problems can be found here:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3304
I did a capture with the original gateway and BHT-002 and HY368.

yeld78, 2021-01-14:
I still have the same problem. Just after adding the time is correct. But after 24 hours it differs by about 20 mins

pawol, 2021-07-29:
reset TRV to defaults and rapair

zhvial, 2022-01-06:
Hello. 
I have 5 of these thermostat valves. Everything worked perfectly for the past 6 months, but some days ago I have found that time is not set correctly. Time is random on all valves. Even week day is random. I've tried to set time manually, but after couple of seconds it jumps back to time set before manual. I've updated all valve firmware to the latest one. I've tried to reconnect valve and to reset it but after reconnecting time is set wrong. I suppose that it might be related to the latest HA or Z2M updated. Nothing else was changed. Does anyone else have the same issue?

zhvial, 2022-01-06:
found that problem is known already. https://github.com/Koenkk/zigbee2mqtt/issues/10489

pawol, 2022-01-06:
It seems to me that time on z2b 1.22.1 eas synchronized properly but now on 1.22.2 I have to set time manually on my TS0601.