Koenkk/zigbee2mqtt #4457: GBAN GB2530 firmware updating

b4dpxl, 2020-09-25:
## What happened
I have an QBKG25LM which I cannot call from z2m, although it pairs and receives updates from the switch fine. I get the "No network route" error. I'm using a GBAN GB2530-S device. I don't know what version of firmware it's running, but it's fairly old. Around 04/2019 I think. To try and fix the issue with this switch, I wanted to try the "source routing" firmware. I flashed both the CC2530_CC2592_SOURCE_ROUTING_20190619 and CC2530_SOURCE_ROUTING_20190619 versions (for some reason I _think_ I had a CC2592 firmware before).

After flashing, neither version could route to existing devices, and I was unable to remove and re-pair a device (e.g. WXKG02LM). It seemed to go through the pairing process correctly, but afterwards pressing the buttons on the re-paired switch did not generate any messages in the logs or send any mqtt messages. I also tried removing and force-removing the switch, which didn't help.

I reverted the firmware (I took a backup before I started) and everything worked as expected.

* What is the correct source-routing firmware version for the GB2530-S?
* Once flashed, should existing devices be able to reconnect automatically, or do they need re-pairing? If so, is there a specific route to remove and re-pair the devices?
* I know there are device limits on the source routing versions. Do I need to remove everything _before_ flashing, then re-add other routers (e.g. tradfri bulbs and wired xiaomi switches) first?

## What did you expect to happen
Ability to connect devices, either with or without re-pairing.

## How to reproduce it (minimal and precise)
Update firmware on GBAN GB2530, restart z2m, try re-pairing devices

## Debug info
Zigbee2MQTT version: Not sure, but updated in the lsat few weeks
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version: CC2530_CC2592_SOURCE_ROUTING_20190619 and CC2530_SOURCE_ROUTING_20190619

b4dpxl, 2020-09-25:
Bit more information. I have 5 routing devices (1 wired Xiaomi switch with neutral, and 4 tradfri bulbs) and only 13 non-routing devices (xiaomi wireless switches, one no-neutral wired switch, and various xiaomi sensors)

Koenkk, 2020-09-25:
Maybe try the CC2591 version?

b4dpxl, 2020-09-25:
Will do. With the limitation on the number of devices on the source routing version, do I need to do anything about this?

> You’ve hit the device limit of the coordinator, especially occurs when using the CC2531 or CC2530 in combination with the source routing firmware. Try reflashing the coordinator and immidiately pair the device after starting Zigbee2MQTT.

Or should it handle that automatically?

Koenkk, 2020-09-25:
I would first pair the routers, after that end devices.

b4dpxl, 2020-09-26:
I was hoping to avoid unpairing and re-pairing everything. If I’ve got to go down that route I’ll just swap to a ZZH instead.

closed 2020-11-04 by github-actions[bot]