Koenkk/zigbee2mqtt #4459: RTCGQ11LM (Aqara Motion sensor) not reporting occupancy: true

vytautasgimbutas, 2020-09-25:
## What happened
Sometimes instead of reporting `occupancy: true` it reports `genBasic`:
```
debug 2020-09-25 18:58:50: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-25 18:58:50: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":90,"occupancy":false,"voltage":3025}'
debug 2020-09-25 18:59:15: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":30,"4":5032,"5":6,"6":[0,65545]}}' from endpoint 1 with groupID 0
info  2020-09-25 18:59:15: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":87,"occupancy":false,"voltage":3025}'
```

I'm creating a new issue as I'm not entirely sure it's related with https://github.com/Koenkk/zigbee2mqtt/issues/2274 as I haven't found anyone mentioning `genBasic`.

This happens with all (and only) aqara motion sensors. With some more frequently with some less.

If this is the same as the 2274 issue - please close this and i'll post there as a comment. Thank you!

## What did you expect to happen
Report occupancy: true

## How to reproduce it (minimal and precise)
Hard to reproduce. It just happens sometimes.

## Debug info
Zigbee2MQTT version: currently latest master
Adapter hardware: CC2531 and zzh! both have this issue
Adapter firmware version:

Koenkk, 2020-09-25:
This is indeed the same issue as #2274, `genBasic` is an hourly report that a lot of Xiaomi end devices do.

vytautasgimbutas, 2020-09-25:
@Koenkk weird. My motion sensor is directly connected to the coordinator. My only other router is Tuya Curtain Motor. Regarding `genBasic`: when I entered the room with motion sensor instead of reporting `occupancy:true` it did that. I doubt that this is a coincidence :(

Koenkk, 2020-09-25:
That is interesting! Can you reproduce this a 5 times and provide the logging of this like you did in the OP.

vytautasgimbutas, 2020-09-25:
Usually I can reproduce this 5-10 times a day so I will try to get back regarding this ASAP.

Koenkk, 2020-09-25:
the rest of the times it works OK?

vytautasgimbutas, 2020-09-25:
Yes. They work very well when they do. Right now I'm wondering if it can be that the device goes to sleep just right after it sends a report? If yes, maybe i'm "lucky" to enter the room within that minute after the report was sent.. I'll try send some notification myself when `genBasic` is seen in the logs and get my ass ASAP to the room :) 

```
debug 2020-09-25 18:59:15: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":30,"4":5032,"5":6,"6":[0,65545]}}' from endpoint 1 with groupID 0
debug 2020-09-25 19:48:22: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":29,"4":5032,"5":6,"6":[0,7]}}' from endpoint 1 with groupID 0
debug 2020-09-25 20:38:49: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":29,"4":5032,"5":6,"6":[1,16842761]}}' from endpoint 1 with groupID 0
debug 2020-09-25 21:29:07: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":29,"4":5032,"5":6,"6":[0,3]}}' from endpoint 1 with groupID 0

vytautasgimbutas, 2020-09-25:
Ok, after all it was just a coincidence. Not sure why that time I haven't received illuminance report. This time I tried to enter the room just right after the occupancy timeout.

It turns off (occupancy_timeout is 150s):
{code}
info  2020-09-25 22:15:01: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":99,"occupancy":false,"voltage":3025}'
{code}

After a few seconds I try to enter the enter the room and this happens:
{code}
debug 2020-09-25 22:15:07: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-25 22:15:07: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":93,"occupancy":false,"voltage":3025}'
{code}

I have wait for a minute for it to able send `occupancy:true` event:
{code}
debug 2020-09-25 22:16:13: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
info  2020-09-25 22:16:13: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":93,"occupancy":false,"voltage":3025}'
debug 2020-09-25 22:16:13: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
info  2020-09-25 22:16:13: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":96,"occupancy":true,"voltage":3025}'
{code}

After 3 minutes it sends a `genBasic`:
{code}
debug 2020-09-25 22:19:04: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":29,"4":5032,"5":6,"6":[1,16908294]}}' from endpoint 1 with groupID 0
{code}

I guess it's very related with that issue but I don't use any OSRAM devices and this is happening with two kind of coordinators. Trying to think any patterns that would help solve this and the only one that comes to my mind is: a few minutes before it sends a `genBasic` report it can misbehave. Could be totally false, but I'll try to check this theory as well. :)

My configuration.yml:
{code}
  '0x00158d00048604f0':
    friendly_name: '0x00158d00048604f0'
    occupancy_timeout: 150
{code}

Not sure if this will help but here's a little bit background: up until a few days ago I've been using CC2531 and sensors  were working for a few weeks without a problem. Then it suddenly becomes worse and worse. I thought maybe switching to a better one like zzh! would help but actually it's the opposite. Once the sensor is repaired it starts skipping `occupancy:true` events the very same day!

Today I even unpacked a brand new sensor and tried replacing but it didn't help.

Here's my network map:

[image]

If there's anything else I can do to help solve this issue please let me know.

Koenkk, 2020-09-26:
Since you have a spare CC2531 now, could you sniff the traffic? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

In the left wireshark show the `No.`. Clearly mention what you did on a certain `No.`. E.g. trigged occupancy at no 140.

vytautasgimbutas, 2020-09-30:
Sorry for not getting back earlier. Sensor worked fine for a couple days and finally today it started skipping occupancy events. I'll try to sniff traffic this week.

``` 
info  2020-09-30 11:41:28: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":94,"illuminance_lux":94,"linkquality":87,"occupancy":false,"voltage":3015}'

### lights turned off i was waving in front of the sensor (11:41:28 - 11:42:20~) but nothing happened ###
### sensor after a minute reported genBasic, together with `occupancy: false` event  ###

debug 2020-09-30 11:42:44: Received Zigbee message from '0x00158d00048604f0', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"10":0,"100":0,"11":0,"3":29,"4":5032,"5":6,"6":[0,131077]}}' from endpoint 1 with groupID 0
info  2020-09-30 11:42:44: MQTT publish: topic 'zigbee2mqtt/0x00158d00048604f0', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":87,"occupancy":false,"voltage":3025}'

McGiverGim, 2020-11-02:
I think I'm suffering the same problem.

Suddenly, two Aqara occupancy sensors (RTCGQ11LM) report the illuminance but not the occupancy. I have read that this can happen if they connect with Osram routers but in my case both are connected using an Aqara router (LLKZMK11LM). This has been working for several months.

Repairing the occupancy sensors make them work at least once, but they start to fail again immediately. I have not an spare cc2531 to sniff the traffic :(

closed 2020-12-10 by github-actions[bot]