Koenkk/zigbee2mqtt #4460: SHOJZJ 378RT aka TuYa TS0601 can't set value

longos, 2020-09-25:
## Can't set value
During changing thermostat set value in Domoticz, TRV don't change this value. After that, when I check value on valve, there still were old value. Beside this, valve send ZigBee message, and overwrite set in Domoticz.

Also, when I [force device to send local_temperature](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html#local-temperature) it doesn't work.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter firmware version: 20190608

Bartekn86, 2020-09-27:
update firmware using tuya gateway, that should fix problem

longos, 2020-09-28:
> update firmware using tuya gateway, that should fix problem

OK, I try this, thx. This issue is also fixed: [#3824](https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-699644902)

closed 2020-09-28 by longos

SLongus, 2020-12-15:
Hello, I have the same problem using cc2531 and iobroker. I don't have a Tuya Gateway.