Koenkk/zigbee2mqtt #4461: Zemismart Switch is not update relay status using conbee 2 as coordinator

juslex, 2020-09-25:
## What happened

I am using conbee 2 as coordinator and everything is working well.

I added Smart light switch - 3 gang v2.0 (HGZB-43) and Smart light switch - 2 gang v2.0 (HGZB-42) device and it's working but the physical switch don't update the MQTT, but if I send command via MQTT it's turns on and off the relay as expected.

I am using the latest stable version 1.14.4.4.2 in Home Assistant

## What did you expect to happen

Physical switch update the MQTT.

## How to reproduce it (minimal and precise)

Just add this device using conbee 2 as coordinator.

## Debug info
Zigbee2MQTT version: 1.14.4.2
Adapter hardware: Conbee II
Adapter firmware version:

Koenkk, 2020-09-26:
Try to reconfigure the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and check the logs if it is OK.

juslex, 2020-09-26:
Dear Koenkk,

I reconfigured according to your guidance, when analyzing the logs, there was confirmation that it was a success, but the problem persists. However, because of the hurry, I didn't explain the whole history. I installed 8 of these switches 4 of 3 gangs and 4 of 2 gangs. All using conbee 2 as coordinator. Of the 8, two who are not the ones who are having a problem, had the same symptom. I had to repairing to solve it. Everyone worked well until a week ago, half of them stopped updating their status. I'll try repairing but without success.

Log files:

Zigbee2MQTT:info  2020-09-26 17:31:12: Configuring 'Interruptor Sala 1'
Zigbee2MQTT:info  2020-09-26 17:31:13: Successfully configured 'Interruptor Sala 1'
Zigbee2MQTT:info  2020-09-26 17:31:26: Configuring 'Interruptor Corredor Acima'
Zigbee2MQTT:info  2020-09-26 17:31:27: Successfully configured 'Interruptor Corredor Acima'

Koenkk, 2020-09-27:
@juslex what we fixed in #4245 is already in for your devices. I don't know why it's not working in your case. Could you sniff the traffic when configuring and manually controlling the switch? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

juslex, 2020-10-01:
Dear Koenkk,

I don’t have anymore the cc2531 then I can’t do the sniffer test.

One more thing: the fix from my devices from #4245  is present in the latest stable version or still in  dev branch?

Koenkk, 2020-10-01:
Those changes will be in todays release (1.15.0)

juslex, 2020-11-22:
Dear @Koenkk,

I'm reopening the issue. After informing you that you did not have a cc2531 at the time to make the sniffer I get a zzh! To replace conbee 2. However, the problem I reported in this issue with zemismart switches did not reappear in zzh!. So I decided to go back to conbee taking into account the new firmware released in early October and version .16.1 of zigbee2mqtt in the hope that the reported problem is gone. After a few days of use, the problem repeated itself again, including the dev version. If I remove the switch and read, the problem disappears, reappearing days later. Even on the same switch, I can have an update of the top, center and not the botton, for example. The same situation is repeated on other switches with 2 or 3 buttons of the same model. And there are some switches of the same model without problem.

And as I said, this problem only occurs with conbee2, zzh and cc2531 do not present this problem of absence of status update when you physically press the switch button.

Can I send you any more files or excerpts to analyze the reason for this change in behavior after a few days? I can make a snifer with zzh! Instead of cc2531?

juslex, 2020-11-22:
AliExpress page of zemismart switch, 1, 2 or 3 gangs:

https://a.aliexpress.com/_mrKwuQv

Page of zigbee2mqtt assistant with device:

[image]

juslex, 2020-11-22:
Two Gangs with problem:

[image]

Three Gangs with problem:

[image]

juslex, 2020-11-22:
Two Gangs with problem: 
[image]

Jhonattan-Souza, 2020-11-22:
Same problem here with Conbee II
If you guys need any help, I'll be glad to help

Jhonattan-Souza, 2020-11-22:
@juslex Call me at Telegram if you want, @Jhoniin

juslex, 2020-11-23:
Extract from database.db: 

Threes Gangs with problems:

{"id":3,"type":"Router","ieeeAddr":"0x00158d00057f3c28","nwkAddr":34744,"manufId":4478,"manufName":"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSW03LX2.0","epList":[1,2,3],"endpoints":{"1":{"profId":49246,"epId":1,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FNB56-ZSW03LX2.0","manufacturerName":"FeiBit","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180808","swBuildId":"1000-0001"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}},"2":{"profId":49246,"epId":2,"devId":0,"inClusterList":[0,4,3,6,5,8],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}},"3":{"profId":49246,"epId":3,"devId":0,"inClusterList":[0,4,3,6,5,8],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180808","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1606096833021}

{"id":6,"type":"Router","ieeeAddr":"0x00158d00054edd44","nwkAddr":18703,"manufId":4478,"manufName":"FeiBit","powerSource":"Mains (single phase)","modelId":"FNB56-ZSW03LX2.0","epList":[1,2,3],"endpoints":{"1":{"profId":49246,"epId":1,"devId":0,"inClusterList":[0,4,3,6,5,4096,8],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"FNB56-ZSW03LX2.0","manufacturerName":"FeiBit","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180808","swBuildId":"1000-0001"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}},"2":{"profId":49246,"epId":2,"devId":0,"inClusterList":[0,4,3,6,5,8],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}},"3":{"profId":49246,"epId":3,"devId":0,"inClusterList":[0,4,3,6,5,8],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05b15b","endpointID":1}],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20180808","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1606096835076}

Two Gangs with problem:

Jhonattan-Souza, 2020-11-24:
Hi @Koenkk , i'ts happening to me as well

Mine is

{"id":2,"type":"Router","ieeeAddr":"0x00124b001c87b0b6","nwkAddr":52739,"manufId":0,"manufName":"Feibit Inc co.  ","powerSource":"Mains (single phase)","modelId":"FB56+ZSW1GKJ2.7","epList":[16],"endpoints":{"16":{"profId":260,"epId":16,"devId":2,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"FB56+ZSW1GKJ2.7","manufacturerName":"Feibit Inc co.  ","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190601        "}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff060d6b","endpointID":1}],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20190601        ","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1606167910048}

devices.js content

    zigbeeModel: ['FB56+ZSW1GKJ2.7'],
    model: 'HGZB-42',
    vendor: 'Feibit',
    description: 'Smart one gang wall switch',
    extend: generic.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(16);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint);
    },

I've followed the tutorial to support new devices, Tried to reconfigure, but nothing gives me feedback from physical button, Conbee just don't catch the physical button message, event with "zigbee_herdsman_debug: true" and "level: debug",
The problem is not ocurring with Smartthings, look at the message that comes from smartthings when i press the physical button

[raw:0104 0006 10 0B 0000 00 6559 00 00 0000 0A 01 00002001, profileId:0104, clusterId:0006, sourceEndpoint:10, destinationEndpoint:0B, options:0000, messageType:00, dni:6559, isClusterSpecific:false, isManufacturerSpecific:false, manufacturerId:0000, command:0A, direction:01, attrId:0000, encoding:20, value:01, isValidForDataType:true, data:[00, 00, 20, 01], clusterInt:6, attrInt:0, commandInt:10]

Any help would be appreciated, thanks in advance

Koenkk, 2020-11-24:
@Jhonattan-Souza can you sniff the traffic while reconfiguring the device? (please provide the wireshark .pcapng file and your network key) https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

juslex, 2020-11-24:
Dear @Koenkk 

is there any way to sniff using ssh? Due to the need to try to solve the problem, I purchased a CC2531, but it will take a few weeks to arrive.

juslex, 2020-12-02:
Dear @Koenkk 

another user with the same problem

Jhonattan-Souza, 2020-12-10:
Hi @Koenkk 

Here it is my traffic file.
https://drive.google.com/file/d/1T_FOzh9Wkl3Fe9QB5PGk0TKYLWRgVEUq/view?usp=sharing

My network key is the default network key -> 01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D

Please, use the filter zbee_nwk.addr eq 0xe647 to see only the specific device on-off events, (i have pressed many times) 

As you can see, the messages are being transmited when i press the button, but nothing happens at z2mqtt

Thanks !

Koenkk, 2020-12-10:
@Jhonattan-Souza thanks!

@chrishae I think the issue here is that the switch sends messages to endpoint 11, could it be that the Conbee firmware does not forward messages to endpoint 11 to zigbee-herdsman?

[image]

Jhonattan-Souza, 2020-12-10:
@ChrisHae good night. I have some new information that might help.

I tried Deconz at HA to verify if it works correctly, and there is feedback from the physical button. It works, but there`s a lag, 2 to 3 seconds to change the status. 

If you need any help to test anything else, just ask ! 

[image]

[image]

[image]

ChrisHae, 2020-12-11:
Is this switch bound to EP11 during configuration? If so, can it changed to be bound to EP1?

Koenkk, 2020-12-11:
@ChrisHae good point

@Jhonattan-Souza can you provide the herdsman debug logging when reconfiguring the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Jhonattan-Souza, 2020-12-11:
https://drive.google.com/file/d/1Aj2CeH_ZZN-8wY1c8W4R3lbpdIFkRfDx/view?usp=sharing

@Koenkk here it is. 

Now the filter is zbee_nwk.addr eq 0x1af1

Thank you !

Jhonattan-Souza, 2020-12-11:
@Koenkk and here, my zigbee-herdsman debug, from the permit join until the end. 

Zigbee2MQTT:debug 2020-12-11 17:52:44: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"value":true}',
Zigbee2MQTT:info  2020-12-11 17:52:44: Zigbee: allowing new devices to join.,
2020-12-11T20:52:44.303Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffc EP:0 SeqNr. 43 request id: 8,
2020-12-11T20:52:44.305Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 8 status: 0,
2020-12-11T20:52:44.305Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.346Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110,
2020-12-11T20:52:44.346Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.402Z zigbee-herdsman:deconz:driver query aps data confirm,
2020-12-11T20:52:44.408Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 44,
2020-12-11T20:52:44.410Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffc request id: 8 confirm status: 0,
2020-12-11T20:52:44.411Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.411Z zigbee-herdsman:deconz:adapter PERMIT_JOIN - 254 seconds,
2020-12-11T20:52:44.412Z zigbee-herdsman:deconz:adapter zclFrame to all - zclFrame.payload:,
2020-12-11T20:52:44.412Z zigbee-herdsman:deconz:adapter { options: 11, commisioningWindow: 254 },
2020-12-11T20:52:44.412Z zigbee-herdsman:deconz:adapter sendZclFrameToAll - message send,
2020-12-11T20:52:44.504Z zigbee-herdsman:deconz:driver send write parameter request from queue. seqNr: 45 paramId: 33 param: 254,
2020-12-11T20:52:44.505Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 46 request id: 9,
2020-12-11T20:52:44.506Z zigbee-herdsman:deconz:frameParser write parameter response - parameter id: 33 - status: 0,
2020-12-11T20:52:44.507Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 9 status: 0,
2020-12-11T20:52:44.507Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.561Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110,
2020-12-11T20:52:44.562Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.603Z zigbee-herdsman:deconz:driver query aps data confirm,
2020-12-11T20:52:44.608Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 47,
2020-12-11T20:52:44.610Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 9 confirm status: 0,
2020-12-11T20:52:44.610Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:52:44.704Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 48 paramId: 5,
2020-12-11T20:52:44.705Z zigbee-herdsman:deconz:frameParser PANID: 1a62,
2020-12-11T20:52:44.804Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 49 paramId: 11,
2020-12-11T20:52:44.806Z zigbee-herdsman:deconz:frameParser APS_EXT_PANID: 0xdddddddddddddddd,
2020-12-11T20:52:44.905Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 50 paramId: 28,
2020-12-11T20:52:44.907Z zigbee-herdsman:deconz:frameParser CHANNEL: 11,
2020-12-11T20:52:45.006Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 51 paramId: 24,
2020-12-11T20:52:45.007Z zigbee-herdsman:deconz:frameParser NETWORK_KEY: hidden,
Zigbee2MQTT:info  2020-12-11 17:52:45: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"1.16.2","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00124b001b7600b2":{"friendly_name":"ventilador_quarto"},"0x00158d0002008d36":{"friendly_name":"botao_quarto"},"0x00158d0002672277":{"friendly_name":"sensor_presenca_corredor"},"0x00158d0002753c41":{"friendly_name":"sensor_portao"},"0x00158d0002b7b980":{"friendly_name":"luz_area_externa"},"0x00158d0002c3ec13":{"friendly_name":"botao_cozinha"},"0x00158d0002e33a0c":{"friendly_name":"sensor_racao"},"0x00158d00045a2c21":{"friendly_name":"botao_terraco"},"0x00158d0004775eec":{"friendly_name":"sala_switch_3"},"0x00158d000542b164":{"friendly_name":"cubo_aqara"},"0x04cf8cdf3c787d93":{"friendly_name":"sensor_de_luz"},"0x04cf8cdf3c7db436":{"friendly_name" [... cut ...]
2020-12-11T20:55:49.881Z zigbee-herdsman:deconz:frameParser device state: 10100010,
2020-12-11T20:55:49.881Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-11T20:55:59.907Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 144,
2020-12-11T20:55:59.909Z zigbee-herdsman:deconz:frameParser device state: 10100010,
2020-12-11T20:55:59.909Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,

Jhonattan-Souza, 2020-12-11:
@Koenkk and here, the zigbee-herdsman logs

https://drive.google.com/file/d/1LPPyXo46ABm0l9tzQaCEODBhjN_NNNE1/view?usp=sharing

My device is the -> '0x00124b001c87b0b6'  that joins the network.

juslex, 2020-12-12:
Dear @Koenkk anda @ChrisHae 

Here is my Zigbee-herdsman logs in configure mode from my two devices: 1) FNB56-ZSW02LX2.0 - HGZB-42 - 2 Gang, named in my network: **Interruptor Corredor Acima** 2) FNB56-ZSW03LX2.0 - HGZB-43 - 3 Gang, named in my network: **Interruptor Escada Abaixo**

https://www.dropbox.com/s/54ukup1079d81b8/Zigbee-herdsman%20debug.txt?dl=0

Koenkk, 2020-12-13:
@Jhonattan-Souza did you reconfigure your device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure in these sniff/logs? (I don't see the expected logging)

Jhonattan-Souza, 2020-12-13:
@Koenkk I get this error trying to reconfigure

INFO: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"luz_escritorio"},"error":"Device 'luz_escritorio' cannot be configured","status":"error"}'

Then , instead of reconfigure, I re-paired the device and sent the logs. Do I have to change the devices.js to reconfigure?

Jhonattan-Souza, 2020-12-13:
And from the Mqtt Configure method

Zigbee2MQTT:debug 2020-12-13 13:56:02: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'luz_escritorio'
Zigbee2MQTT:warn  2020-12-13 13:56:02: Skipping configure of 'luz_escritorio', device does not require this.

Koenkk, 2020-12-13:
@Jhonattan-Souza ah I see this device has no configure in devices.js, can you  change it to:

```js
{
    zigbeeModel: ['FB56+ZSW1GKJ2.5', 'LXN-1S27LX1.0', 'FB56+ZSW1GKJ2.7'],
    model: 'HGZB-41',
    vendor: 'Nue / 3A',
    description: 'Smart one gang wall switch',
    extend: generic.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        console.log('bind time!');
        console.log(coordinatorEndpoint);
        const endpoint = device.getEndpoint(16);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

Jhonattan-Souza, 2020-12-13:
@Koenkk 
Here it goes !

Zigbee2MQTT:debug 2020-12-13 14:50:07: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'luz_escritorio',
Zigbee2MQTT:info  2020-12-13 14:50:07: Configuring 'luz_escritorio',
bind time!,
Endpoint {,
  ID: 1,,
  profileID: 260,,
  deviceID: 5,,
  inputClusters: [ 25, 10 ],,
  outputClusters: [ 1280 ],,
  deviceNetworkAddress: 0,,
  deviceIeeeAddress: '0x00212effff060d6b',,
  clusters: {},,
  _binds: [],,
  meta: {},
},
2020-12-13T17:50:07.654Z zigbee-herdsman:controller:endpoint Bind 0x00124b001c87b0b6/16 genOnOff from '0x00212effff060d6b/1',
2020-12-13T17:50:07.656Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xafd2 EP:0 SeqNr. 38 request id: 6,
2020-12-13T17:50:07.659Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 6 status: 0,
2020-12-13T17:50:07.659Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T17:50:07.681Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110,
2020-12-13T17:50:07.681Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T17:50:07.692Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110,
2020-12-13T17:50:07.692Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T17:50:07.772Z zigbee-herdsman:deconz:driver query aps data confirm,
2020-12-13T17:50:07.871Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 39,
2020-12-13T17:50:07.873Z zigbee-herdsman:deconz:driver query aps data indication,
2020-12-13T17:50:07.874Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xafd2 request id: 6 confirm status: 0,
2020-12-13T17:50:07.874Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T17:50:07.874Z zigbee-herdsman:deconz:driver query aps data indication,
2020-12-13T17:50:07.972Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 40,
2020-12-13T17:50:07.974Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 40 srcAddr: 0xafd2 destAddr: 0x0 profile id: 0x0 cluster id: 0x8021,
2020-12-13T17:50:07.975Z zigbee-herdsman:deconz:frameParser response payload: 6,0,
2020-12-13T17:50:07.975Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.,
2020-12-13T17:50:07.975Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T17:50:07.975Z zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0xafd2 status: 0,
Zigbee2MQTT:info  2020-12-13 14:50:07: Successfully configured 'luz_escritorio'

Koenkk, 2020-12-13:
Looks good, do you know get status updates when pressing the button? (and can you sniff this again)

Jhonattan-Souza, 2020-12-13:
@Koenkk Here it goes, Reconfigured the device, and pressed the on off button many times 
In the sniff, log appears, but in the z2mqtt container, nothing changes. No logs.

https://drive.google.com/file/d/1Gk7feWU-CDpr9i2IkYHREozt7uee8Vmb/view?usp=sharing

Koenkk, 2020-12-13:
@Jhonattan-Souza can you try with this, reconfigure and check if you get updates?

```js
{
    zigbeeModel: ['FB56+ZSW1GKJ2.5', 'LXN-1S27LX1.0', 'FB56+ZSW1GKJ2.7'],
    model: 'HGZB-41',
    vendor: 'Nue / 3A',
    description: 'Smart one gang wall switch',
    extend: generic.switch,
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        console.log('bind time!');
        console.log(coordinatorEndpoint);
        const endpoint = device.getEndpoint(16);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
         await configureReporting.onOff(endpoint);
    },
},
```

Jhonattan-Souza, 2020-12-13:
@Koenkk Reconfigured with the new configuration, no status changes at logs, and there`s no status feedback when i press the physical button.

however, when i press this button, the status changes to the correct status

[image]

**and the logs, when i press this button**

2020-12-13T18:27:11.538Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":15,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":32,"attrData":1}]},"address":45010,"endpoint":16,"linkquality":245,"groupID":null}'
2020-12-13T18:27:11.542Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-13T18:27:11.542Z zigbee-herdsman:deconz:adapter response received
Zigbee2MQTT:debug 2020-12-13 15:27:11: Received Zigbee message from 'luz_escritorio', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 16 with groupID null

**here it goes the logs reconfiguring the device:**

Zigbee2MQTT:debug 2020-12-13 15:21:53: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'luz_escritorio',
Zigbee2MQTT:info  2020-12-13 15:21:53: Configuring 'luz_escritorio',
bind time!,
Endpoint {,
  ID: 1,,
  profileID: 260,,
  deviceID: 5,,
  inputClusters: [ 25, 10 ],,
  outputClusters: [ 1280 ],,
  deviceNetworkAddress: 0,,
  deviceIeeeAddress: '0x00212effff060d6b',,
  clusters: {},,
  _binds: [],,
  meta: {},
},
2020-12-13T18:21:53.090Z zigbee-herdsman:controller:endpoint Bind 0x00124b001c87b0b6/16 genOnOff from '0x00212effff060d6b/1',
2020-12-13T18:21:53.095Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xafd2 EP:0 SeqNr. 39 request id: 6,
2020-12-13T18:21:53.098Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 6 status: 0,
2020-12-13T18:21:53.098Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.129Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110,
2020-12-13T18:21:53.129Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.139Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110,
2020-12-13T18:21:53.139Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.196Z zigbee-herdsman:deconz:driver query aps data confirm,
2020-12-13T18:21:53.296Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 40,
2020-12-13T18:21:53.297Z zigbee-herdsman:deconz:driver query aps data indication,
2020-12-13T18:21:53.298Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xafd2 request id: 6 confirm status: 0,
2020-12-13T18:21:53.298Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.298Z zigbee-herdsman:deconz:driver query aps data indication,
2020-12-13T18:21:53.395Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 41,
2020-12-13T18:21:53.398Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 41 srcAddr: 0xafd2 destAddr: 0x0 profile id: 0x0 cluster id: 0x8021,
2020-12-13T18:21:53.398Z zigbee-herdsman:deconz:frameParser response payload: 6,0,
2020-12-13T18:21:53.398Z zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr.,
2020-12-13T18:21:53.398Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.398Z zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0xafd2 status: 0,
2020-12-13T18:21:53.401Z zigbee-herdsman:controller:endpoint ConfigureReporting 0x00124b001c87b0b6/16 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}),
2020-12-13T18:21:53.402Z zigbee-herdsman:deconz:adapter zclFrame.payload:,
2020-12-13T18:21:53.403Z zigbee-herdsman:deconz:adapter [,
  {,
    direction: 0,,
    attrId: 0,,
    dataType: 16,,
    minRepIntval: 0,,
    maxRepIntval: 3600,,
    repChange: 0,
  },
],
2020-12-13T18:21:53.496Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xafd2 EP:16 SeqNr. 43 request id: 7,
2020-12-13T18:21:53.497Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 42,
2020-12-13T18:21:53.499Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 7 status: 0,
2020-12-13T18:21:53.499Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.499Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr.: 42 status: 5,
2020-12-13T18:21:53.508Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110,
2020-12-13T18:21:53.508Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
[... cut ...]
2020-12-13T18:21:53.800Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":7},"Payload":[]},"address":45010,"endpoint":16,"linkquality":245,"groupID":null}',
2020-12-13T18:21:53.800Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1,
2020-12-13T18:21:53.800Z zigbee-herdsman:deconz:adapter response received,
Zigbee2MQTT:info  2020-12-13 15:21:53: Successfully configured 'luz_escritorio',

juslex, 2020-12-13:
> however, when i press this button, the status changes to the correct status

Here the same thing: when i press this button, the status changes to the correct status.

Koenkk, 2020-12-13:
I'm afraid I have a bad news then. @ChrisHae it's not possible to bind this device to endpoint 1, it is bound out-of-the-box to endpoint 11 and doesn't allow setup any other bindings. To fix this issue the Conbee II firmware should be updated to pass through message to endpoint 11. This issue is not Zigbee2MQTT specific but will also happen when using this device with Deconz or ZHA.

juslex, 2020-12-13:
Very bad news ://///

My zigbee network is formed by 10 routers zemismart switch’s. These routers are the core of my mesh.

Jhonattan-Souza, 2020-12-13:
> @ChrisHae good night. I have some new information that might help.
>
> I tried Deconz at HA to verify if it works correctly, and there is feedback from the physical button. It works, but there`s a lag, [...]

@Koenkk Look at this post, I configured the device on deconz, and feedback works , should I try to sniff the traffic using Deconz instead of z2mqtt? It might help

Koenkk, 2020-12-13:
@Jhonattan-Souza then I was wrong in https://github.com/Koenkk/zigbee2mqtt/issues/4461#issuecomment-744048928 , yes please can you sniff the traffic when getting feedback with Deconz?

Jhonattan-Souza, 2020-12-13:
@Koenkk @ChrisHae 

Here it is, deconz traffic, the device reports status changes from the physical button, but there`s a huge lag, 2 to 3 seconds to update the status.

Please use the zbee_nwk.addr eq 0x7e91 filter, to see only the specific device logs.

https://drive.google.com/file/d/19kg9HnZG55oOpZjf2xnNt7LCWKxxwaJp/view?usp=sharing

Koenkk, 2020-12-13:
@Jhonattan-Souza thanks

@ChrisHae it seems that with Deconz the messages are also send to endpoint 11 so it should be possible to support this. Can you look into it?

ChrisHae, 2020-12-15:
I think the report attributes don't work because it is send to EP11. But deCONZ and the reload switch from that z2mqtt GUI send a read Attributes to the switch and that works. I think there is a way to get this work. I will respond again.

ChrisHae, 2020-12-16:
@Jhonattan-Souza Was this switch at another zigbee gateway before? And have you ever reset it?

juslex, 2020-12-16:
> @Jhonattan-Souza Was this switch at another zigbee gateway before? And have you ever reset it?

Dear @Koenkk and @ChrisHae 

A friend uses this switch with the cc2531 without any problem.

I used these switches with zzh! And this issue also didn't happen to him.

I had another type of problem that I suspect is also the fault of the zemismart switch with zzh!

Just to make my situation clear: I have 10 switches on my network and they are the only type of router in my mesh network. After I opened the issue in September and in front of the non-resolution, I bought a zzh and started using it. However, after a while, when trying to pair some end device in a place where the switch was nearby, it did not transmit data to the coordinator and the pairing did not happen.

I had to go close to the zzh and then the pairing took place. Soon after, when I took the end device to the definitive location, the next switches did not route the data from this new end devices, causing it to run out of signal.

After this problem in zzh, I decided to go back to the conbee again, but I noticed the return of the problem of not sending the status change by physically pressing the button.

@Koenkk I intend to try zzh again, does this problem I narrate have a chance of being solved? The cc2531 for the sniff arrived yesterday, but incomplete unfortunately, I'm hoping to get a downloder cable that I decided to buy right here in my country.

I'm sorry to mention this zzh problem here, but I'm very tired of all this, and I'd like my network to stay 100%.

Jhonattan-Souza, 2020-12-16:
> @Jhonattan-Souza Was this switch at another zigbee gateway before? And have you ever reset it?

@ChrisHae Yes, in Samsung Smartthings, and it works perfectly, there is feedback from physical button without lag .

 Yes, I've reseted many many times to pair with Conbee. I also reseted to return device to smartthings to collect logs, now the switch is back to conbee again.

ChrisHae, 2020-12-16:
@Jhonattan-Souza Ok if reseting does not change anything then there are two ways that could solve the problem. As it is said the current ConBee FW does not forward messages send to EP11. Until this is fixed in FW you can try two things.
1. Using Zigbee2mqtt to Unbind the switch from EP 11 according this guide: https://www.zigbee2mqtt.io/information/binding.html
`zigbee2mqtt/bridge/unbind/[SOURCE_DEVICE_FRIENDLY_NAME]/11`
with payload `TARGET_DEVICE_FRIENDLY_NAME`
I am not sure if unbinding a specific EP works because in this guide only binding to a specific EP is described.
And then bind it to EP1
`zigbee2mqtt/bridge/bind/[SOURCE_DEVICE_FRIENDLY_NAME]/1`
with payload `TARGET_DEVICE_FRIENDLY_NAME`
2. If this does not work you can Change the second EP of the ConBee to EP 11. This can be done with the deCONZ Software and this means the EP F2 will be lost. That means that Green Power Devices will not work any longer.
in the deCONZ GUI click on Network Settings (the top middle button over the network map) and then click on the tab Enpoints.
There you can describe two endpoints. You have to change the second one to 0x0B. Copy the other fields(Profile ID, Device ID, Device Version, In Clusters and Out Clusters from the first one (0x0104, 0x0005, 0x1, 0x0019, 0x0500)

Jhonattan-Souza, 2020-12-16:
It seems that nor bind nor unbind is working in this case

I prefer to wait to be fixed in FW rather than lost the green power devices endpoint

@Koenkk and @ChrisHae, I really appreciate your help and effort, I'll be waiting for the new firmware version !

Logs:
[image]

juslex, 2020-12-16:
> It seems that nor bind nor unbind is working in this case
> 
> I prefer to wait to be fixed in FW rather than lost the green power devices endpoint
> 
> @Koenkk and @ChrisHae, I really appreciate your help and effort, I'll be waiting for the new firmware version !
> 
> Logs:
> [image]

the same thing happened here

ChrisHae, 2020-12-16:
Ok it was just guess. I see now this could not work because the EP11 must be the EP of the target not the source. I think this is not possible.

ChrisHae, 2020-12-16:
I got another Idea. in deCONZ GUI you can bind and unbind using the Bind Dropbox. Activate it in the menu under Panels/Bind Dropbox. You drag and drop the On/Off Cluster of the switch (grey text) to the source field. and the 0B endpoint of the coordinator to the Destination field. Then press unbind. That means you have to configure EP0B first in the Network Settings as I described before. Afterwards you can change it back to the Green Power Endpoint.
And then you can bind it to the Coordinator EP1.
I can't test it because I don't have the switch but it is worth a try (On the picture is another device in the source field as an example).
[image]

Jhonattan-Souza, 2020-12-16:
@ChrisHae Changed the endpoint in the configuration from 0x0B to 0x11 just to test, but , same result

Error: no entry

[image]

ChrisHae, 2020-12-17:
Then I think it just don't work to bind/unbind. I think there will be a fix in one of the next ConBeeFW or deCONZ version where you can simply add a third EP.

ChrisHae, 2020-12-17:
@Jhonattan-Souza Could you test something for me?
In deCONZ can you right click on the switch node and then "read binding table". And please activate the debug window in the menu under help/debug view. And there check ZDP.

It would be interesting to know how the response looks like.

Jhonattan-Souza, 2020-12-17:
[image]

@ChrisHae

juslex, 2021-01-12:
> @Jhonattan-Souza Could you test something for me?
> In deCONZ can you right click on the switch node and then "read binding table". And please activate the debug window in the menu under help/debug view. And there check ZDP.
> 
> It would be interesting to know how the response looks like.

Dear @ChrisHae,

any news about the firmware fix for this issue?

Jhonattan-Souza, 2021-02-14:
> any news about the firmware fix for this issue?

let's hope for the best

juslex, 2021-03-18:
let's hope for the best (2)

roumano, 2021-05-07:
New firwmare of conbee 2 is available since few days (29-03-2021) to be exact : 0x266b0700, but the issue remain.

ChrisHae, 2021-05-12:
Would anybody test the new firmware that should fix this issue?
https://www.dropbox.com/s/8fh73071knq3y2a/deCONZ_ConBeeII_0x266c0700.bin.GCF?dl=0
here is a manual on how to update the firmware of the ConBee2: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

And when testing just make sure to revert the second Endpoint to the GP Endpoint or just undo the changes if you made any in the Endpoint Config of deCONZ.

roumano, 2021-05-12:
Hi @ChrisHae ,
Tested just now 
```
# /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /tmp/deCONZ_ConBeeII_0x266c0700.bin.GCF
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2416690-if00 (ConBee II)
deCONZ firmware version 266B0700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
flashing 161494 bytes: |===============================|
verify: .
SUCCESS
Wait 10 seconds until application starts

```
But Sadly it's not fix the issue

[9 more replies]

closed 2021-12-22 by github-actions[bot]