Koenkk/zigbee2mqtt #4462: Legrand 412015 doesn't report power

jd766, 2020-09-25:
## What happened
I connected a Legrand 412015 DIN power consumption module, Bticino version, to z2m on Home Assistant, the device paired successfully but does not send any message, power consumption is never updated. 
**I can make it blink by publishing a message to its "/set" topic, but the communication is one-direction only.** 

If I try to publish { "power": ""} to its /get topic, I get this error:

> {"message":"Publish 'get' 'power' to 'Wattmetro_Casa' failed: 'Error: Read 0x0004740000899492/1 haElectricalMeasurement([\"activePower\"], {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 43098 - 1 - 180 - 2820 - 1 after 10000ms)'","meta":{"friendly_name":"Wattmetro_Casa"},"type":"zigbee_publish_error"}

Pairing and re-pairing, resetting doesn't change anything.

## Debug info
Zigbee2MQTT version: Hassio Add-on 1.14.4.2
Adapter hardware: slaesh’s CC2652RB stick

jd766, 2020-09-26:
I don't know how but this morning I just resetted it for the nth time and now it reports power correctly.
I'll write again if it stops working.

closed 2020-09-26 by jd766

PaoloTK, 2020-09-26:
Same issue here, I've reset the device many times but it always reports power as 0W.

jd766, 2020-09-26:
A bit off-topic here but I'm now having a similar issue with some of my 067775 smart plugs, again Bticino version. 
I can turn them off and on, make them blink, but they don't report the power consumption.

PaoloTK, 2020-12-15:
For anyone that stumbles upon this, I was able to fix the issue by inverting the direction of the sensor on the power line.

Cr4z33, 2021-07-06:
> 
> 
> For anyone that stumbles upon this, I was able to fix the issue by inverting the direction of the sensor on the power line.

I hope this will work for me! 🙏

I already changed two of these DIN modules and they both were reporting 0W (no matter how many times I reset them).

jd766, 2021-07-06:
After connecting them, did you try power cycling them (cutting off the
power), instead of resetting?

Cr4z33, 2021-07-07:
> 
> […](#)

Yes of course. 😉

Cr4z33, 2021-07-08:
> 
> 
> For anyone that stumbles upon this, I was able to fix the issue by inverting the direction of the sensor on the power line.

I tried inverting the direction, but the meter keeps reporting 0 W... 😡

Cr4z33, 2021-07-09:
@jd766 is the power consumption automatically updated?

I am not getting updates until I refresh it in the Z2M Web UI.

jd766, 2021-07-09:
@Cr4z33 
It updates every time it changes, both on Z2M and of course the MQTT broker. It's actually kinda spammy.

Cr4z33, 2021-07-09:
@jd766 pfff alright then.

Yet another issue (on my side)... 😖

benjR, 2022-03-30:
@Cr4z33 Any success on getting the power measurement updates ? I have a 412171, shows up correctly but never updates itself (it's working when pressing the button though).
Thanks!

benjR, 2022-04-02:
@Cr4z33 haha, I would need a bit of luck then, I recreated several times already (for other purposes) but it never pushed power, I could trigger something from HA but the power get request is not exposed to HA :(

benjR, 2022-04-03:
So I ended up setting up an automation directly in HA publishing a mqtt message to pull power every x minute

pirracas77, 2022-04-13:
@jd766 do you have also this issue #12152 ?

rainbowbreeze, 2022-05-27:
@benjR could you please provide the code for your automation?

I'm in the same situation as you were, and I want to measure how much energy my houses consumes. I can [convert](https://www.home-assistant.io/docs/energy/faq/#creating-an-energy-sensor-out-of-a-power-sensor) the power provided by 412015 into energy consumption thanks to a [Riemann sum integral integration](https://www.home-assistant.io/integrations/integration#energy), but I suppose I need to manually launch the update of the power value exposed by 412015, as it doesn't update by its own.

benjR, 2022-05-27:
@rainbowbreeze I run this every 5 minutes or so through an automation on HA:
```
service: mqtt.publish
data:
  topic: zigbee2mqtt/0x000474000083cxxx/get
  payload: '{"power":""}'
```
Interested to see how you turn this into energy consumption!

rainbowbreeze, 2022-05-27:
Here what I've done:

My Legrand 412015 is configured as `power_meter_001` in z2m, and expose in HA a `sensor.power_meter_001_power`. 
It has `unit_of_measurement` = W, and `device_class` = power

Following [Creating an Energy Sensor out of a Power Sensor](https://www.home-assistant.io/docs/energy/faq/#creating-an-energy-sensor-out-of-a-power-sensor) guidelines, I created a new energy sensor:
```
sensor:
  - platform: integration
    source: sensor.power_meter_001_power
    name: house_total_energy
    unique_id: house_total_energy
    unit_prefix: k
    unit_time: h
    round: 2
```

This sensor doesn't update if power doesn't update, and we all know 412015 doesn't publish power updates, but you've to force them. So, I created an automation to force read the power measurement from `power_meter_001`:
```
alias: Energy - Force read total house power
description: ''
trigger:
  - platform: time_pattern
    seconds: /30
condition: []
action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/power_meter_001/get
      payload: '{ "power": "" }'
mode: single
```
This new sensor has `device_class` = energy, and `unit_of_measurement` = kWh
30 secs because of what's written in the aforementioned guidelines. But I think also 1 minute is OK, unless you've important spikes in your energy grid you want to measure. For example, a water boiler that works only for 10-20 secs, etc.

Icing on the cake: I added `sensor.house_total_energy` in the HA Energy dashboard, and I can now control how much energy my house needs

Thanks everyone for the many suggestions I captured here and there in order to arrive to this solution!