Koenkk/zigbee2mqtt #4464: Schlage BE468 != Schlage BE468GBAK

orditeck, 2020-09-26:
Hi,

I bought this lock thinking that because its model number is beginning with `BE468` it'd work but no. It locks/unlocks but the status is not updated. It "works" with the Optimistic mode but that's not what I'm looking for.

I'd be totally willing to help make it work with some directions :) 

## Information about the device + link
Model number BE468GBAK
Link: https://www.amazon.com/gp/product/B07H41WCZJ

I'm getting this error if it can be of any help:
```
Zigbee2MQTT:info  2020-09-25 21:05:29: Configuring '0x90fd9ffffe64431d'
Zigbee2MQTT:error 2020-09-25 21:05:29: Failed to configure '0x90fd9ffffe64431d', attempt 1 (TypeError: Cannot read property 'bind' of undefined
    at bind (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman-converters/devices.js:44:24)
    at Object.configure (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman-converters/devices.js:11694:19)
    at Configure.configure (/zigbee2mqtt-1.14.4/lib/extension/configure.js:143:45)
    at Configure.onZigbeeStarted (/zigbee2mqtt-1.14.4/lib/extension/configure.js:111:28)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.4/lib/controller.js:365:44))
```

Koenkk, 2020-09-26:
Can you share the `data/database.db` entry of this device?

orditeck, 2020-09-26:
I'm trying to find indications that could help me doing that. Is there any link you could send me?

When opening a new issue, it says to look at IMPORTANT: Before submitting a request, first follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html but there's no reference to `database.db`

I'm using HA in a VM in Proxmox, so using "Hass.io" (VM host running HA in docker)

Koenkk, 2020-09-26:
Are you able to access the `configuration.yaml` file of Zigbee2MQTT? The `database.db` file is next to that file in a folder called `data`.

orditeck, 2020-09-26:
Figured it out. Here's the `database.db` part about the lock:

```
{
  "id": 4,
  "type": "EndDevice",
  "ieeeAddr": "0x90fd9ffffe64431d",
  "nwkAddr": 62664,
  "manufId": 4662,
  "manufName": "Schlage",
  "powerSource": "Battery",
  "modelId": "BE468",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 10,
      "inClusterList": [
        0,
        1,
        3,
        9,
        32,
        257,
        2821,
        64512
      ],
      "outClusterList": [
        10,
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "BE468",
            "manufacturerName": "Schlage",
            "powerSource": 3,
            "zclVersion": 2,
            "appVersion": 0,
            "stackVersion": 6,
            "hwVersion": 65
          }
        }
      },
      "binds": []
    }
  },
  "appVersion": 0,
  "stackVersion": 6,
  "hwVersion": 65,
  "zclVersion": 2,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1601076860398
}
```

orditeck, 2020-09-26:
I'm gonna try to change `https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L12405` to use endpoint `1` instead of `2`.

orditeck, 2020-09-26:
Yay! Works #1 when changing `getEndpoint(2)` to `getEndpoint(1)` (as we can see in the database.db, the endpoint is `1`)

So the following works as expected!
```js
    // Schlage
    {
        zigbeeModel: ['BE468'],
        model: 'BE468',
        vendor: 'Schlage',
        description: 'Connect smart deadbolt',
        supports: 'lock/unlock, battery',
        fromZigbee: [fz.lock, fz.lock_operation_event, fz.battery],
        toZigbee: [tz.generic_lock],
        meta: {configureKey: 3},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await configureReporting.lockState(endpoint);
            await configureReporting.batteryPercentageRemaining(endpoint);
        },
    },
```

Koenkk, 2020-09-26:
In that case `const endpoint = device.endpoints[0];` may be a solution.

@jdorrance I see you added support for this device in https://github.com/Koenkk/zigbee-herdsman-converters/pull/1255 , could you share me your `data/database.db` to ensure it doesn't break on your side?

closed 2020-10-20 by orditeck