Koenkk/zigbee2mqtt #4465: Zgemma ZM25tq rollershutter

Dylantje, 2020-09-26:
Dear all..
https://zigbee.blakadder.com/Zemismart_ZM25TQ.html
I do have his one working.. Only can use 1 of this motor. The second one will not learn. Is there perhaps a option to repair or set the motor into pair mode? Or a manual add option?

My config:
`homeassistant: false
permit_join: true
log_level: error
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
  disable_led: false
  channel: 11
devices:
  '0x00158d0001a9e2ba':
    friendly_name: 'ZBX: Fietshok pir'
  '0x000d6ffffed65967':
    friendly_name: 'ZBI: Carportcam R'
  '0x00158d00010e4dac':
    friendly_name: 'ZBX: WC Pir'
  '0x00158d0001a4b86d':
    friendly_name: 'ZBXA: Woonkamer'
  '0x00158d00044457a5':
    friendly_name: 'ZBX:  33 BVTHL'
  '0x00158d0001b76f68':
    friendly_name: 'ZBX: Schakelaar over'
  '0x00158d000117b3fd':
    friendly_name: 'ZBX: Ryan Pir'
  '0x00158d000130d60b':
    friendly_name: 'ZBX: Dylan Pir'
  '0x00158d0001ddcd53':
    friendly_name: 'ZBXA: Zolder Overl Pir'
  '0x00158d0001a35d4e':
    friendly_name: 'ZBXA: 22'
  '0x00158d0004102c94':
    friendly_name: 'ZBX: 21 Over'
  '0x00158d000213872d':
    friendly_name: 'ZBXA: Vierkant 1'
  '0x00158d00010dcd46':
    friendly_name: 'ZBX: Rond 1'
  '0xec1bbdfffe47c996':
    friendly_name: 'ZB: Ikea repeater 1 [ Ryan ]'
  '0x00158d0001db35fb':
    friendly_name: 'ZBX: Carport Pir'
  '0x0017880100d7c02a':
    friendly_name: 'ZBHW: Garage R'
  '0x00158d0001f29be2':
    friendly_name: 'ZBX: Bijkeuken Pir'
  '0x00158d0001ddccef':
    friendly_name: 'ZBXA: Keuken Pir'
  '0x00158d00018373ad':
    friendly_name: 'ZBXA: Achterdeur'
  '0x00158d0001a64c71':
    friendly_name: 'ZBXA: Bijkeukendeur'
  '0x00158d0001a8d32e':
    friendly_name: 'ZBX: Gang Pir'
  '0x0017880100ed186c':
    friendly_name: 'ZBHW: Gang R'
  '0x0017880100dbcce7':
    friendly_name: 'ZBHW: Zolder R'
  '0x00178801063405a8':
    friendly_name: 'ZBHR1: Nieuw'
  '0x0017880100e56e0e':
    friendly_name: 'ZBHW: Over'
  '0x0017880100e612b6':
    friendly_name: 'ZBHK: Eettafel 1 R'
  '0x0017880100ffc053':
    friendly_name: 'ZBHK: Eettafel 2 R'
  '0x0017880100fedd0f':
    friendly_name: 'ZBHK: Zitkamer R'
  '0x001788010117afe5':
    friendly_name: 'ZBHG: Kamer TV R'
  '0x0017880100e8dd9d':
    friendly_name: 'ZBHKS: Kachel 1 R'
  '0x0017880100e84bd3':
    friendly_name: 'ZBHKS: Kachel 2 R'
  '0x00158d00014dc9a2':
    friendly_name: 'ZBX: Overloop Pir'
  '0x00158d000117b661':
    friendly_name: 'ZBX: Voordeur'
  '0x00158d000121b9c5':
    friendly_name: 'ZBX: Badkamer T&V'
  '0x00158d00015a9ce3':
    friendly_name: 'ZBX: Garage Pir'
  '0x00158d000149a348':
    friendly_name: 'ZBX: Garagedeur'
  '0x00158d00016da147':
    friendly_name: 'ZBX: Pir over'
  '0x34c69a000bec3df7':
    friendly_name: '0x34c69a000bec3df7[ weet niet? ]'
  '0x00158d00016569cf':
    friendly_name: 'ZBX: PIR over'
  '0x00158d0001537631':
    friendly_name: 'ZBX: Zolder Wastafel'
  '0x00124b00210b624c':
    friendly_name: 'Zgeldop: Douche L'
  '0x00124b00213acb58':
    friendly_name: 'Zgeldop: WC'
  '0x00124b00213b124d':
    friendly_name: 'Zgeldop: Meubel rechts'
  '0x00124b0021382f36':
    friendly_name: 'Zgeldop: Bad R'
  '0x00124b002117b04f':
    friendly_name: 'Zgeldop: Douche R'
  '0x00124b00213ac72f':
    friendly_name: 'Zgeldop: Bad M'
  '0x00124b002117c4a7':
    friendly_name: 'Zgeldop: Meubel Links'
  '0xbc33acfffe363dcc':
    friendly_name: Ryan Voorkant
groups:
  '1':
    friendly_name: 'ZB: Lampen'
    devices:
      - 0x0017880100e612b6/11
      - 0x0017880100ffc053/11
      - 0x0017880100e84bd3/11
      - 0x0017880100fedd0f/11
      - 0x0017880100dbcce7/11
      - 0x001788010117afe5/11
      - 0x0017880100ed186c/11
      - 0x0017880100d7c02a/11
      - 0x00124b0021382f36/11
      - 0x00124b00210b624c/11
      - 0x00124b002117b04f/11
      - 0x00124b00213b124d/11
      - 0x00124b00213acb58/11
  '2':
    friendly_name: Rolgordijnen
    devices:
      - 0xbc33acfffe363dcc/1
`
Everthing working fine. only 1 device...
[image]

levitatejay, 2020-09-28:
I only have 1 at the moment so I can't confirm this... but maybe you need to pair the motors to different channels on the remote first?

Dylantje, 2020-09-28:
I was also thinking to that direction. Only o remove the 1 that was working.. and now i cannot learn 1 again. Do you know a factory reset and how i get the motor in to pair mode for z2m? And in z2m to pair mode? Or new device? What is the difference?

nogueirasa, 2020-10-01:
@Dylantje , same problem for me.

@levitatejay , I tryed with the motors configured in different channels, but the same problem

nogueirasa, 2020-10-01:
Worked for 2 motors. Copied the device.js configuration twice:

    {
        zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja', 'u1rkty3'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_xuzcvlku'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_nkoabg8w'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
        ],
        model: 'TS0601_curtain',
        vendor: 'TuYa',
        description: 'Curtain motor',
        whiteLabel: [
            {vendor: 'Yushun', model: 'YS-MT750'},
            {vendor: 'Zemismart', model: 'ZM79E-DT'},
            {vendor: 'Binthen', model: 'BCM100D'},
            {vendor: 'Binthen', model: 'CV01A'},
            {vendor: 'Zemismart', model: 'M515EGB'},
            {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },

and then again:

    {
        zigbeeModel: ['mcdj3aq', 'mcdj3aq\u0000'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
        ],
        model: 'mcdj3aq',
        vendor: 'TuYa',
        description: 'Tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },

Dylantje, 2020-10-01:
Nice. When i am @ home i give it also a try... Please tell me how firt reset the motor? Then set the motor in pairing/learning mode? Only how do i do that?

Dylantje, 2020-10-01:
You use domoticz?

nogueirasa, 2020-10-01:
> You use domoticz?

Home Assistant

nogueirasa, 2020-10-01:
> Nice. When i am @ home i give it also a try... Please tell me how firt reset the motor? Then set the motor in pairing/learning mode? Only how do i do that?

Press the motor setup button 3 times very fast

Dylantje, 2020-10-01:
That is the reset option?
Or is this also the learning/pairing way?

Or after a reset the motor go automatically into learn option?

Dylantje, 2020-10-01:
Did update to last version of z2m 1.15
Only now i cannot learn any of my 3 motors. Also not the 1 i had working
so i think i do something wrong?
Please the steps i need to make to a factory reset
And then ho i learn the good way

Dylantje, 2020-10-01:
mmm Did learn 2 of 3.. [ fast push the button ] 
Ony not vissable..in z2m, so perhaps just the 2 that are not in the db?
[image]

Dylantje, 2020-11-01:
No other options?  Only wait until new version ?

Dylantje, 2020-12-04:
Hope this topic wil not be closed...
Ordered a new stick so perhaps then is this going to work.

Wait.. for the post.. very bussy here at the moment

closed 2021-02-13 by github-actions[bot]