Koenkk/zigbee2mqtt #4466: [RFC] Exposing device capabilities

Koenkk, 2020-09-26:
Currently there is no way for clients to determine the capabilities of a device. With this change I would like to introduce this. Please let me know what you think!

## Benefits
- Avoid having to maintain a manual mapping for integrations e.g.:
  - [homebridge-z2m](https://github.com/itavero/homebridge-z2m) @itavero
  - [domoticz-zigbee2mqtt-plugin](https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin) @stas-demydiuk
  - [homeassistant.js](https://github.com/Koenkk/zigbee2mqtt/blob/0cd4607856b00aa50098f8960046ed8d99d56946/lib/extension/homeassistant.js#L837)
  - [ioBroker.zigbee](https://github.com/ioBroker/ioBroker.zigbee) @kirovilya
- Provide more info on what values to expect for a certain device on the [device specific pages](https://www.zigbee2mqtt.io/devices/DJT11LM.html)
- Improve controlling of devices through the frontend (as currently it does not know what it can send) (@nurikk)
- Get rid of non-uniform [`supports` attribute](https://github.com/Koenkk/zigbee-herdsman-converters/blob/7c6d8d63f068a53762e6364c188a2b0cb6bec443/devices.js#L457)

## Proposal
Expose the device capabilities through the [new api](https://github.com/Koenkk/zigbee2mqtt/issues/3281) on the `zigbee2mqtt/bridge/devices` topic, example:

```js
[
    {
        "ieee_address":"0x00158d00018255df",
        "type":"Router",
        "network_address":29159,
        "supported":true,
        "friendly_name":"my_plug",
        "endpoints":{"1":{"bindings":[],"clusters":{"input":["genOnOff","genBasic"],"output":[]}}},
        "definition":{
            "model":"ZNCZ02LM",
            "vendor":"Xiaomi",
            "description":"Mi power plug ZigBee",
            "supports":"on/off, power measurement"
            "exposes": [{.....}] // <------ HERE ----
        },
        "power_source":"Mains (single phase)",
        "date_code":"02-28-2017",
        "interviewing":false,
        "interview_completed":true
    }
]
```

The `exposes` attribute will be added directly in the device definition (`devices.js` file). Possible `exposes` entries are below. Note that I think this is the bare minimum we currently need (extending is always possible :smile:).

### Generic

#### Boolean

```js
{
    "type": "boolean",
    "property": "occupancy"
}
```

#### Numeric

```js
{
    "type": "numeric",
    "property": "temperature",
    "unit": "°C"
}
```

### Specific
For devices exposing multiple endpoints (e.g. switch with left and right) a property `endpoint` will be added. e.g. `"endpoint": "left"`.

#### Action

Indicates the device can produce an action (e.g. button click).

```js
{
    "type": "action"
}
```

#### Light
```js
{
    "type": "light",
    "features": ["state", "brightness", "color_temp", "xy"],
}
```

#### Switch

```js
{
    "type": "switch",
}
```

#### Cover
```js
{
    "type": "cover",
    "features": ["position", "tilt"],
}
```

#### Lock
```js
{
    "type": "lock"
}
```

#### Fan
```js
{
    "type": "fan",
    "modes": ["off", "low", "medium", "high", "on", "auto", "smart"]
}
```

#### Climate
```js
{
    "type": "climate",
    "modes": ["off", "auto", "heat"],
    "minimum_temperature": 7,
    "maximum_temperature": 30,
    "temperature_step_size": 1
}
```

nurikk, 2020-09-26:
what about reading attributes from the device (something like power on state). Some devices can expose the same attribute for multiple endpoint (per endpoint configuration, like ZCL_CLUSTER_ID_GEN_ON_OFF_SWITCH_CONFIGfoe every button )

nurikk, 2020-09-26:
For boolean(or other) attribute isn't clear whether the user can modify this state or no

sjorge, 2020-09-26:
> For boolean(or other) attribute isn't clear whether the user can modify this state or no

Yes! Having a ‘writable’ indicator is a good idea.
Maybe with options range for int, or enum for int and str.

E.g.

color_mode = read, int, enum [0=Hue/Sat, 1=CIExy,2=ColorTemp]

kirovilya, 2020-09-27:
I have a positive attitude.
This partially overlaps with the classification of devices (as it is done here https://zigbee.blakadder.com/).
I also suggest putting this in devices.js as a list, since there are devices with different functionality. For example, distinguish light bulbs from sockets, switches or relays.

This all resembles the standardization of device capabilities, as is done in Apple HomeKit, Google Home, Amazon Alexa, Yandex Alice.
We declare what capabilities the device has, and the end systems themselves interpret how to display the work with this capability. If the device is specified as a "temperature sensor" or "socket with consumption measurement", then it is clear which attributes to display and which commands can be sent to the device.

But will we be able to describe all the capabilities of the device in this way? Basic - yes, but there are additional ones that are very different.
For example, it is possible to call selftest of a smoke detector or change its sensitivity, or switch the binding mode on the Opple remote control.
This should be described in more detail. And this possibility should have a name ("Sensor sensitivity", "Binding mode"), not just a type.

Koenkk, 2020-09-27:
> what about reading attributes from the device (something like power on state). Some devices can expose the same attribute for multiple endpoint (per endpoint configuration, like ZCL_CLUSTER_ID_GEN_ON_OFF_SWITCH_CONFIGfoe every button )

I don't understand what you mean here.

@kirovilya @sjorge @nurikk regarding the settable part, if we e.g. take the sensitivity of a sensor we can easily add it with:

```
{
    "type": "enum",
    "property": "sensitivity",
    "values": ["low", "medium", "high"],
    "direction": 2, // 0 = get only, 1 = set only, 2 = get and set
}
```

Selftest:

```
{
    "type": "enum",
    "property": "self_test",
    "values": [""],
    "direction": 1, // 0 = get only, 1 = set only, 2 = get and set
}
```

nurikk, 2020-09-27:
Can we have typescript definitions for new exposes properties?

kirovilya, 2020-09-28:
OK. I think I need to have an optional description of the property.
```
{
    "type": "enum",
    "property": "self_test",
    **"description": "Invoke self-test mode (send any value)",**
    "values": [""],
    "direction": 1, // 0 = get only, 1 = set only, 2 = get and set
}
```

Koenkk, 2020-09-28:
@nurikk typescript defs shouldn't be a problem

@kirovilya `description` field is no problem!

Since I believe we agree on the general idea and structure, I will start implementing this for lights (since those represent the biggest share of supported devices).

sjorge, 2020-09-29:
What is the plan for groups? For uniform groups we can probably publish the info too, not sure for mixed groups... only the common ones?

E.g. 4 bulbs that are ws and one that is cws, we could still publish the ws part as all 5 support it?

But probably something for after we have devices exposed

Koenkk, 2020-09-29:
@sjorge lets do that after this indeed, I think we can also add discovery then.

Koenkk, 2020-10-01:
`exposes` is now available for all lights (+- 50% of the supported devices). Docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#exposes

nurikk, 2020-10-01:
Sneak peek :)
[image]

nurikk, 2020-10-01:
I would like to see implementation of sensors and "set converters" (ex: power on state, how to read/set values)

nurikk, 2020-10-02:
First try, already available in `dev`
[image]

nurikk, 2020-10-02:
Currently implemented color_xy, color_temp, state, brightness
Kindly share device states with other color_XX features (I don't have anything but color_xy)

Koenkk, 2020-10-02:
Also added for switch devices!

sjorge, 2020-10-02:
@Koenkk what about lights like `Innr RB 250 C` which takes
- color_temp
- color_xy
- hue + saturation

(and it properly switches color_mode!)

Currently it is exported as color_xy + color_temp I believe.

sjorge, 2020-10-02:
> Also added for switch devices!

Looks like the Ubisys S1 and S2 are not exposed? Or will you do them in a later set, as they also support power metering (and well voltage, but the later is not hooked up but the cluster is there)

Koenkk, 2020-10-02:
@sjorge feel free to also add `.withColorHs` to that, about S1/S2 a device is either mapped in homeassitant.js OR has `exposes`. Since it also has the sensor which are not supported with `exposes` yet I didn't migrate them.

sjorge, 2020-10-02:
Will open a PR against converters, I also discovered the E27 variant was missing the meta to enable Hue/Sat support, I just verified it works, will include that too.

nurikk, 2020-10-03:
I've implemented switch support, it's quite dull component.
We need to test lights and switches with different endpoints.

[image]

itavero, 2020-10-05:
Sorry I didn't get back to you sooner. The proposal looks interesting and it will definitely help to "speed" up the discovery process in homebridge-z2m.

One thing I'm really interested in would be remote controls.
If we could define a good definition for that, that would help a lot with itavero/homebridge-z2m#2.
With the current release of zigbee2mqtt, I don't see a good way to convert a remote control to a HomeKit entity without manual configuration, because I don't know all the possible values and how they relate to each other upfront.

Some background information: in HomeKit (and therefor in Homebridge) you have [Stateless Programmable Switches](https://developers.homebridge.io/#/service/StatelessProgrammableSwitch) and [Stateful Programmable Switches](https://developers.homebridge.io/#/service/StatefulProgrammableSwitch).
I think for homebridge-z2m, I will (mostly) stick to the stateless service, as the state will probably be exposed as an "On/Off" characteristic on a Switch service.
Anyway, for both of them you must define a [Programmable Switch Event](https://developers.homebridge.io/#/characteristic/ProgrammableSwitchEvent) characteristic.
HomeKit "assumes" that every button has the following events:
* Single Press
* Double Press
* Long Press

Ofcourse, these actions don't always exists on the actual devices compatible with z2m (and the long press sometimes even seems to send the same command for different physical buttons on a remote), so I reckon I either need to try and detect these things in the homebridge plugin or not support them, if the actual device does not support them.
But, if the actions do exist, they are different textual strings in the `click` property (for instance).

So for homebridge-z2m, what I would like to get out of the `exposes` information is which buttons exist on a remote control and what the different values are for different kind of events/actions for each button.

Koenkk, 2020-10-05:
@itavero no problem, for the action we can publish something like:

```js
{
    "type": "action",
     "actions": ["single", "double", "triple", '"many"],
}
```

Would that solve your issue?

itavero, 2020-10-07:
@Koenkk I think that's a bit limited, at least for my plugin to be able to automatically configure it. I'll capture some of the values currently transmitted by the different remotes I have and see if I can make a sensible/flexible definition based on those.

Koenkk, 2020-10-07:
`numeric` and `boolean` exposes are now also available (https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#numeric). 634 out of 978 supported devices have `exposes` now.

sjorge, 2020-10-07:
Can we mix those, switch + numeric for power. e.g. Ubisys S1 ?

nurikk, 2020-10-07:
> Can we mix those, switch + numeric for power. e.g. Ubisys S1 ?

Yes, this is totally fine
[image]

Koenkk, 2020-10-12:
Added the `cover` exposes type. Currently 705 out of 981 devices have `exposes`.

bodiroga, 2020-10-12:
Hi @Koenkk!

One question: where do you get those "705 out of 981" numbers from? :confused: I have had a look at the latest version of the [devices.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js) file and I haven't been able to confirm that proportion (71.87% per your numbers).

Do you want us to create small pull requests adding the "exposes" field to our devices or you have an automatic tool to add them? Perhaps I haven't checked the correct file, sorry :wink: 

Anyway, many many thanks for your awesome work, the zigbee2mqtt project is amazing! :+1: 

Best regards,

Aitor

Koenkk, 2020-10-13:
@bodiroga a lot of devices (almost all lights, +- 500) get it via e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/b40bd9ba117b40119a5dc609ebd6a11c25bad39b/devices.js#L263 , https://github.com/Koenkk/zigbee-herdsman-converters/blob/b40bd9ba117b40119a5dc609ebd6a11c25bad39b/devices.js#L1275

I have an automatic tool to do this: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/scripts/transformExposes.js (so no PRs required from users).

Hacker-CB, 2020-10-13:
@Koenkk 
How about to add  somethink like this:

```javascript
const PropertyAccessMode = {
    READ_ONLY: 1,
    WRITE_ONLY: 2,
    READ_WRITE: 3,
};

class Base {
    withEndpoint(endpointName, accessMode = PropertyAccessMode.READ_WRITE) {
        this.endpoint = endpointName;
        this.accesMode = accessMode;
        return this;
    }
}
```

But better to remove default value, like this:

```javascript
class Base {
    withEndpoint(endpointName, accessMode) {
        this.endpoint = endpointName;
        this.accesMode = accessMode;
        return this;
    }
}
```

Also I think it will be great to make some standartization for speicifc devices, like thermostats, which can have weekly schedule for example.

Koenkk, 2020-10-13:
@Hacker-CB I think access mode only makes sense for the generic types (currently `numeric`, `boolean`) and not for specific types.

For thermostat we can do something like `{"type":"thermostat","features":["weekly_schedule"]}`, for this to work the `weekly_schedule` format has to be the same for all thermostats, otherwise the client does not know what to send.

bodiroga, 2020-10-13:
> @bodiroga a lot of devices (almost all lights, +- 500) get it via e.g. https://github.com/Koenkk/zigbee-herdsman-converters/blob/b40bd9ba117b40119a5dc609ebd6a11c25bad39b/devices.js#L263 , [...]

Great, thanks for the explanation @Koenkk!

What about something like [WSDCGQ01LM](https://github.com/Koenkk/zigbee-herdsman-converters/blob/b40bd9ba117b40119a5dc609ebd6a11c25bad39b/devices.js#L851)? That is the only device I have tried in my test environment and it seems that it doesn't support the "exposes" field, bad luck :rofl: Do you mind if I create a PR for it? :+1: 

Thank you very much for your help!

Hacker-CB, 2020-10-13:
> For thermostat we can do something like `{"type":"thermostat","features":["weekly_schedule"]}`, for this to work the `weekly_schedule` format has to be the same for all thermostats, otherwise the client does not know what to send.

@Koenkk you are right. In this case I suggest for thermostats/fans something like this:

```javascript
const Dict extends Base {
    constructor(property, values, accessMode = PropertyAccessMode.READ_WRITE) {
        super();
        this.type = 'dict';
        this.property = property;
        this.values = values;
    }
}

class Climate extends Base {
    constructor(climateMode){
        super();
        this.type = 'climate';
        this.climateNode = climateMode;
        this.features = [];
    }

    withSystemModes(systemModes){
        this.features.push(Dict('system_mode', systemModes));
        return this;
    }

    withPresets(presets){
        this.features.push(Dict('preset',presets, PropertyAccessMode.READ_ONLY));
        return this;
    }

    withRoomTemperature(){
        this.features.push(Numeric('room_temperature' PropertyAccessMode.READ_ONLY,));
        return this;
    }

    withFloorTemperature(){
        this.features.push(Numeric('floor_temperature' PropertyAccessMode.READ_ONLY,));
        return this;
    }

    withFan(fanModes){
        this.features.push(Dict('fan',fanModes));
        return this;
    }

    withChildrenLock(){
        this.features.push('children_lock');
        return this;
    }

    withWeeklySchedule(scheduleMode, maxTransitions) {
        // HERE will be more complicated logic
    }
}
```

I can try to implement this if you agree with base logic.

Koenkk, 2020-10-13:
@Hacker-CB yes, looks good! But maybe `features` should be an object, this would allow for more extensions in the future. e.g.

```js
{
  "type": "climate",
  "features": {
    "children_lock": {"type": "numeric", "direction": 0},
    "floor_temperature": {"type": "numeric", "direction": 0, "max": 30, "min": 7}
  }
}
```

If you agree, then we should also do the same for `lights` and `cover`.

@bodiroga I will migrate some Xiaomi devices to `exposes` soon.

Hacker-CB, 2020-10-13:
@Koenkk , You are right. It will be better if `features` should be object.
For the `direction` - I think it will be better to name it `access` or `accessMode`, or may be separate properties `readable` and `writable`. For my mind `direction` is not clear.

I will try to implement thermostat based on `ETOP HT-08/HT-10` after you finish refactoring for `lights` and `cover`.

Hacker-CB, 2020-10-13:
One more comment about `min` and `max`, it is possible to change `min/max` boundaries for some thermostats.
Also `min/max` will work only during writing. It is useless when reading.

bodiroga, 2020-10-13:
> @bodiroga I will migrate some Xiaomi devices to `exposes` soon.

Perfect, thank you Koen!

Hacker-CB, 2020-10-13:
I forgot about devices with actions with params, like [heiman_ir_remote](https://github.com/Koenkk/zigbee-herdsman-converters/blob/d81ba2e5a1f70e0df752c5d795453cdeda3fa834/converters/fromZigbee.js#L2994):
```javascript
{
    action: 'learn',
    action_result: msg.data.result === 1 ? 'success' : 'error',
    action_key_code: msg.data.keyCode,
    action_id: msg.data.result === 1 ? msg.data.id : undefined,
};
```

I don't have idea how to describe such action with `exposes` now.

Hacker-CB, 2020-10-13:
For my mind its better to wrap action to object, but it can break a lot.
```json
{
  "action": {
     "type": "learn",
    "params": {...}
  }
}
```

Koenkk, 2020-10-14:
@Hacker-CB  currently @itavero is looking how we should define actions (https://github.com/Koenkk/zigbee2mqtt/issues/4466#issuecomment-704829177)

Hacker-CB, 2020-10-14:
> @Hacker-CB currently @itavero is looking how we should define actions ([#4466 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4466#issuecomment-704829177))

Ok, Iet's wait.

> For the `direction` - I think it will be better to name it `access` or `accessMode`, or may be separate properties `readable` and `writable`. For my mind `direction` is not clear.

@Koenkk what about this? For `boolean` it is very important for HA because it will be different domains. `readonly  - binary sensor`, `read/write  - switch`.

Koenkk, 2020-10-14:
@Hacker-CB this depends on the context right? We can do this in `homeassistant.js`.

sjorge, 2020-10-14:
> @Hacker-CB this depends on the context right? We can do this in `homeassistant.js`.

I think it's important to have the exposes to be as generic as possible as this should not be homeassistant specific.

That might mean some extra mappings in homeassistant.js but that should, I hope, not be a issue.

Koenkk, 2020-10-14:
I think the lights exposes should be changed to: 

```js
{
  "type": "light",
  "features": {
    "state": {"type": "enum", "values": ["ON", "OFF"], "access": 2},
    "brightness": {"type": "numeric", "min": 0, "max": 254, "access": 2},
    "identify": {"type": "enum", "values": ["blink", "channel_change"], "access": 1},
  }
}
```

I like the idea of @Hacker-CB that every  `features` element is one of the generic exposes type (`numeric`/`enum`/`boolean`, etc..)

sjorge, 2020-10-14:
Is there a reason why 'access' is numeric? Having it be 'r', 'w', or 'rw' seems easier to read as a human (which should hopefully also make it easier/less error prone to write if people add new devices)

Should we also split the values to a 'state' and an 'action' like seperation?

For device.state is is indeed 'ON' or 'OFF' but you can also use 'toggle' as input... but 'toggle' will never be the returned value?

kirovilya, 2020-10-14:
I will comment a little about the `state` ...
In the case of a light / socket, we need to map the values ​​corresponding to the `on` and `off`, and not just a list of available ones.
Which value from the list indicates that the lamp should be turned on? :))

kirovilya, 2020-10-14:
I think a predefined type (not logical, not digital, not string) is suitable here - for example `switch` (and this type already has fixed values ​​and their correspondences for state)

Koenkk, 2020-10-14:
@sjorge rw/r/w indeed makes much more sense, toggle is indeed a tricky one, not sure how to deal with this case yet

@kirovilya can you make  a proposal by changing the example of: https://github.com/Koenkk/zigbee2mqtt/issues/4466#issuecomment-708242173

kirovilya, 2020-10-14:
```
{
  "type": "light",
  "features": {
    "state": {"type": "switch", "access": 2},
    "brightness": {"type": "numeric", "min": 0, "max": 254, "access": 2},
    "identify": {"type": "enum", "values": ["blink", "channel_change"], "access": 1},
  }
}
```
and somewhere in our code we know what the `switch` type is: ON - is for turn on, OFF - is for turn off, TOGGLE - for change state.

sjorge, 2020-10-14:
Wouldn't `switch` be an abstraction that needs to go in homeassistant.js?

nurikk, 2020-10-14:
for switch type we can do something like
```json
 {
     "name": "state",
     "type": "binary",
     "access": 2,
     "value_true": "LOCK",
     "value_false": "UNLOCK",
     "toggle": "TOGGLE"
 }
```

sjorge, 2020-10-14:
> for switch type we can do something like
> 
> ```json
>  {
>      "name": "state",
>      "type": "binary",
>      "access": 2,
>      "value_true": "LOCK",
>      "value_false": "UNLOCK",
>      "toggle": "TOGGLE"
>  }
> ```

Maybe
```
 {
     "name": "state",
     "type": "binary",
     "access": 2,
     "value_true": "LOCK",
     "value_false": "UNLOCK",
     "value_input": ["LOCK", "UNLOCK", "TOGGLE"]
 }
```

If memory serves me right a bulb takes ON, OFF, toggle, alarm, ...

Hacker-CB, 2020-10-14:
> this depends on the context right? We can do this in `homeassistant.js`.

@Koenkk of cause we will do this in `homeassistant.js`.

> I think it's important to have the exposes to be as generic as possible as this should not be homeassistant specific.

@sjorge I agree with you. I just told that `access` or `direction` is very important.  

> If memory serves me right a bulb takes ON, OFF, toggle, alarm, ...

@sjorge I think better to have only `ON/OFF` values. Toggle can be implemented on the client-side of the MQTT API.

> Is there a reason why 'access' is numeric? Having it be 'r', 'w', or 'rw' seems easier to read as a human (which should hopefully also make it easier/less error prone to write if people add new devices)

Looks nice.

```javascript
const PropertyAccessMode = {
    R: 'r',
    W: 'w',
    RW: 'rw',
};
```

sjorge, 2020-10-14:
> @sjorge I think better to have only `ON/OFF` values. Toggle can be implemented on the client-side of the MQTT AP

toggle and alarm are already implemented though and are valid zigbee payloads to send, not all bulbs respond the same to alarm either for example but in general they all flash when send.

Hacker-CB, 2020-10-14:
> toggle and alarm are already implemented though and are valid zigbee payloads to send, not all bulbs respond the same to alarm either for example but in general they all flash when send.
@sjorge as @Koenkk already told, it's important to have the exposes to be as generic as possible and all lights should have one API in this case. So, for my opinion binary switch should have only `ON/OFF` exposes.

sjorge, 2020-10-14:
> toggle and alarm are already implemented though and are valid zigbee payloads to send, not all bulbs respond the same to alarm either for example but in general they all flash when send.

I don't see a problem in having a 'value_input' as in https://github.com/Koenkk/zigbee2mqtt/issues/4466#issuecomment-708261352, you can just ignore it and use value_true and value_false for your implementation.

But it does provide hits about other inputs you can give it. This for example would easily allow say mapping of an identify function for bulbs if they support alarm, if not you can still create a fallback logic if needed.

But this doesn't take away from the basic binary state with a true/false value as defined.

nurikk, 2020-10-14:
>If memory serves me right a bulb takes ON, OFF, toggle, alarm, ...

Btw things like `alarm` and `toggle` can be exposed as separate features

```json
{
    "name": "state",
    "type": "enum",
    "values": ["toggle"]
}
```

Koenkk, 2020-10-14:
Instead of `value_true`/`value_false` I prefer `value_on`/`value_off`, i feel it's more generic I also don't see a problem with `value_input` although I would prefer to name it `values` instead, a binary expose will only have 2 values anyway (which you can discover through `value_on`/`value_off`. Combining everything together, we currently have: 

```js
{
    "type": "light",
    "features": [
        {"name": "state", "type": "binary", "value_on": "ON", "value_off": "OFF", "values": ["ON", "OFF", "TOGGLE"], "access": "rw"},
        {"name": "brightness", "type": "numeric", "min": 0, "max": 254, "access": "rw"},
        {"name": "identify", "type": "enum", "values": ["blink", "channel_change"], "access": "w"},
    ]
}
```

nurikk, 2020-10-14:
> Instead of `value_true`/`value_false` I prefer `value_on`/`value_off`, i feel it's more generic I also don't see a problem with `value_input` although I would prefer to name it `values` instead, a [...]

I really don't like the `values` in binary, if it's binary it only should have two values described in `value_on`/`value_off`, adding `TOGGLE` only adds inconsistency, if you need something more complex then use `enum`

[84 more replies]

closed 2021-01-01 by Koenkk