Koenkk/zigbee2mqtt #4468: Support for "Lonsonho Tuya ZigBee Wireless Smart Scene Switch 4 Gang" needed

Spirituss, 2020-09-26:
## Information about the device + link
Lonsonho Tuya ZigBee Wireless Smart Scene Switch 4 Gang
[image]

[image]

[image]

## data/database.db entry of the device
```
{"id":70,"type":"EndDevice","ieeeAddr":"0xbc.....","nwkAddr":...,"manufId":4098,"manufName":"_TZ3000_vp6clf9d","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,10,1,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"^q�&\u0012","modelId":"TS0044","manufacturerName":"_TZ3000_vp6clf9d","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[]},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[]}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1600948923006}
```

lcse66, 2020-09-28:
Hi,
I have one of this.

It works with these changes:

folder /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters
```
--- devices_orig.js     2020-09-28 19:16:53.095887508 +0200
+++ devices.js  2020-09-28 19:16:19.846152157 +0200
@@ -1231,6 +1231,18 @@
         toZigbee: [],
     },
     {
+        zigbeeModel: ['TS0044'],
+        model: 'TS0044',
+        vendor: 'TuYa',
+        description: 'Wireless switch with 4 buttons',
+        whiteLabel: [
+            {vendor: 'Lonsonho', model: 'TS0044'},
+        ],
+        supports: 'action',
+        fromZigbee: [fz.tuya_on_off_action, fz.battery],
+        toZigbee: [],
+    },
+    {
         zigbeeModel: ['TS0001'],
         model: 'TS0001',
         vendor: 'TuYa',
```
folder /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters
```
--- fromZigbee_orig.js  2020-09-28 19:36:08.504529036 +0200
+++ fromZigbee.js       2020-09-27 15:53:23.344632355 +0200
@@ -3521,6 +3521,7 @@
             let buttonMapping = null;
             if (model.model === 'TS0042') buttonMapping = {1: 'left', 2: 'right'};
             if (model.model === 'TS0043') buttonMapping = {1: 'right', 2: 'middle', 3: 'left'};
+            if (model.model === 'TS0044') buttonMapping = {1: 'button_1', 2: 'button_2', 3: 'button_3', 4: 'button_4'};
             const button = buttonMapping ? `${buttonMapping[msg.endpoint.ID]}_` : '';
             return {action: `${button}${clickMapping[msg.data[3]]}`};
         },
```

Koenkk, 2020-09-28:
@lcse66 thanks, instead of `button_1`, can you change it to `left_top`, `right_bottom`, etc.? Then I can add support for it out of the box.

lcse66, 2020-09-28:
```
--- fromZigbee_orig.js  2020-09-28 19:36:08.504529036 +0200
+++ fromZigbee.js       2020-09-27 15:53:23.344632355 +0200
@@ -3521,6 +3521,7 @@
             let buttonMapping = null;
             if (model.model === 'TS0042') buttonMapping = {1: 'left', 2: 'right'};
             if (model.model === 'TS0043') buttonMapping = {1: 'right', 2: 'middle', 3: 'left'};
+            if (model.model === 'TS0044') buttonMapping = {1: 'left_bottom', 2: 'right_bottom', 3: 'right_top', 4: 'left_top'};
             const button = buttonMapping ? `${buttonMapping[msg.endpoint.ID]}_` : '';
             return {action: `${button}${clickMapping[msg.data[3]]}`};
         },
```

Koenkk, 2020-09-29:
Thanks, device will be supported in next Zigbee2mqtt release on 1 October.

Spirituss, 2020-10-04:
I upgraded z2m to 115 and this devices still doesn’t work. I still can see device interview and announcement but there is no device in the device list.

Koenkk, 2020-10-05:
@Spirituss can you provide me your Zigbee2MQTT startup log?

Spirituss, 2020-10-05:
@Koenkk I cut lines with other devices.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5329640/log.txt)

Koenkk, 2020-10-06:
@Spirituss did you set `zigbee_shepherd_devices: true` in your configuration? (this should be `false`).

Spirituss, 2020-10-06:
@Koenkk I didn't set this option at all, should I do?

VoyteckPL, 2020-11-18:
I just connected it and reaction to press is sometimes over 3-4 seconds. Why is that? Im on HA and Z2M 1.16.1.

lcse66, 2020-11-25:
> I just connected it and reaction to press is sometimes over 3-4 seconds. 
It happens to me too. This delay is already visible on the serial device (/dev/ttyACM0, USB dongle based on CC2531, raspberry PI 3 B+), even with good link quality.

VoyteckPL, 2020-11-27:
So it's software problem as it happends on conbee 2 and cc2531.

VoyteckPL, 2020-11-30:
@Koenkk can we have this ticket reopened for futher investigation of latency?

VoyteckPL, 2020-11-30:
Thanks, how can I investigate latency? I have other buttons (Aqara for example) and didn't experience big latency.

Koenkk, 2020-11-30:
You can check by sniffing (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) , check how much time there is between clicking and seeing the message in wireshark.

VoyteckPL, 2020-11-30:
That looks complicated... I have Conbee 2, Hassos 5.5 on RPI 4. Is there any simpler way?

Koenkk, 2020-11-30:
I don't think (assuming your Conbee 2 is already connected through a USB extension cable I don't know what could cause this)

jeroenkl, 2020-11-30:
I've got a ZZH! with TS0044 and not experiencing any delay.

barmazu, 2020-12-01:
Actually I came here because of the matter you are now discussing. I've got ZZH! and I'm seeing quite delays with TS0044. This happens when I try to multiple actions fast - ex. do a "double" click after doing a "single" click (or vice versa) - the second and any other subsequent action will be delayed, not always but for most cases. If I run single action or allow device to "cool down" between actions for 2-3s it works w/o any delay. So this looks like TS0044 hardware fault rather than anything else?

VoyteckPL, 2020-12-02:
> Actually I came here because of the matter you are now discussing. I've got ZZH! and I'm seeing quite delays with TS0044. This happens when I try to multiple actions fast - ex. do a "double" click [...]

I have exactly the same experience.

jeroenkl, 2020-12-02:
I checked again and I do not have any delay. Clicking, single, double, hold subsequential (so many actions after in a row).
I ordered a second TS0044 and see if it is the same (or not)...
Or you running the latest ZZH! Koenkk firmware? Healty ZIgbee network?

VoyteckPL, 2020-12-02:
his is weird. I'm on conbee 2, RPI4 4GB, SSD with USB 3.0 support. Small zigbee network (10 nodes) signal 100%. I get this delays too.

barmazu, 2020-12-02:
> I checked again and I do not have any delay. Clicking, single, double, hold subsequential (so many actions in a row).
> I ordered a second TS0044 and see if it is the same (or not)...
> Are you running the latest ZZH! Koenkk firmware? Healty ZIgbee network?
Except TS0044 I have no issues within the network, all other zigbee buttons I have (apart TS0044 I've got 4 x Mi Wireless Zigbee and one Ikea Dimmer/Switch) work flawlessly, so the rest of devices - bulbs, temperature sensors, door/window sensors, wall switches, etc, ~35 devices in total. No drop outs, no delays. ZZH! is running 20200925 Koenkk's  Z-Stack firmware - rock solid, no plans for update at the moment.

prestonsmartuk, 2021-01-05:
Can someone help me please.  Within the exposes of Zigbee2MQTT WebUI, it discovered it, recognised it was Tuya brand etc but it's showing this:
battery
Remaining battery in % N/A%
action
Triggered action (e.g. a button click) N/A

I can't use any of the buttons as they aren't showing up as useable in Home Assistant.  
Can someone help me fix please?

Rincovanderbeek, 2021-01-20:
I have exactly the same, only the power status is visible as integration. No switches

jeroenkl, 2021-01-21:
very strange, it is working fine on my site. No delay's (got 2 TS0044) and all the button's are visible.
I'm using the latest koenk version with Domoticz

Spirituss, 2021-01-21:
For me it works too, but regarding delays - it takes place. Most of the time there is no delay, but sometimes I have significant (about 5 sec!!) delay for it.

Rincovanderbeek, 2021-01-21:
Strange indeed; I use HA, latest version, zigbee integration and a Ziig-A-Zag USB flashed withCC26X2R1_20201026.
I tried it a couple of times, no succes.
Other devices are binding correctly.

Rincovanderbeek, 2021-01-25:
Any idea why it only detects the powerstate?
Does it help when I list the device signing?

{
  "node_descriptor": "NodeDescriptor(byte1=2, byte2=64, mac_capability_flags=128, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006",
        "0x000a"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    },
    "4": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0001",
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZ3000_vp6clf9d",
  "model": "TS0044",
  "class": "zhaquirks.tuya.ts0044.TuyaSmartRemote0044"
}

barmazu, 2021-01-31:
Finally I was able to run Zigbee sniff, and this TS0044 is a garbage, at least it looks so when comparing how Xiaomi switch/button (WXKG01LM)

After pressing a button on TS0044 3 zbee ZCL payload packets are sent:  First one  - a battery state attribute ("remaining battery power in %"). The second actual action payload - and action is performed INSTANTLY (eg light goes off). The weird thing is Wireshark is showing ZCL payload  as Malformed OnOff command, but nonetheless - it worked. And third zbee packet is shown around one second after the second one (this is the with empty action payload). In between there are Data Request packets sent by TS0044 where most major delays are seen (2x0.8s). What data is being requested? I have no idea. Now the tricky part: so imagine, if I press a button once again but somewhere in between, before 3rd zbee packet is processed and sent out by the device you will get a lag - at least with my device. Depending on how fast I have pressed the button again you will see higher or lower lag. So this second press will be processed only after first press of the button is fully completed (ie all 3 packets are sent out). How much does take to complete a "press a button action"? Around 2 - 2,5s. So this cool down thing I mentioned before make sense now (I mean it doesn't but you get the point). This is 100% reproducible, no matter what: how close to coordinator, sender receiver distance, firmware version (same for latest one as well as previous one). 

For mentioned Xiaomi it takes ~0.2-0.3s process and send action payload. No Data Requests packets are visible.
Under a cover I found a sticker which says: ESW-0ZAA-EU HRBG30LSE003C6
[image]

Rincovanderbeek, 2021-01-31:
I just found out that the actions per button must be defined as automations.
So not as integrations. Not sure if that is normal but that is how it works with my device.

dummys, 2021-02-04:
Hi, I have the same switch but I was not able to pair it with z2m. Are you able to pair it ? I opened the device and the MCU is a TUYA TYZS3 chip inside which should do Zigbee.

Sjworks, 2021-02-14:
I'm having the same problem.
By the way, I found out that this button works well without any delay when linked with Smartthings Hub(with custom DTH https://github.com/YooSangBeom/SangBoyST/blob/master/devicetypes/sangboy/zemismart-button.src/zemismart-button.groovy ), and I tested a few things.

I tried sniffing packets respectively when connected to Zigbee2Mqtt and Smartthings.
All of the following situations are packets when two buttons are pressed at the same time.

With Smartthing Hub (filter by `zbee_newk`)
[image]

With Zigbee 2 Mqtt (filter by `zbee_newk`)
[image]

I was able to see the `ZCL: Default Response` packet only in the Smartthings environment that is running normally,
In the `Z2M` environment, it seemed as if transmission failed and retryed several times.
I am not familiar with the packet structure of Zigbee, but there seems to be a delay due to a function that is not implemented.

[zigbee sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/5977560/zigbee.sniff.zip)

Is it possible to solve this problem?

JulioC, 2021-03-02:
I was able to pair my Zemismart 4 gang Scene Switcher with the following devices.js definition (fingerprint taken from their [official specs](https://drive.google.com/file/d/1zQuF92Hs2ryj_nuzIPmwgKJB3o_rzFKL/view))

```js
    {
        zigbeeModel: ['TS0044'],
        model: 'TS0044',
        vendor: 'TuYa',
        description: 'Wireless switch with 4 buttons',
        whiteLabel: [{vendor: 'Lonsonho', model: 'TS0044'}, {vendor: 'Haozee', model: 'ESW-OZAA-EU'}],
        fromZigbee: [fz.tuya_on_off_action, fz.battery],
        exposes: [e.battery(), e.action(['1_single', '1_double', '1_hold', '2_single', '2_double', '2_hold',
            '3_single', '3_double', '3_hold', '4_single', '4_double', '4_hold'])],
        toZigbee: [],
        fingerprint: [
            {manufacturerName: "_TZ3400_key8kk7r", modelID: "TS0043", },
            {manufacturerName: "_TYZB02_key8kk7r", modelID: "TS0043", },
            {manufacturerName: "_TZ3000_bi6lpsew", modelID: "TS0043", },
            {manufacturerName: "_TZ3000_vp6clf9d", modelID: "TS0044", }
        ]
    }
```

matlab22, 2021-03-11:
> 
> 
> Hi, I have the same switch but I was not able to pair it with z2m. Are you able to pair it ? I opened the device and the MCU is a TUYA TYZS3 chip inside which should do Zigbee.

I have the same problem. All 4 LEDs are blinking, but it does not show up in the info log :(

rensenti, 2021-04-10:
> I am not familiar with the packet structure of Zigbee, but there seems to be a delay due to a function that is not implemented.
> 
> [zigbee sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/5977560/zigbee.sniff.zip)
> 
> Is it possible to solve this problem?

Has anyone had a chance to investigate this? The delay is quite annoying for my particular use-case, remote(s) for roller shutter blinds.

What can we do to help?

Ramblurr, 2021-05-13:
I'm interested in a multi button Zigbee switch like this and i have zzh! 

Over in ZHA land they have an additional module/library called zha-quirks that seems to fix this switch  

https://community.home-assistant.io/t/tuya-4-gang-scene-switch/264164/9

kuradi, 2021-05-15:
i really dont understand now... https://www.zigbee2mqtt.io/information/supported_devices.html saying this switch is supported but i cant get it working. i got "device announced" but when i push switch log shows: {"level":"warning","message":"Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html."}. please advise.

closed 2021-06-22 by github-actions[bot]

VoyteckPL, 2021-06-22:
Is it just me or anyone else experience delay with this switch? Sometimes if I press it it takes 1-1,5 seconds before it reacts.

dummys, 2021-06-24:
> Is it just me or anyone else experience delay with this switch? Sometimes if I press it it takes 1-1,5 seconds before it reacts.

same here, never was able to fix

HB9VCT, 2021-07-06:
I have to confirm that TS0043 has a random delays up to 10 seconds with Z2M. After sniffing the protocol I have made the conclusion that this device is waiting for some kind of acknowledge message. Since it is not seen it repeat the command several times and is blocked during this time.
Should be nice is somebody (Koenkk) can have a look at it.

Sjworks, 2021-07-29:
@dummys @VoyteckPL @rensenti @Spirituss 
The delay problem is being solved with Koenkk's help on #8149..
Please check comments in there.