Koenkk/zigbee2mqtt #4469: Strange "recursive" MQTT topics created on mosquitto.

risa2000, 2020-09-26:
I am running `zigbee2mqtt` on RaspberryPi 3B+ (running Raspbian Buster), with `homeassistant` and `mosquitto`. Just did a full system update on all components. (For `zigbee2mqtt` I used `update.sh` script.)

When trying to figure out, which topics exist on `mosquitto` I tried `sudo kill -SIGUSR2 <mosquitto_PID>` which should dump all known topics to the system log. Then, after running `journalctl -u mosquitto.service` I am seeing this:

```syslog
Sep 26 20:59:29 hass mosquitto[17875]:         zigbee2mqtt
Sep 26 20:59:29 hass mosquitto[17875]:           bridge
Sep 26 20:59:29 hass mosquitto[17875]:             state (5dadaed6-867a-4f65-99f2-589e0c39daae, 0) (r)
Sep 26 20:59:29 hass mosquitto[17875]:             device
Sep 26 20:59:29 hass mosquitto[17875]:               +
Sep 26 20:59:29 hass mosquitto[17875]:                 get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                 +
Sep 26 20:59:29 hass mosquitto[17875]:                   get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                   +
Sep 26 20:59:29 hass mosquitto[17875]:                     get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                     +
Sep 26 20:59:29 hass mosquitto[17875]:                       get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                       +
Sep 26 20:59:29 hass mosquitto[17875]:                         get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                         +
Sep 26 20:59:29 hass mosquitto[17875]:                           get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                           +
Sep 26 20:59:29 hass mosquitto[17875]:                             get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                             +
Sep 26 20:59:29 hass mosquitto[17875]:                               get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                               +
Sep 26 20:59:29 hass mosquitto[17875]:                                 get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                 +
Sep 26 20:59:29 hass mosquitto[17875]:                                   get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                   +
Sep 26 20:59:29 hass mosquitto[17875]:                                     get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                     +
Sep 26 20:59:29 hass mosquitto[17875]:                                       get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                       +
Sep 26 20:59:29 hass mosquitto[17875]:                                         get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                         +
Sep 26 20:59:29 hass mosquitto[17875]:                                           get_group_membership (mqttjs_d0f3bcdc, 0)
Sep 26 20:59:29 hass mosquitto[17875]:                                           +
Sep 26 20:59:29 hass mosquitto[17875]:                                             get_group_membership (mqttjs_d0f3bcdc, 0)
```
plus some other strange repetitions. The full log follows here:

[mosq.log](https://github.com/Koenkk/zigbee2mqtt/files/5287216/mosq.log)

I am not very familiar with `mosquitto` nor `zigbee2mqtt` to really comment on the log, as I have no idea how to read `+` topics, but the recursive nature looks suspicious. It also seems like they are getting longer every time I restart `zigbee2mqtt` and/or `mosquitto`. I assume they are created by `zigbee2mqtt`, so I am asking here, hoping, someone might explain them.

sjorge, 2020-09-26:
This is expected behavior as a group or device friendly_name can contain one or more / and there and some devices also have multiple endpoints which also use /

e.g a device called bedroom/switch with has a left and right endpoint will result for in topics like
zigbee2mqtt/bedroom/switch/left/set/state for example to update the left endpoint of the switches state.

As there is no wildcard way to do this z2m will listen on multiple topics so it can both devices without a / in the name, device switch no or multiple endpoints etc.

Koenkk, 2020-09-27:
As @sjorge explained this is indeed expected. Note that the new api does not do this (https://github.com/Koenkk/zigbee2mqtt/issues/3281)

closed 2020-09-27 by Koenkk