Koenkk/zigbee2mqtt #4470: Having several issues with zigbee2mqtt in Home Assistant

riddledaxis, 2020-09-26:
## What happened

All of my devices stopped communicating after the update to 1.14.4.2, and I am unable to control them at all. They appear to be working through the HA add-on logs but when you look at the logs there is no communication. My kitchen lights are on but HA thinks they are off and I am unable to switch them off. I have to turn them off at the wall.

I am also getting errors in the logs when zigbee2mqtt tries to configure my devices

My zigbee2mqtt configuration (with redaction) is here: https://pastebin.com/a4xdd2HP
I have attached a sample of the logs here: https://pastebin.com/RLx0RUEZ 

I tried to go back to version 1.14.4.1 of the add-on from an HA snapshot, but the issues remained so I stripped everything out, factory reset the devices and started with the add-on from scratch on version 1.14.4.2 but the issues remained

Below is a screenshot of the network map and you can see that it shows no communication:
[image]
  
## What did you expect to happen

Everything to continue to work as it did prior to the upgrade

## How to reproduce it (minimal and precise)

I have removed everything including the add-on and the topics in MQTT an reset everything up including a new network key and everything pairs and appears to be working ok and then the routing and LQI errors start in the logs

## Debug info
Zigbee2MQTT version: 1.14.4.2
Adapter hardware: Conbee II
Adapter firmware version: 0x264a0700

Koenkk, 2020-09-27:
The logs are private, please make them public.

riddledaxis, 2020-09-27:
Sorry, they are public now

Koenkk, 2020-09-27:
What do you get in the log when controlling a device? (e.g. turning a bulb on?)

riddledaxis, 2020-09-27:
I turned on Bob's light at the physical switch and Mosquitto* registered the LQI and power consumption, but the state of the switch showed as Off. In HA it was showing the power consumption in my graph but the switched showed as off. I then clicked the switch in HA and Mosquitto then set the state to On and the logs  then started reflecting that. At this point I had also turned on the 6 spots in the kitchen but the logs do not reflect this.

https://pastebin.com/VTtfAja8

*I am using MQTT Explorer to check Mosquitto

Koenkk, 2020-09-27:
I see a network scan is executed while executing this commands. A network scan will drastically reduce the performance of your network, does it also happen when no network scan is executed?

riddledaxis, 2020-09-27:
yes

riddledaxis, 2020-09-27:
I had an automation running the network scan every 10 minutes, I have changed it to every 40 minutes. 

This automation was running before the update and I didn't get any of these issues, everything worked. But it would explain why the kitchen lights took a little while to turn on and off sometimes.

riddledaxis, 2020-09-29:
So, I think I have fixed all of the issues apart from 1. 

I moved over to the Edge version of the Zigbee2MQTT HA app and then factory reset and reattached my devices to the Zigbee network and they just started working (not really sure why) and communicating across the Zigbee network.

[image]
[image]

My remaining issue is that when zigbee2mqtt tries to configure a device it does not work, especially with Bob's Light (SmartThings Zigbee Outlet UK ). The errors from log are below:

https://pastebin.com/r0Q1WGEC

Thank you for your help

Koenkk, 2020-09-29:
What is the model of the plug? (`F-APP-UK-V2`?)

riddledaxis, 2020-09-29:
Yes, it is an F-APP-UK-V2

Koenkk, 2020-09-29:
Can you try changing the definitions in `devices.js` to:

```js
{
    zigbeeModel: ['3200-Sgb'],
    model: 'F-APP-UK-V2',
    vendor: 'SmartThings',
    description: 'Zigbee Outlet UK with power meter',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.on_off, fz.electrical_measurement_power],
    toZigbee: [tz.on_off],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
        await configureReporting.onOff(endpoint);
        try {
            await readEletricalMeasurementPowerConverterAttributes(endpoint);
        } catch (e) {
        }
        // Limit updates to 3V and max 600s (10m)
        await configureReporting.rmsVoltage(endpoint, {max: 600, change: 3});
        // Limit updates to 0.01A and max 600s (10m)
        await configureReporting.rmsCurrent(endpoint, {max: 600, change: 10});
        // Limit updates to 4.0W and max 600s (10m)
        await configureReporting.activePower(endpoint, {max: 600, change: 40});
    },
},
```

It could be that after this change the reported power is way too much (e.g. x10 or x100). Please check if the reported `voltage`, `current` and `power` make sense.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

riddledaxis, 2020-09-29:
I made the change, and it as far as I can tell it is all working :)

``` 
info  2020-09-29 18:43:24: Configuring 'Bob's light'	--- I could not find a corresponding error as before :)

info  2020-09-29 18:43:31: MQTT publish: topic 'zigbee2mqtt/Bob's light', payload '{"linkquality":44,"power":92,"state":"ON","voltage":241}'
debug 2020-09-29 18:43:31: Received Zigbee message from 'Bob's light', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":61}' from endpoint 1 with groupID null
info  2020-09-29 18:43:31: MQTT publish: topic 'zigbee2mqtt/Bob's light', payload '{"current":61,"linkquality":44,"power":92,"state":"ON","voltage":241}'
debug 2020-09-29 18:45:04: Received Zigbee message from 'Bob's light', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":238}' from endpoint 1 with groupID null
info  2020-09-29 18:45:04: MQTT publish: topic 'zigbee2mqtt/Bob's light', payload '{"current":61,"linkquality":44,"power":92,"state":"ON","voltage":238}' 
```
As I am using a custom devices.js file, will I need to update it when Daniel updates the add-on?

riddledaxis, 2020-09-29:
Sorry to be a pain.

I am assuming the issues I seem to be having with the Osram AB35996 bulbs is similar to the plug issue?

https://pastebin.com/dA1sGTWK

Koenkk, 2020-09-29:
@riddledaxis looks that the power/current is off, can you connect a known load to the plug and provide the logging of this and the load in watts?

Other isssue is a different one, looks conbee related, try switching to the edge version of the addon as it contains some fixes for conbee.

riddledaxis, 2020-09-30:
The power consumption for the plug always seems to be approximately 91 watts. It does very during the evening. I will test the plug during the day tomorrow, and get back to you.

I had already moved to the Edge version of the add-on, as this is how I managed to get everything talking again.

My Conbee II is running 0x264a0700, should I upgrade it? If so is there a recommend one I should update to as this firmware is from 27/05/2019?

Thanks

dwest576, 2020-10-03:
> So, I think I have fixed all of the issues apart from 1.
>
> I moved over to the Edge version of the Zigbee2MQTT HA app and then factory reset and reattached my devices to the Zigbee network and they [...]

Off topic, how do you make your coord to router lines green?  I love it!

riddledaxis, 2020-10-03:
@dwest576 it is this HACS custom card. 

You can change the colours when you add the card, but it's set up with green by default, and as I use a dark theme thought it would look good 

https://github.com/azuwis/zigbee2mqtt-networkmap

riddledaxis, 2020-10-03:
@Koenkk I have updated my ConBee II to the latest firmware 0x26650700, but I am still getting these errors:

https://pastebin.com/uv0AQbh5

Also, when I updated the HA Edge app to 0.115 by removing and reinstalling (as per Daniel's instructions) the OSRAM lights disappeared from HA.

The weird thing is they still show up in Zigbee2MQTT and I can change their state in the app's web interface, but their entities don't exist in HA anymore.

I am going to try and re-add them tomorrow.

closed 2020-11-10 by github-actions[bot]