Koenkk/zigbee2mqtt #4472: SmartThings Arrival (F-ARR-UK-2) - Beep

danieledwardgeorgehitchcock, 2020-09-27:
## What happened
I can see in this issue here #459 and this commit #118 that the SmartThings Arrival sensor should be supported in Zigbee2MQTT and indeed, when I added the device, it paired without any herdsman errors. The device seems to however, be reported as a STS-PRS-251 - Not sure if this is correct? But that model seems to align with the same feature set as my device.

## What did you expect to happen
When I publish to the topic `zigbee2mqtt/Daniel Presence/set` with the payload `{"beep":5}` from MQTT Explorer, nothing happens with the device. I can confirm that all other functionality seems to work (presence, battery, etc).

## How to reproduce it (minimal and precise)
Bind -ARR-UK-2 (Firmware R860_#35)
Send payload to devices set topic of `{"beep:5"}`

## Debug info
Zigbee2MQTT version: 1.15 (dev)
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200805
Relevant debug info:
```
ERROR: Publish 'set' 'beep' to 'Daniel Presence' failed: 'Error: Cluster 'genIdentify' has no command 'identifyTime''
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'beep' to 'Daniel Presence' failed: 'Error: Cluster 'genIdentify' has no command 'identifyTime''","meta":{"friendly_name":"Daniel Presence"},"type":"zigbee_publish_error"}'
```

closed 2020-11-06 by github-actions[bot]

Jhonattan-Souza, 2021-04-19:
@danieledwardgeorgehitchcock Did you solve your problem ? I am with the same error here

danieledwardgeorgehitchcock, 2021-04-20:
Still an issue for me

Jhonattan-Souza, 2021-04-20:
Hi @Koenkk, can you help us find the problem? 
I don`t know where to start off , zigbee-herdsman debug or sniffing?

Koenkk, 2021-04-20:
@pixeye33  does this still work for you?

pixeye33, 2021-04-20:
It never worked reliably for me, and the lack of community support for that device was not reassuring : i've send the device back quickly after the support was merged.

danieledwardgeorgehitchcock, 2021-04-21:
@Koenkk I have a couple of these devices and can provide debug logs / database records / sniffs if required? This is a little bit beyond my expertise when it comes to converters...

Koenkk, 2021-04-21:
@danieledwardgeorgehitchcock sniffing when sending this command might help: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

danieledwardgeorgehitchcock, 2021-04-28:
@Koenkk I have captured the traffic - I sent the beep command 5 times so, hopefully you can see that in the capture.

Source ID: 0x0C2B

Koenkk, 2021-04-28:
@danieledwardgeorgehitchcock I don't see the command being send, are you sure you started sniffing before sending the command?

Can you also provide the herdsman debug logging when sending the command. 
See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

danieledwardgeorgehitchcock, 2021-04-28:
@Koenkk - I think this may be because the Zigbee payload isn't even getting genereated by Zigbee2MQTT.

I get this message in the debug logs:

```
Zigbee2MQTT:info  2021-04-28 18:48:48: MQTT publish: topic 'zigbee2mqtt/Daniel Presence', payload '{"action":"","battery":60,"linkquality":93,"presence":true,"voltage":2800}'
Zigbee2MQTT:debug 2021-04-28 18:49:02: Received MQTT message on 'zigbee2mqtt/Daniel Presence/set' with data '{"beep":""}'
Zigbee2MQTT:debug 2021-04-28 18:49:02: Publishing 'set' 'beep' to 'Daniel Presence'
Zigbee2MQTT:error 2021-04-28 18:49:02: Publish 'set' 'beep' to 'Daniel Presence' failed: 'Error: Cluster 'genIdentify' has no command 'identifyTime''
Zigbee2MQTT:debug 2021-04-28 18:49:02: Error: Cluster 'genIdentify' has no command 'identifyTime'
    at Object.getCommand (/app/node_modules/zigbee-herdsman/dist/zcl/utils.js:107:19)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:468:37)
    at Generator.next (<anonymous>)
    at /app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:23:12)
    at Endpoint.command (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:466:16)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1992:26)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:224:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:384:44)
Zigbee2MQTT:info  2021-04-28 18:49:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'beep' to 'Daniel Presence' failed: 'Error: Cluster 'genIdentify' has no command 'identifyTime''","meta":{"friendly_name":"Daniel Presence"},"type":"zigbee_publish_error"}'
```

Koenkk, 2021-04-28:
lol, good catch, should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

danieledwardgeorgehitchcock, 2021-04-29:
Thanks @Koenkk - Tested this morning and it works when sending a payload of `{"beep":"5"}` to the /set topic via Home Assistant / MQTT Explorer.

I cannot say the same for activating the beep via the frontend GUI - I will try and look in to see if there is anything I can fix on that today.

@Jhonattan-Souza - can you confirm the same?

Jhonattan-Souza, 2021-05-01:
@danieledwardgeorgehitchcock It is the same for me, beep is now working fine! But only sending the payload via mqtt.

Thanks @Koenkk !

danieledwardgeorgehitchcock, 2021-05-02:
@Jhonattan-Souza the reason for this is that the button in the GUI does not have a value bound to it - it just submits a blank request.

I think the desired behaviour can be accomplished using a composite (which will have the button alongside a numeric slider) - @Koenkk introduced me to these in a different post.

I am just trying to get my head around how composites work, but will let you know if / when I have managed to figure it out and have submitted a PR.