Koenkk/zigbee2mqtt #4473: CC2652RB - ser2net not working

matoxp, 2020-09-27:
## What happened
zigbee2mqtt is unable to communicate with slaesh's CC2652RB stick via ser2net. After starting, the following error message is raised:

```
Zigbee2MQTT:error 2020-09-27 13:17:50: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-09-27 13:17:50: Failed to start zigbee
Zigbee2MQTT:error 2020-09-27 13:17:50: Exiting...
Zigbee2MQTT:error 2020-09-27 13:17:50: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/.../zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:90:31)

```

My ser2net configuration looks like this:

```
20108:raw:0:/dev/ttyUSB0:115200 8DATABITS NONE 1STOPBIT
```

And in zigbee2mqtt's configuration.yaml I have

```
serial:
   port: 'tcp://192.168.1.101:20108'
```

The stick gets detected properly:

```
[9206513.153910] usb 1-1.5: new full-speed USB device number 6 using dwc_otg
[9206513.291860] usb 1-1.5: New USB device found, idVendor=10c4, idProduct=ea60
[9206513.291879] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[9206513.291890] usb 1-1.5: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[9206513.291900] usb 1-1.5: Manufacturer: Silicon Labs
[9206513.291910] usb 1-1.5: SerialNumber: ...
[9206513.295064] cp210x 1-1.5:1.0: cp210x converter detected
[9206513.302716] usb 1-1.5: cp210x converter now attached to ttyUSB0
```

This setup was previously working fine with a CC2531 stick (albeit with *ttyACM0*). Looking at traffic with tcpdump, I can see that the TCP connection to ser2net gets established successfully and data is sent, but there is zero response from the stick.

As a workaround, I am now using *usbip* to remotely access the adapter via TCP and this works fine.

## What did you expect to happen
zigbee2mqtt should start up correctly.

## How to reproduce it (minimal and precise)
Attach a CC2652RB to a linux box, install, configure and start ser2net, point your zigbee2mqtt configuration to the ser2net host and start zigbee2mqtt.

## Debug info
* Zigbee2MQTT version: 1.14.4
* Adapter hardware:  slaesh’s CC2652RB stick
* Adapter firmware version: {"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200830,"transportrev":2},"type":"zStack3x0"} (preflashed by slaesh)
* ser2net version: 2.10.1-1 (running on a RPi with Raspian 9 (stretch)

tim-brand, 2020-10-22:
I have the exact same issue.
@matoxp , how is that usbip working for you? Is it stable? And can you just installed it at the side of where the USB-device is attached and can insert the address in z2mqtt? Or do I also need to install usbip at the z2mqtt side?

matoxp, 2020-10-23:
> @matoxp , how is that usbip working for you? Is it stable? And can you just installed it at the side of where the USB-device is attached and can insert the address in z2mqtt? Or do I also need to install usbip at the z2mqtt side?

@tim-brand It takes a little more effort to set up compared to ser2net. But once its running, it is quite stable (I have it running without any issuess for weeks now).  If something breaks though (e.g. due to network issues), it won't automatically recover, so you have to manually do unbind/bind/detach/attach.

You will need usbip on the zigbee2mqtt side as well. The [instructions on on the ArchWiki](https://wiki.archlinux.org/index.php/USB/IP) are quite helpful. The device will show up as a virtual usb-serial port on the z2m side, so you have to configure this like a regular serial port (e.g. serial.port: /dev/ttyUSB0).

tim-brand, 2020-11-18:
@matoxp Still thanks for the tip. I've used the usbip with success, but it's not a very portable solution. Like on a connection loss, the usbipd doesn't unbind automatically.

@Koenkk Is there any possibility the CC2652RB will be supported via ser2net?

Koenkk, 2020-11-19:
AFAIK nothing special was added to get this working for CC2531 (but tbh I don't know anything about ser2net). @slaesh maybe you can check why it doesn't work?

slaesh, 2020-11-19:
Hey Tim!

It will work.. BUT!

I am no ser2net expert.. If we would try to use my "debug" coordinator firmware.. Which uses the onboard LEDs to visualize the CPU state..
I assume we will see that ser2net will force the MCU into reset state. Cause of the auto-bootloader-schematic. RTS is used to reset the MCU.

So.. Maybe there is a way to tell ser2net to NOT use RTS and DTR.. That would be awesome and the easiest approach!

The hammer method would be to cut the solder-pads like described here:

https://slae.sh/projects/cc2652/#cant-get-any-serial-debug-messages

BUT you need to know, that updating the stick becomes a bit more complicated. You have to use the buttons instead of just pushing the new image via USB 😎

tim-brand, 2020-11-19:
Thanks @Koenkk and @slaesh for the replies!
I'm neither a ser2net expert but will take a look into it. (I'm not a noob in code, so hope this helps ;) 

If cutting the solder-pads will be necessary, I will do it, thanks for the tip.

matoxp, 2020-11-19:
@tim-brand Looking at the man page for ser2net, there is a section about 'device configuration options':

>  [-]RTSCTS turns on (- off) hardware flow control.
>  [-]LOCAL ignores (- checks) the modem control lines (DCD, DTR, etc.)
>  [-]HANGUP_WHEN_DONE lowers (- does not lower) the modem control lines (DCD, DTR, etc.) when the connection closes

So maybe adding -RTSCTS would do the trick?

slaesh, 2020-11-19:
Unfortunately, that would not be enough I guess.

This function looks promising:

https://sourceforge.net/p/ser2net/code/ci/master/tree/portinfo.c#l363

Which sets RTS/DTR to a given requested value!!

It will be called from here:

https://sourceforge.net/p/ser2net/code/ci/master/tree/controller.c#l522

Docs found here: 

https://linux.die.net/man/8/ser2net

[image]

BUT: `These do not stay between connections` =//

tim-brand, 2020-11-21:
@slaesh I've done some testing with the debug firmware.
And I indeed noticed that the moment a connection is started through ser2net, the blue and red lights are turning off. The moment it gets disconnected, they turn back on again.
I've been trying to change the RTS and DTR by using the `setportcontrol`, but that didn't seems to do anything.
However. When my connection was open (using a telnet session) and the lights were off, I started the miniterm command including the rts and dtr parameters, and then the lights turned on again. When I send some lines using my telnet session, I see the red light turning off and on. So during the session it's possible to change them, and getting that `active` connection.

tim-brand, 2020-11-23:
I've made a simple nodejs application where I proxy the data of the stick from the serial port to a socket, where zigbee2mqtt is connecting to. This works perfect. So this tool is actually just replacing ser2net.
I will make it production ready and release it as a npm package, which can be run as a daemon. So it will be a working replacement of ser2net.

Unfortunately after this testing I started having the issue as reported in #4135 .
At first I thought it was related to my tool. But the problem still exists when I connect it back directly to the host where z2m is running on. I will add some details of this into the other issue.
The issue is stale, @Koenkk could you maybe provide some help?

mbrackjr, 2020-12-15:
Would someone be able to provide some details on how to make the CC2652RB work with something like ser2net.  I use the Z2M plugin for Hassio, which also has native support for 'socat'.  What settings to use to make this work?  Thanks!

asimonen, 2020-12-16:
I'm using a bit different USB adapter, but a quick suggestion anyway; check your ser2net version, as apparently at least version 2 may not work with all CC26x adapters.

I had been running CC2531 with an ancient ser2net 2.10 with socat 1.7 for some time without problems, but once I upgraded the adapter to zzh!, I was just getting these timeouts and nothing worked. I was about to debug that with my laptop but realized that things started working right away, and the main difference seemed to be the ser2net version.

Compiling ser2net 4.2.3 with gensio 2.1.7 for my regular "USB bridge box" resolved all those problems. It also required no special settings and my config is actually pretty much the same as the one shown in https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html . But then, zzh!'s schematic suggests it doesn't have RTS or DTR wired in CH340, so maybe it's not as "picky" as slaesh's adapter regarding port settings(?).

tim-brand, 2020-12-17:
O that's good news @asimonen .  I just checked my version of ser2net, which is currently 3.5. So I might try to upgrade.

I can image it's working fine as I created a serial to socket proxy myself, in NodeJS and I didn't had to do anything special to make it work as well. Even when my tool is working fine, I prefer to use ser2net, so will try to upgrade it.

MarcelRoozekrans, 2020-12-21:
@tim-brand  I tried the ser2net 4.2.3 as suggested by @asimonen suggested unfortunatly with no luck for me.

sjorge, 2020-12-24:
I'm hitting this too with zzhp-lite prototype, using a FreeBSD on the raspberry pi. It does indeed look to be pulling RTS and DTS to low imediatly after connecting. (Also when opening locally).

I wonder if the telnet mode support my ser2net could help? I think with the remctl param set those can be toggled. z2m would need to speak RFC 2217 and toggle those though, which I believe it currently does not.

sjorge, 2020-12-26:
@Koenkk some more poking about...

I believe it does not work because of this: https://github.com/Koenkk/zigbee-herdsman/blob/e171094f8609cf6129e66baf0cb051ae665bf292/src/adapter/z-stack/znp/znp.ts#L208

We can't do it on a plain socket, supporting RFC 2217 might indeed allow us to do this. But I found no node implementation.

tim-brand, 2020-12-26:
@MarcelRoozekrans I've also already tested that version, same result. It still doesn't work.

tim-brand, 2020-12-26:
I created a node version ser2net (named ser2nodenet :D )
On my macbook my service is working fine and does proxy all the data over tcp to zigbee2mqtt. When I run that same service on my raspberry pi, it doesn't work. I notice that I'm not getting any data from the device into my service. Haven't found any reason for the difference between my mac and the rpi yet. I tried different types of raspberry pi's.

MarcelRoozekrans, 2020-12-27:
@tim-brand I tested your ser2nodenet as well on a rapsberry pi with the same result.

sjorge, 2020-12-27:
Maybe doing the skip bootloader step in ser2nodenet could help?

tim-brand, 2020-12-27:
@sjorge I will give that another try. I've tested that, but not sure if it was on the correct conditions.
But, when I start ser2nodenet on my macbook I already get some data when the port gets opened. I'm not getting this on the rpi.
I think one of the dependent libraries is acting different on the other OS.

FYI. If you run ser2nodenet with `DEBUG=ser2nodenet*` prefixed, it will output detailed logs.

sjorge, 2020-12-27:
Might just be that macOS initializes the port in the correct state for the soc to start.

While other OS’s might initialize them in a way we end up in the bootloader instead.

~ sjorge

sjorge, 2020-12-27:
@tim-brand I got it to work with the change on a freebsd rpi3!
But... the code is horrible and needs a bit of cleanup... because I can't do the sleep thing... it works like 2/3 times!

```javascript
root@freebsd:~/ser2nodenet # git diff
diff --git a/lib/port-proxy.js b/lib/port-proxy.js
index 6ce71bb..f184f33 100644
--- a/lib/port-proxy.js
+++ b/lib/port-proxy.js
@@ -89,6 +89,16 @@ module.exports = class PortProxy {
           }
         }
         log.info('SerialPort opened')
+
+        log.info('SerialPort toggling dtr/rts to skip bootloader')
+
+        await this.serialPort.set({dtr: false, rts: false})
+        // wait 150ms
+        await this.serialPort.set({dtr: false, rts: true})
+        // wait 150ms
+        await this.serialPort.set({dtr: false, rts: false})
+        // wait 150ms
+
         resolve()
       })
     })
```

sjorge, 2020-12-27:
It probably also needs to be gated behind a 'skipBootloader' flag in the config, the config took me ages to figure out.

It shouldprobably be called this z2m-proxy or something at that point though.

tim-brand, 2020-12-27:
Thanks @sjorge , I will take a look at the patch and try it myself.

The idea was to make this ser2nodenet configurable to also run other devices. But I agree what it might be better to focus the project as a proxy for z2m. I will probably refactor the project.

tim-brand, 2020-12-27:
Got it working! Thanks @sjorge !
I already know why I stopped using that skip-bootloader part, because it didn't seem to be required on my macbook, during development. Didn't thought it would make the difference on the pi.
I only changed the patch a bit so that the method for setting the serialport options is really async, this will probably help to make it work 3 out of 3 times ;)

As said, I will change the project to make it a z2m proxy only. And of course add some documentation ;)

sjorge, 2020-12-27:
I noticed a huge difference in drivers and default assumptions...
macOS: seems to initialize at 115200 baud with DTR/RTS in the correct state
freeBSD: initializes at 9600 baud with DTR/RTS so that it always hits the bootloader
illumos: initializes at 9000 baud, DTR/RTS so it always hits the bootloader

I haven't tried linux, but I'd guess based on the issues on the rpi it is the same as freebsd/illumos and that it comes up in the bootloader mode.

closed 2021-02-04 by github-actions[bot]

MrRodz, 2021-02-18:
hi is there any update on this please

sjorge, 2021-02-18:
https://github.com/tim-brand/zigbee2mqtt-adapter-proxy is working fine for me, exporting form a rPI 3 running FreeBSD to a zone running on illumos. (As illumos lacks the required serial drivers)

lolorc, 2021-02-19:
I had success with usbip flashing my cc2652r plugged on a rpi3 from my laptop.
it's not very straightforward, I now temporarily plug it to my laptop :-)

MrRodz, 2021-02-21:
I’m looking at using the cc2652rb in a pi 3b+. I’ll be honest, I don’t have a clue where to start installing Jason files for the proxy. Any advice would be great full

sjorge, 2021-02-21:
You need to clone the repo and run:
npm install

~ sjorge

KorbinianP, 2021-02-22:
> 
> You need to clone the repo and run: npm install ~ sjorge

Hey, a few more details would help me. The repo sadly has no README. I cloned and installed. Whats next?
Thank you!

Cheers

sjorge, 2021-02-22:
> > You need to clone the repo and run: npm install ~ sjorge
> 
> Hey, a few more details would help me. The repo sadly has no README. I cloned and installed. Whats next?
> Thank you!
> 
> Cheers

You need to add a config file

```yaml
[iris :: sjorge][/opt/ser2nodenet][master *]
[.]$ cat config.yaml
---
ports:
  - path: /dev/cuaU4
    baudrate: 115200
    socketPort: 2652
    socketAddress: "::"
    rtscts: false
    skipBootloader: true
```

I think you can have multiple ports but I have not tried. I have a entry in /etc/rc.local (freebsd) that starts it in a screen.

```
/usr/local/bin/screen -dmS zzh-lite /usr/local/bin/bash -c 'cd /opt/ser2nodenet; while true; do /usr/local/bin/npm start -- config.yaml; done'
```

KorbinianP, 2021-02-22:
Hey,
thank you for your answer! But now I am a bit confused. 4 days ago you wrote that you use [zigbee2mqtt-adapter-proxy](https://github.com/tim-brand/zigbee2mqtt-adapter-proxy), in your answer today you use [ser2nodenet](https://github.com/tim-brand/ser2nodenet). On second I can start once I have the config in place, but it crashes. Done npm install, created the config.yaml, started first manually with `npm start -- config.yaml`. First doesn't have a start script.

My Config:
`ports:
  - path: /dev/ttyUSB0
    baudrate: 115200
    socketPort: 20108
    socketAddress: "::"
    rtscts: false
    skipBootloader: true
`

Result

> ~/ser2nodenet $ npm start -- config.yaml
> > serialport-proxy@1.0.0 start /home/pi/ser2nodenet
> > node index.js "config.yaml"
> 
> /home/pi/ser2nodenet/lib/port-proxy.js:155
>   onSocketData = (data) => {
>                ^
> 
> SyntaxError: Unexpected token =
>     at Module._compile (internal/modules/cjs/loader.js:723:23)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
>     at Module.load (internal/modules/cjs/loader.js:653:32)
>     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
>     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
>     at Module.require (internal/modules/cjs/loader.js:692:17)
>     at require (internal/modules/cjs/helpers.js:25:18)
>     at Object.<anonymous> (/home/pi/ser2nodenet/index.js:2:19)
>     at Module._compile (internal/modules/cjs/loader.js:778:30)
>     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! serialport-proxy@1.0.0 start: `node index.js "config.yaml"`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the serialport-proxy@1.0.0 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /home/pi/.npm/_logs/2021-02-22T21_12_50_748Z-debug.log

The changes that you mentioned on 27 December are in.

Could you please help me once more? Do you use ser2nodenet? Or zigbee2mqtt-adapter-proxy? Or both? And did you modify anything else?

sjorge, 2021-02-22:
I'm using https://github.com/tim-brand/zigbee2mqtt-adapter-proxy, before that existed I had https://github.com/tim-brand/ser2nodenet with some hacks to get it to work, I didn't feel like changing all my paths. zigbee2mqtt-adapter-proxy = ser2nodenet + some fixes for sleash/zzhp(-lite) based on my hacks mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/4473#issuecomment-751504465

So you did:
```
git clone https://github.com/tim-brand/zigbee2mqtt-adapter-proxy.git /opt/zigbee2mqtt-adapter-proxy
vim /opt/zigbee2mqtt-adapter-proxy/config.yaml
cd /opt/zigbee2mqtt-adapter-proxy
npm install .
npm start -- config.yaml
```

?

KorbinianP, 2021-02-22:
Hey,
I just cloned again, this time in /opt. Did just like you wrote. No change.

```bash
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ cat config.yaml
ports:
  - path: /dev/ttyUSB0
    baudrate: 115200
    socketPort: 20108
    socketAddress: "::"
    rtscts: false
    skipBootloader: true
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ ll /dev/ttyUSB0
crw-rw---- 188,0 root 22 Feb 19:53 /dev/ttyUSB0
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ npm start -- config.yaml
npm ERR! missing script: start

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-02-22T21_50_34_791Z-debug.log
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ npm --version
6.11.3
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ ll
drwxr-xr-x   - pi 22 Feb 22:45 .git
.rw-r--r--  25 pi 22 Feb 22:45 .gitignore
.rw-r--r-- 140 pi 22 Feb 22:47 config.yaml
.rw-r--r-- 883 pi 22 Feb 22:45 index.js
drwxr-xr-x   - pi 22 Feb 22:45 lib
drwxr-xr-x   - pi 22 Feb 22:49 node_modules
.rw-r--r-- 23k pi 22 Feb 22:45 package-lock.json
.rw-r--r-- 780 pi 22 Feb 22:45 package.json
```

sjorge, 2021-02-22:
I can't get either to work with a fresh install now 🤔 , well ser2nodenet not working was expected.

Try with `node index.js config.yaml` instead of `npm start -- config.yaml`... that at least throws an error for me on a fresh test system (that does not have a stick to export).

KorbinianP, 2021-02-23:
Hello,
thank you for your support! It does change the behaviour. Now I get an error message similar to ser2nodenet:

```bash
pi@zigbeepi:/opt/zigbee2mqtt-adapter-proxy $ node index.js config.yaml
/opt/zigbee2mqtt-adapter-proxy/lib/serialport-proxy.js:140
  onSocketData = (data) => {
               ^

SyntaxError: Unexpected token =
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt-adapter-proxy/index.js:3:25)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
```

https://github.com/tim-brand/zigbee2mqtt-adapter-proxy/blob/8ca0961cd4d2e3ca885b7fc7d9cda171627f6d46/lib/serialport-proxy.js#L140-L150

My Setup: RasPi Zero W with latest Raspian, my Stick is a [zig-a-zig-ah (zzh)](https://electrolama.com/projects/zig-a-zig-ah/)

Maybe I should create an issue at the proxy repo?

sjorge, 2021-02-23:
I’d you’re using zzh you can just use ser2net (the c program) or socat. 

Probably still worth filing an issue against the proxy repo.

~ sjorge

KorbinianP, 2021-02-23:
Hmm, I was using ser2net in the past, last week it suddenly stopped working after I updated my host (proxmox) and zigbee2mqtt. Was going cracy, something about zigbee2mqtt could not open the port and ser2net complaining about not being able to perform some action (io...). Today I tried it with the original device (/dev/ttyUSB0), not the symlink that I create based on the vendor id with udev (/dev/ttyUSBZigbee), and voilla it is working again. One week lost because of a broken symlink :-( 

Thank you for your support! As you see above I create a ticket and a PR for the proxy nontheless.

Cheers

> 
> I’d you’re using zzh you can just use ser2net (the c program) or socat. Probably still worth filing an issue against the proxy repo. ~ sjorge

zombielinux, 2021-03-03:
https://github.com/cminyard/ser2net/issues/46

Take a look at the latest version of ser2net. Should have a fix for our DTR/CTS problem.

timdonovanuk, 2021-09-26:
ZZH stick but after a Pi reboot my USB devices swapped around, e.g. I had ser2net connecting on USB0 and after the reboot...

```
lrwxrwxrwx 1 root root 13 Sep 26 22:17 usb-1a86_USB_Serial-if00-port0 -> ../../ttyUSB1
lrwxrwxrwx 1 root root 13 Sep 26 22:17 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0
```

I wonder if there is a way to make sure this doesn't happen.

KorbinianP, 2021-09-30:
> ZZH stick but after a Pi reboot my USB devices swapped around, e.g. I had ser2net connecting on USB0 and after the reboot...
>
> ```
> lrwxrwxrwx 1 root root 13 Sep 26 22:17 [...]

Hey, you can use UDEV Rules to get a symlink that stays stable after a reboot.
Example: https://wiki.ubuntuusers.de/udev/

zombielinux, 2021-09-30:
> ZZH stick but after a Pi reboot my USB devices swapped around, e.g. I had ser2net connecting on USB0 and after the reboot...
>
> ```
> lrwxrwxrwx 1 root root 13 Sep 26 22:17 [...]

The better solution is to point instead to /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 instead of /dev/ttyUSB0. That way it maps to the correct device every time. As far as I know, that path is valid on raspbian, ubuntu, and the other usual suspects.