Koenkk/zigbee2mqtt #4474: Trust ZYCT-202 not supported?

2N3819, 2020-09-27:
Hi
I tried to pair a newly bought Trust ZYCT-202.

https://www.zigbee2mqtt.io/devices/ZYCT-202.html
I understand these can be challenging but I have not seen this issue reported.

The log reports that the device is not supported:

info  2020-09-27 12:49:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d00038e5fe1"}}'
warn  2020-09-27 12:49:04: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:04: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:04: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:04: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:05: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:05: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:05: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:06: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:06: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:06: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
warn  2020-09-27 12:49:06: Received message from unsupported device with Zigbee model 'ZLL-NonColorController'
warn  2020-09-27 12:49:06: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2020-09-27 12:49:07: Successfully interviewed '0x00158d00038e5fe1', device has successfully been paired
warn  2020-09-27 12:49:07: Device '0x00158d00038e5fe1' with Zigbee model 'ZLL-NonColorController' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Any help would be appreciated.

Koenkk, 2020-09-27:
What zigbee2mqtt version?

2N3819, 2020-09-27:
Hi Sorry for not including the zigbee2mqtt version.
It was release 1.14.1 should have been OK as zyct-202 was added to
zigbee2mqtt in release 1.2.0

I have now updated to 1.14.4 and the original problem has been fixed - I do
not know why.

Now have these errors:
info  2020-09-27 18:37:52: MQTT publish: topic
'zigbee2mqtt/0x00158d00038e5fe1', payload
'{"action":"off","action_group":24550,"linkquality":49}'
error 2020-09-27 18:37:59: Failed to configure '0x00158d00038e5fe1',
attempt 2 (Error: Bind 0x00158d00038e5fe1/1 genOnOff from
'0x00124b00014d96f2/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout
(/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
info  2020-09-27 18:38:18: Configuring '0x00158d00038e5fe1'
info  2020-09-27 18:38:18: MQTT publish: topic 'zigbee2mqtt/bridge/log',
payload
'{"message":"announce","meta":{"friendly_name":"0x00158d00038e5fe1"},"type":"device_announced"}'
info  2020-09-27 18:38:19: MQTT publish: topic 'zigbee2mqtt/bridge/log',
payload
'{"message":"announce","meta":{"friendly_name":"0x00158d00038e5fe1"},"type":"device_announced"}'
error 2020-09-27 18:38:28: Failed to configure '0x00158d00038e5fe1',
attempt 3 (Error: Bind 0x00158d00038e5fe1/1 genOnOff from
'0x00124b00014d96f2/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout
(/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

but I *think* the ZYCT-202 and zigbee2mqtt are functional, I am not 100%
sure as there are errors reported in the Domoticz zigbee2mqtt plugin.
Heading over to the plugin github project to ask for help re the plugin!

closed 2020-11-05 by github-actions[bot]