Koenkk/zigbee2mqtt #4475: Tuya Window covering DT82LEMA-1.2N

czm97176, 2020-09-27:
## Information about the device + link

    Product name: DT82LEMA-1.2N
    Model identifier: u1rkty3
    Device type : Battery Curtain rail track motor

[image]

## data/database.db entry of the device

```
{"id":3,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe688015","nwkAddr":63015,"manufId":0,"manufName":"_TYST11_xu1rkty3","powerSource":"Battery","modelId":"u1rkty3\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"u1rkty3\u0000","manufacturerName":"_TYST11_xu1rkty3","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1599379472957}
```

## converter/devices.js entry of the device

```
    {
        zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja','u1rkty3','u1rkty3\u0000'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'}],
        model: 'TS0601_curtain',
        vendor: 'TuYa',
        description: 'Curtain motor',
        whiteLabel: [
            {vendor: 'Yushun', model: 'YS-MT750'},
            {vendor: 'Zemismart', model: 'ZM79E-DT'},
            {vendor: 'Binthen', model: 'BCM100D'},
            {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },
```
## Minor Issue

Since it's battery driven, but i don't know how to get the battery information, i don't see it from cluster so far

czm97176, 2020-09-30:
@Koenkk Would you consider to add this device code along with `owvfni3` in the devices.js ?

Koenkk, 2020-09-30:
Added, thanks

closed 2020-11-07 by github-actions[bot]