Koenkk/zigbee2mqtt #4477: Some devices change state after zigbee2mqtt startup

fedebruni84, 2020-09-28:
## What happened

When I start zigbee2mqtt some devices change state. In particular one Innr Smart Plug switch OFF and another one switch ON, regardless what was precedent state. 

## What did you expect to happen 

I expect no state change

## How to reproduce it (minimal and precise)

I have zigbee2mqtt (+mosquitto server) running on a Raspberry Pi and when I reboot it or restart after an update, the problem occurs.
Looking at old issues, at the beginning I thought it was a problem of state.json/cache but I deleted the file and the problem persists. Also adding "cache_state_send_on_startup: false" to configuration doesn't resolve this issue.
In the log I can see that during startup, zigbee send a message to the plug topic with payload "ON" (or "OFF" for the other plug) and I really don't know why.
I have also home assistant mqtt integration but this happens also when home assistant is shutted down.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-09-28:
Could you provide the debug logging of this? First enable debug logging, reboot your system (and watch the issue happen), then go into the `data/log` directory and post the log here.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

fedebruni84, 2020-09-28:
> Could you provide the debug logging of this? First enable debug logging, reboot your system (and watch the issue happen), then go into the `data/log` directory and post the log here.
> 
> To enable debug logging set in `configuration.yaml`:
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

For sure...
Look at device innr_samsungTV for example.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5291895/log.txt)

Koenkk, 2020-09-28:
Zigbee2MQTT receives command to control devices:

```
debug 2020-09-28 13:28:23: Received MQTT message on 'zigbee2mqtt/innr_umidificatore/set' with data 'OFF'
debug 2020-09-28 13:28:23: Publishing 'set' 'state' to 'innr_umidificatore'
debug 2020-09-28 13:28:23: Received MQTT message on 'zigbee2mqtt/innr_samsungTV/set' with data 'OFF'
debug 2020-09-28 13:28:23: Publishing 'set' 'state' to 'innr_samsungTV'
debug 2020-09-28 13:28:23: Received MQTT message on 'zigbee2mqtt/innr_plant/set' with data 'ON'
debug 2020-09-28 13:28:23: Publishing 'set' 'state' to 'innr_plant'
debug 2020-09-28 13:28:23: Received MQTT message on 'zigbee2mqtt/innr_frigo/set' with data 'ON'
debug 2020-09-28 13:28:23: Publishing 'set' 'state' to 'innr_frigo'
```

You have to figure out who sends these MQTT messages, probably these are retained MQTT messages and you need to clear them. (http://www.steves-internet-guide.com/mqtt-retained-messages-example/). If that's the case you can clear them by using an MQTT client and sending an empty message to e.g. `zigbee2mqtt/innr_umidificatore/set` with retained set to true. 

Since it's not a Zigbee2MQTT problem I will close this.

closed 2020-09-28 by Koenkk