Koenkk/zigbee2mqtt #4478: Can't integrate TY0A01 - Door Lock

MUzair030, 2020-09-28:
I'm using latest version of zigbee2mqtt & followed this guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html   and tried many times before posting the req.

Here's the log:
```Zigbee2MQTT:debug 2020-09-28 16:38:43: Device '0xec1bbdfffea147e0' announced its                                                                             elf
Zigbee2MQTT:info  2020-09-28 16:38:43: MQTT publish: topic 'zigbee2mqtt/bridge/l                                                                             og', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffea147e0"                                                                             },"type":"device_announced"}'
Zigbee2MQTT:debug 2020-09-28 16:38:48: Device '0xec1bbdfffea147e0' announced its                                                                             elf
Zigbee2MQTT:info  2020-09-28 16:38:48: MQTT publish: topic 'zigbee2mqtt/bridge/l                                                                             og', payload '{"message":"announce","meta":{"friendly_name":"0xec1bbdfffea147e0"                                                                             },"type":"device_announced"}'
Zigbee2MQTT:debug 2020-09-28 16:38:53: Received Zigbee message from '0xec1bbdfffea147e0', type 'attributeReport', cluster 'genBasic', data '{"65506":28,"appVersion":88}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2020-09-28 16:38:53: Received message from unsupported device with Zigbee model 'TY0A01'
Zigbee2MQTT:warn  2020-09-28 16:38:53: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2020-09-28 16:38:54: Received Zigbee message from '0xec1bbdfffea147e0', type 'attributeReport', cluster 'genBasic', data '{"65503":"��\u0004'\u0013��\u0004'\u0012\r�\u0004'\u0012"}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2020-09-28 16:38:54: Received message from unsupported device with Zigbee model 'TY0A01'
Zigbee2MQTT:warn  2020-09-28 16:38:54: Please see: `https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`
Zigbee2MQTT:debug 2020-09-28 16:38:54: Received Zigbee message from '0xec1bbdfffea147e0', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2020-09-28 16:38:54: Received message from unsupported device with Zigbee model 'TY0A01'
Zigbee2MQTT:warn  2020-09-28 16:38:54: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

I with this and also with different available door locks:
```   {
        zigbeeModel: ['TY0A01'],
        model: 'TY0A01',
        vendor: 'Unknown',
        description: 'Chinese Smart Lock',
        supports: 'Lock',
        fromZigbee: [fz.xiaomi_lock_report, fz.ignore_time_read],
        toZigbee: [],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            const binds = ['genBasic'];
            await bind(endpoint, coordinatorEndpoint, binds);
        },
    },
```
including:
```
        fromZigbee: [fz.lock],
        toZigbee: [tz.generic_lock],
```
but it didn't work! :( 
Here is my `database.db`
```{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b000be891f4","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId">
{"id":2,"type":"Router","ieeeAddr":"0x588e81fffed0dba1","nwkAddr":7853,"manufId":4098,"manufName":"_TZE200_9i9dt8is","powerSource":"Mains (single phase)","m>
{"id":3,"type":"EndDevice","ieeeAddr":"0x00606efffe9563e7","nwkAddr":18933,"manufId":4412,"manufName":"OWON","powerSource":"Battery","modelId":"RC204_4C","e>
{"id":4,"ieeeAddr":"0x00606efffe919e3e","nwkAddr":6867,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1600770348810}
{"id":5,"ieeeAddr":"0x00606efffe919e3e","nwkAddr":38911,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1600773616307}
{"id":6,"ieeeAddr":"0x00158d0003a472e7","nwkAddr":59155,"modelId":"lumi.ctrl_86plug.aq1","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClus>
{"id":7,"ieeeAddr":"0xec1bbdfffea1486e","nwkAddr":9519,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic">
{"id":8,"type":"EndDevice","ieeeAddr":"0xec1bbdfffea147e0","nwkAddr":43437,"manufId":0,"manufName":"_TYST12_eyfzd1gi","powerSource":"Battery","modelId":"TY0>
{"id":9,"ieeeAddr":"0x00606efffe919e3e","nwkAddr":784,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1600928335242}
{"id":10,"type":"EndDevice","ieeeAddr":"0x00158d000396115b","nwkAddr":742,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_s>
{"id":11,"type":"Router","ieeeAddr":"0x04cf8cdf3c7e157b","nwkAddr":56768,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"l>
{"id":12,"type":"Coordinator","ieeeAddr":"0x00124b000be891f4","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,47,110,242],"endpoints":{"1":{"profId>
{"id":13,"type":"Router","ieeeAddr":"0x00158d00028bc171","nwkAddr":55719,"manufId":4478,"manufName":"Feibit Co.,LTD","powerSource":"Mains (single phase)","m>
```

@Koenkk , kindly help or add device support for "TY0A01"

Koenkk, 2020-09-28:
TuYa uses non-standard commands, you will need to sniff the traffic with the original gateway and follow https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d

MUzair030, 2020-10-07:
Hello @Koenkk ,
I followed the tutorial but its not complete. I made a zigbee sniffer, connected the device with its hub and sniffed the data:
i dont know what to do now, How to integrate it with zigbee2mqtt??
Every time i do a single operation on device (i,e. Unlock the door), I receive different Data.. can you plz help?

[image]

[image]

[image]

Waiting for any guidance/tutorial that can help me figure out.

kvvoff, 2020-10-07:
I also bought this door lock (tuya M500). And I was also unable to add its support to z2m. Any tests are welcome, ready to help

kvvoff, 2020-10-07:
@Koenkk @MUzair030 
I also bought this door lock (tuya M500). And I was also unable to add its support to z2m. Any tests are welcome, ready to help

My z2m log:
`Zigbee2MQTT:info  2020-08-18 08:14:07: Successfully interviewed '0x14b457fffe3cb090', device has successfully been paired
Zigbee2MQTT:warn  2020-08-18 08:14:07: Device '0x14b457fffe3cb090' with Zigbee model 'TY0A01' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-08-18 08:14:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x14b457fffe3cb090","supported":false}}'`

Koenkk, 2020-10-07:
@MUzair030 I don't have knowledge about TuYa devices, all I know is that they use a custom protocol (which makes integration very hard and time consuming). There are some people which know a lot about integrating TuYa devices (e.g. in #4185). Personally I don't have time to dive deep into the TuYa protocols, so I'm sorry I cannot help further here.

UzairMushtaq-Hoo, 2020-10-08:
okay Thank you @Koenkk 👍

yoprogramo, 2020-12-03:
Someone had luck on supporting this lock?

szaman-89, 2021-01-13:
> Hello @Koenkk ,
> I followed the tutorial but its not complete. I made a zigbee sniffer, connected the device with its hub and sniffed the data:
> i dont know what to do now, How to integrate it with [...]

I ended up in the same place. Have no idea how to identify payload structure. Did you moved forward with this?

rospogrigio, 2021-02-02:
Hi @MUzair030 and @szaman-89 , I am trying too to sniff the traffic of this device but I cannot get any packet, perhaps you could provide some help?
Thank you

szaman-89, 2021-02-02:
> Hi @MUzair030 and @szaman-89 , I am trying too to sniff the traffic of this device but I cannot get any packet, perhaps you could provide some help?
> Thank you

Did you pair it with tuya gateway? Did you changed channels to find traffic in wireshark?

rospogrigio, 2021-02-03:
> Did you pair it with tuya gateway? 
what do you mean by "it"? the device or the sniffer? I tried several times to sniff the traffic (both with the Rpi and with Windows), and then re-pair the smart lock with the gateway,

> Did you changed channels to find traffic in wireshark?
I tried channels 11 and 12, should I try to use different channels?

szaman-89, 2021-02-03:
> what do you mean by "it"? the device or the sniffer? I tried several times to sniff the traffic (both with the Rpi and with Windows), and then re-pair the smart lock with the gateway,

I mean pairing lock with tuya gateway.

> I tried channels 11 and 12, should I try to use different channels?

Yes, try to start with highest channel and then check lower ones. Chinese gateways usually works on channels 20-25.

rospogrigio, 2021-02-03:
> I mean pairing lock with tuya gateway.

OK then I already did that.

> Yes, try to start with highest channel and then check lower ones. Chinese gateways usually works on channels 20-25.

Ooooh, good to know, did not imagine that. Will try the other channels and see if anything comes out, thank you.

One question: if you use linux, do you see a device in /dev associated to the sniffer? Because I don't, and I don't know if it's a problem...

szaman-89, 2021-02-03:
> One question: if you use linux, do you see a device in /dev associated to the sniffer? Because I don't, and I don't know if it's a problem...

I don't know. I'm sniffing with wireshark on windows.

rospogrigio, 2021-02-03:
OK , and what channels you are using, as a curiosity?

szaman-89, 2021-02-03:
> OK , and what channels you are using, as a curiosity?

Mine gateway works on channel 20.

rospogrigio, 2021-02-05:
@szaman-89 thank you for your help, I tried a lot of channels and finally discovered my device is using channel 15 so I was able to capture both pairing and opening traffic. I'll try to have a look at it and see if I can figure out how it works.
Thank you!

szaman-89, 2021-02-05:
> @szaman-89 thank you for your help, I tried a lot of channels and finally discovered my device is using channel 15 so I was able to capture both pairing and opening traffic. I'll try to have a look at it and see if I can figure out how it works.
> Thank you!

I'm counting on you :D I have no idea how to interprate payloads.

rospogrigio, 2021-02-05:
OK I am at the same point as you guys. We have to figure out what that 16-bytes payload is. It looks encrypted in some way, or maybe just encoded. Will try to ask the manufacturer and see if I can get something...

rospogrigio, 2021-02-11:
OK I've done some sniffing and this is what I've collected. Please confirm that you see the same with your devices. I see that 5 commands are exchanged between the ZC (the gateway) and the ZED (the smart lock). In the following list, "-->" means from ZC to ZED, and "<--" from ZED to ZC.
1) --> , command id 0x00, sequence number N 
2) <-- , command id 0x02, sequence number M 
3) <-- , command id 0x02, sequence number M+1 
4) <-- , command id 0x23, sequence number M+2 
5) --> , command id 0x23, sequence number N+1 

Now, each of these commands has a 16-bytes payload which looks somehow encrypted, and changes at each interaction. My impression is that it could be a sort of key-exchange mechanism, in which the lock sends a certain key and it expects the gateway to respond in the correct way in order to unlock successfully. Unfortunately, it is very hard (if not impossible) to determine how this acknowledge works without knowing the involved protocol. Maybe we could ask the manufacturer if he can disclose something but I highly doubt he will. I've ben asking to my manufacturer (which is Airbnk, wonder if yours is the same), but with no luck so far, even if the interaction is still proceeding.
If you happen to have any info by your manufacturer or if you have some ideas on how to decrypt this please share them.
Bye!

Faustino-ai, 2021-02-15:
>

> OK I've done some sniffing and this is what I've collected. Please confirm that you see the same with your devices. I see that 5 commands are exchanged between the ZC (the gateway) and the ZED (the [...]

I am also working on this. One slightly out of topic question though, can you find the 'tuya.dump.txt' which according to the guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html
should be in /data but I couldn't find mine. As for the protocol, you can try creating an account on tuya's site: https://developer.tuya.com/en/

rospogrigio, 2021-02-15:
> I am also working on this. One slightly out of topic question though, can you find the 'tuya.dump.txt' which according to the guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html
> should be in /data but I couldn't find mine. 

I think you should find it in /opt/zigbee2mqtt/data , but I can't find it either, even though I am not at that point yet.

> As for the protocol, you can try creating an account on tuya's site: https://developer.tuya.com/en/
> I remember someone found something there but it was for the thermostat. Not sure if it works the same for smart lock since the security level is very different for locks.

Yeah I already have an account but couldn't find anything useful, if you have a specifiv place to search to suggest tell me, thank you.

Faustino-ai, 2021-02-15:
> > I am also working on this. One slightly out of topic question though, can you find the 'tuya.dump.txt' which according to the guide: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html
> > should be in /data but I couldn't find mine.
> Nope couldn't find it either. I am suspecting the smart lock to work very differently from the other tuya devices (like thermostats and lights) that people worked on here. And I am relatively new in this area as well so I am extremely confused right now.
> I think you should find it in /opt/zigbee2mqtt/data , but I can't find it either, even though I am not at that point yet.
> 
> > As for the protocol, you can try creating an account on tuya's site: https://developer.tuya.com/en/
> > I remember someone found something there but it was for the thermostat. Not sure if it works the same for smart lock since the security level is very different for locks.
> 
> Yeah I already have an account but couldn't find anything useful, if you have a specifiv place to search to suggest tell me, thank you.

Probably nothing useful on that website. I am trying to reach out to the tuya customer service but I am not so sure what to ask them.

rospogrigio, 2021-02-15:
I think you can just tell them you would like to operate the device for integration on a third party software, and that you need support and see where the conversation goes.

Faustino-ai, 2021-02-15:
> I think you can just tell them you would like to operate the device for integration on a third party software, and that you need support and see where the conversation goes.

I remember seeing on some posts that someone actually did that. But their reply was something along the way of its only possible to use with tuya gateway. Anything outside of that they can't disclose. So I guess someone has to come out with a more indirect question

Faustino-ai, 2021-02-15:
https://developer.tuya.com/en/docs/iot/smart-product-solution/product-solution-lock/product-solution-lock-zigbee/zigbee-doorlock-function-developer?id=K9fe9esofkzlf
I am not sure how useful this is but this is all I can find from the official site. As for outside of this site, there's little to no proper explanation on how it works (at least for a newbie like me).

rospogrigio, 2021-02-15:
Yeah, I've read that document dozens of times, and keep looking at it since it's being updated quite regularly.  I also asked to the Tuya tech staff because it is unclear what they mean by "The password that is configured by the user on the app" or "The secret key that is created by the Tuya cloud platform". I was answered that my device lacks DPs 48 or 49 and that I should ask the reseller (Airbnk) to have them available (I was expecting with some sort of fw update, but they said it is actually needed to replace the device 😮  ). 
I have also tried to operate it using Cloud API commands but apparently the only function available to the user is "doorbell_volume". They have also suggested me to try to launch this API command `GET /v1.0/devices/{device_id}/door-lock/remote-unlocks`, but it returned an empty structure so it means that the function is not available.
I finally also asked almost the same questions to the Airbnk support team, and they replied that they would investigate, but now 2 weeks have passed since their last reply, even though I've been asking for a reply.
Will keep you posted, my personal need would be to be able (in any way) to lock the door remotely (in case I forgot to do it, it would be also nice to be able to schedule it) since I am frankly shocked that this feature is missing (I don't know if yours is missing it too).
Bye!

Faustino-ai, 2021-02-16:
> Yeah, I've read that document dozens of times, and keep looking at it since it's being updated quite regularly. I also asked to the Tuya tech staff because it is unclear what they mean by "The [...]

Oh wow you're already quite deep into it already. Frankly speaking I don't know much but it seems like mine does have the auto lock feature but its the on board function (where it auto locks after 5 - 10 seconds. And yes only within this range) I can't seem to adjust the lock time from the app as well, at least not from my out of the box gateway. My lock's is relatively simple, just lock and unlock. I couldn't even change the door password from the native app (not sure if it is the real native app since my door lock is somehow unbranded) but can set temporary password, which would work as of now if I can succesfully implement it using Home Assistant. 
I will also keep you posted if I found out anything. Somehow the tuya page links to some page that only shows Chinese and I do know Chinese. Unfortunately I am quite new to be able to make use of those information so I will try to study on those and see what I can come up with.

Faustino-ai, 2021-02-16:
[MCU Integration Protocol for Zigbee Lock_Tuya Smart_Docs Center.pdf](https://github.com/Koenkk/zigbee2mqtt/files/5986476/MCU.Integration.Protocol.for.Zigbee.Lock_Tuya.Smart_Docs.Center.pdf)
@rospogrigio Have you gone through this? I got this by trying to develop a new smart lock from the same website.

rospogrigio, 2021-02-16:
> [MCU Integration Protocol for Zigbee Lock_Tuya Smart_Docs Center.pdf](https://github.com/Koenkk/zigbee2mqtt/files/5986476/MCU.Integration.Protocol.for.Zigbee.Lock_Tuya.Smart_Docs.Center.pdf) [...]

Yeah I think I have gone through **everything** 😆 .  I think that doc is quite useless because it describes the internal communication between the Zigbee module and the hardware, while we would need the protocol for communicating with the Zigbee module.
My device also has the auto lock feature but I find it quite useless for my purposes, especially because I bought the smart lock mostly for emergency purposes (if I forgot the keys inside), but would really like to lock the door remotely or at scheduled times (before going to bed for example). As I wrote to the manufacturer, currently both the app and the device are quite basic, compared to the competitors.
What is shocking (and I wrote this both to Tuya support and to Airbnk) is that the app allows remote unlocking, but this feature is not available in the cloud API or by setting a DP, but as far as I understood this is because the app is using a different communication channel (and maybe also protocol?).
Good to know that you know a bit of Chinese, maybe here you can find something useful:
https://developer.tuya.com/en/docs/iot/smart-product-solution/product-solution-lock/product-solution-lock-zigbee/zigbee-doorlock-dp?id=K9fembhbeab0p
OK let me know if you find out something, I'll do the same 😉

Faustino-ai, 2021-02-17:
> Yeah I think I have gone through **everything** 😆 . I think that doc is quite useless because it describes the internal communication between the Zigbee module and the hardware, while we would need [...]

Yeah after reading through the doc, it seems like its just a more specific internal communication documentation. Nothing about the Zigbee module. On the other hand, I actually looked around and saw that my Zigbee module is TYS7 module https://developer.tuya.com/en/docs/iot/device-development/module/zigbee-module/tyzs-series-module/zigbeetyzs7module?id=K989ri7qsqekx
I found the lock's Zigbee module through the hole at the back (where the power connector is). Not sure if this is anything useful.

Faustino-ai, 2021-02-18:
> Yeah I think I have gone through **everything** 😆 . I think that doc is quite useless because it describes the internal communication between the Zigbee module and the hardware, while we would need [...]

A small update on what I've gone through. So apparently, the functions of the door lock is preset by the manufacturer (Airbnk in your case). Since the remote scheduled locking is not part of the compulsory function but instead an "additional function", the manufacturer probably did not choose to include that. So the zigbee module inside the door will not include codes for this function, unless you have a way to flash it (this is how I understood anyway, might be wrong). It is like we can both be using the same zigbee module but with different functions available.
And also, I've contacted the seller and as expected, he doesn't know anything about the Tuya cloud to zigbee module communication. He did however tell me a customer of his from Brazil apparently managed to control the exact same door lock using his own server. I am asking for that person's contact and hopefully it can shed some light for us.

Faustino-ai, 2021-02-18:
[image]
Ok so I found this from a Chinese website and I tried my best to translate it. It basically explains how the password is being managed. Not sure how helpful this is but feel free to point out anything if you have any idea to move forward. @rospogrigio

rospogrigio, 2021-02-18:
Yeah, I had also saw something similar and probably the hex payload that are exchanged are encrypted passwords or messages. Unfortunately we are missing the "developer's access key" you read in the diagram and I doubt they'll disclose it.
I don't know if it's breakable in some way... I'll try to study.

Faustino-ai, 2021-02-19:
> Yeah, I had also sen something similar and probably the 16-bytes payloads that are exchanged are encrypted passwords or messages. Unfortunately we are missing the "developer's access key" you read in the diagram and I doubt they'll disclose it.
> I don't know if it's breakable in some way... I'll try to study.

Do you think the manufacturer will have the accessKey? On the other hand, it seems like the customer from Brazil that I mentioned doesn't want to reveal how he did it, so it's a dead end on this side. But according to seller, it is doable. Not sure if he's  truthful though

rospogrigio, 2021-02-19:
I believe they have their own access key and for obvious reasons they would not disclose it.
I believe that, however, it should be possible to enroll for a developer access key by registering as a developer on Tuya site, but don't know how or where.
If they say it's doable, try to ask the exact sequence of commands that need to be used to do it (I tried to ask the same but with no luck, though).
Why doesn't the brazilian guy want to reveal how he did it? Maybe you could ask the reason behind it, and maybe get some hint?

Faustino-ai, 2021-02-19:
> I believe they have their own access key and for obvious reasons they would not disclose it.

Yea the seller just replied me that he doesn't have it. He's probably just a reseller instead of the developer.

> I believe that, however, it should be possible to enroll for a developer access key by registering as a developer on Tuya site, but don't know how or where.

Well it looks like the access is unique for each device so I wouldn't count on it. 

> If they say it's doable, try to ask the exact sequence of commands that need to be used to do it (I tried to ask the same but with no luck, though).

I did however try out the IOT development and managed to connect my gateway as well as the door to their cloud. I had the access key but doing it twice made me realized they generate a different key every time (which essentially means all developers will have a different key I guess? Again still learning so might be awfully wrong). And also I followed the guide for Tuya cloud api from github and managed to go into the cli part. But then when I tried to do anything, it says not supported for this device. In addition to that, I believed I read from somewhere that for tuya smart lock, parts of the payload are actually from the cloud itself, so what we have on the device is just part of it, which makes it even more difficult.
> Why doesn't the brazilian guy want to reveal how he did it? Maybe you could ask the reason behind it, and maybe get some hint?

Thing is the seller never gave me the guy's contact. He said the guy didn't want to reveal and I have no idea why. And honestly speaking I doubt that he did it anyway. Maybe the seller was not being honest.

Faustino-ai, 2021-02-19:
Also, if you want to try on the Tuya IOT site, they require you to pay or use the trial version (will expire at midnight of the same day for my case). And if you managed to configure everything and see the devices connected but not showing on the device list, you have to change the region. For some reason mine is in Ameirca (yes the spell it this way) although there are much closer region like India or China for my case. So you might need to try all the regions.

rospogrigio, 2021-02-19:
So, I believe the diagram you posted is taken from: https://developer.tuya.com/en/docs/iot/open-api/api-reference/proprietary-category/smart-door-lock?id=K9jgsgd4cgysr#title-4-Password%20creation

Well, I've been trying a lot of API calls using the attached python script.

You have to configure it by replacing MY_CLIENTID, MY_SECRET, the "base" URL with your region, and MY_DEVID.
Now, I am starting to believe that the developer accessKey they are referring to is just the "secret" you get from your area in iot.tuya.com site. If you look at all the attempts I've been making, you'll see:
- the "functions" call: it only returns the "doorbell_volume" function, and the Tuya tech guy told me it is because the unlock feature is not available
- I've been trying to get a "temporary key for encryption", but then I think I should decrypt it in order to use it to "Create a temporary password" but I cannot succeed in decrypting it with my "secret". Either there is something I am missing, or I am doing something wrong.
- should I succeed in getting a temporary password, it is still unclear to me how it should be used to unlock the lock

Play with the tool a bit if you can, and first of all tell me if you get the same result with the "functions" call...
bye 

[api_test.zip](https://github.com/Koenkk/zigbee2mqtt/files/6009874/api_test.zip)

Faustino-ai, 2021-02-19:
> So, I believe the diagram you posted is taken from: https://developer.tuya.com/en/docs/iot/open-api/api-reference/proprietary-category/smart-door-lock?id=K9jgsgd4cgysr#title-4-Password%20creation
> 
Yes it from the site. However the translation itself is not so accurate. According to the Chinese version, the secret key itself is actually encrypted, and which is why I translated the Chinese version instead of using this. But still I will try and see the api_test. Thanks.

> Well, I've been trying a lot of API calls using the attached python script.
>
> You have to configure it by replacing MY_CLIENTID, MY_SECRET, the "base" URL with your region, and MY_DEVID.
> Now, I am [...]

rospogrigio, 2021-02-19:
Take your time, there's no hurry, every little step takes us closer to our goal. Or maybe not 😆

Faustino-ai, 2021-02-19:
> Take your time, there's no hurry, every little step takes us closer to our goal. Or maybe not 😆

Yea haha. At least I learn something. Mind giving me a direction on what kind of crash course should I do? My programming knowledge is quite limited and almost nothing about cloud. I guess at this point its not too bad to learn from the very basic and strengthen my foundation before tacking this Tuya lock.

rospogrigio, 2021-02-19:
Well I wouldn't know what to suggest, my knowledge is based on my experience acquired in years, I'd say that google would give you better suggestions for sure. What I can tell you is just that API calls are actually just HTTP calls that need to receive certain parameters and then give some responses.
Let me add that I am actually a C++ developer, so I am quite a noobie with python and I myself have to google around most of the times when I must do something with it... good study 😉

Faustino-ai, 2021-02-19:
Oh nice. Sounds like I just need to read more on http calls and mqtt calls

rospogrigio, 2021-02-19:
I know very little (or should I say nothing) about MQTT but I'd say that it's not related to those APIs...

Faustino-ai, 2021-02-19:
Yea my goal is to add support in zigbee2mqtt so I guess in a way should help.

rospogrigio, 2021-02-19:
Yeah ok but that will be the last step... we have to find any way to operate the device before that, then we can concentrate on how to make it available for zigbee2mqtt.

Faustino-ai, 2021-02-19:
> Yeah ok but that will be the last step... we have to find any way to operate the device before that, then we can concentrate on how to make it available for zigbee2mqtt.

Good point. Need to brush up with programming as well.

Faustino-ai, 2021-02-22:
> So, I believe the diagram you posted is taken from: https://developer.tuya.com/en/docs/iot/open-api/api-reference/proprietary-category/smart-door-lock?id=K9jgsgd4cgysr#title-4-Password%20creation [...]

I tried to learn to run it but it seems like more questions just came up. My first question is should I run the api_test in HA? If so what should I use to connect to the lock? I can't seem to figure out how the messages can reach the lock. What I currently have is the cc2531 so I am not sure if it works for https calls. So I am hoping if you can share your setup (what adapter and what you use to run the python script) with me as I am really confused here.

rospogrigio, 2021-02-22:
No, it is supposed to be run via command line. You can execute it directly (but you have to change permissions in order to make it executable probably) or launch it with the command "`python3 ./api_test.py`". I am assuming you are usign linux.
The messages should reach the lock via the Tuya cloud infrastructure since we are using the cloud API calls. Again, I am assuming you are registered in the IOT Tuya site, that you have your API client_id and secret, and that you can see your device among the registered devices. 
Another thing you could try is to install this tool: https://github.com/TuyaAPI/cli
and then launch the command `tuya-cli wizard` . It will ask your API client_id and secret (you can then store these values) and the device_id of one of your devices connected to your account, and will then show you all your devices with their secretKey (very handy!!).
All of this is done via your wifi connection so at the moment cc2531 is not needed. Finally, my setup is just this Airbnk bluetooth smart lock (M510), connected through a Zigbee gateway.
Let me know if you succeed with the above commands, bye!

Faustino-ai, 2021-02-22:
> No, it is supposed to be run via command line. You can execute it directly (but you have to change permissions in order to make it executable probably) or launch it with the command "`python3 [...]

I guess my python version is the main culprit... I am using version 2.7 during the earlier setup due to new versions not supported. Now I know why people hate windows

Faustino-ai, 2021-02-23:
Ok so for postman I am still having some issues with the easy_sign and timestamp. But from the IOT platform, I was able to use the GET command and receive a return. Turns out for my case, even doorbell_volume isn't there. It showed something like "function: [ ] "
However, for "status", I do see a bunch of them including unlock, card_unlock and so on. From my limited understanding, they are storing those function payloads on the cloud itself, making it almost impossible to do anything locally unless we can manage to simulate Tuya cloud API with our own server. Also I forgot to mention, my lock is a residential lock with password, fingerprint and card unlocking option.

Faustino-ai, 2021-02-23:
@rospogrigio 
Ok somehow I managed to run the api_test. What I recevied is very similar to what I got from the Tuya cloud development website. For the "GET status" I saw a bunch of them like mentioned before. However when it gets to "GET... functions", it returns with a message : "not support this device" . I do get additional information like  "remote_unlock_type" = remoteUnlockWithPwd. I do not understand the password ticket type even though it returns true. I will attach my output if it helps.
https://pastebin.com/Pa0fmP2g

On a side note, I do notice that while using postman, they require "sign" and "t", which is the cloud signature and time stamp. Sign is basically encrypted with md5 or something as well as some other stuff including the time stamp itself. So I have no idea how to put this in postman because the value for "t" always change. I tried to browse through your code but couldn't find information on this. And when I asked the Tuya engineers, they told me useless stuffs with slow replies so I am not counting on them. Hopefully this will help.

rospogrigio, 2021-02-24:
Hi, what do you mean by "using postman"?? I think that the fact you get "remoteUnlockWithPwd" is a good sign (for you, since I have an empty reply for that command 😭  ).
The password ticket is the mysterious part. I believe it should be used for the first part of the diagram you posted. I believe the "ticket_key" you get is AES encrypted and should be decrypted with the developer's secret but I couldn't succeed in this. Then the unencrypted key should be used to get the temporary key for unlocking, according to https://developer.tuya.com/en/docs/iot/open-api/api-reference/proprietary-category/smart-door-lock?id=K9jgsgd4cgysr#title-4-Password%20creation but I can't make it work though I did not spend much time on this.

plizer123, 2021-02-25:
What result.
  did you manage to deal with the lock?

Faustino-ai, 2021-02-25:
Ok so here's what I've been gotten so far and I believe I am pretty much at a dead end. Postman is a platform for API development and I was planning to use it to simulate Tuya cloud API to control the lock. The codes generated work almost the same as what you gave me but somehow more specific and update. From it I can see that both the AccessKey and secret are unique for every device (or manufacturer). I guess if we cant obtain those information from our seller then it is almost impossible for us to crack them, well at least for me. Maybe some pro hackers can managed to get their hands on but I am definitely not one of them. Besides that, getting "remoteUnlockWithPwd" is good for me, it is basically under "status" instead of "function", which I believe is only a reporting message instead of the real unlocking function. When I run the code for function, I got the message function not support. And when I asked the Tuya engineers, I got similar answers from when you did, in which they said those functions are being set by the developers (manufacturer, not Tuya themselves, although I believe they do have data on them since it's being developed on their platform). Worse still, they said developers might have a different payload structure from the standard one, which is why we couldn't get the function list at all. From what I understood, Tuya will accomodate those unique payloads from different developers and adapt those on their cloud services. My seller just told me he doesn't know anything about the protocols being used and is unwilling to help me obtain them. 
On the other hand, I managed to find another seller of Tuya lock, and they told me they've successfully use their own server to control the tuya lock, this is however from their own developed lock, which means they are the developer of their own tuya lock, where they have access to all the UART protocols and so on. So my conclusion is that without the developers coorperation and information, it is close to impossible to control a tuya zigbee/bluetooth lock locally. Even the wifi version of it had to be flashed with custom firmware in order to make it work.

Faustino-ai, 2021-02-25:
In the meantime I will try to get in touch with the engineers who sucessfully bypass Tuya cloud and see what information I can get from them. I highly doubt they will disclose anything crucial since this is kind of a bussiness secret

Faustino-ai, 2021-02-25:
> What result.
> did you manage to deal with the lock?

Nothing useful to be honest. I am just trying to share whatever I had and see if someday someone could make any sense out of them because I can't.

cristi9flavors, 2021-03-17:
Any new progress on this?

rospogrigio, 2021-03-17:
Nope... I don't think it is possible to interface with this device unless somebody can obtain the proper encryption keys by the Tuya guys, and they probably won't give them away,

[105 more replies]

closed 2026-10-04 by github-actions[bot]