Koenkk/zigbee2mqtt #4479: Tasmota Zigbee Bridge

hkrob, 2020-09-28:
## Information about the device + link
https://github.com/arendst/Tasmota/issues/8583

Sonoff have a handy Zigbee wifi bridge.
The above project creates an MQTT flow, but it is not easy to process in HA, no rules and such.
It would be great to able to have zigbee2mqtt enrich the messages coming out of the Tasmota Zigbee adaptor.

Apologies if this is not the correct way to request this feature.

## data/database.db entry of the device
..

Koenkk, 2020-09-28:
Zigbee2MQTT and Tasmota Zigbee cannot be used (and are not meant to) together, you either use Tasmota Zigbee or Zigbee2MQTT.

closed 2020-09-28 by Koenkk

hkrob, 2020-09-29:
That's a shame because Zigbee2MQTT contains heaps of rules/hacks for the myriad devices out there...
With Tasmota Zigbee alone, you receive an MQTT stream which is very basic and not very useful in HA.
With Zigbee2MQTT you get a rich environment in HA (at least with my Xiaomi devices over CC2531...)