Koenkk/zigbee2mqtt #4480: Home Assistant auto discovery: null config since 1.14.4

cybe, 2020-09-28:
Upgrading from 1.14.3 to 1.14.4 causes a duplicate config publication, where the second one contains a `null` payload.

## What happened

```
Zigbee2MQTT:info  2020-09-28 18:51:47: wsdcgq11lm_a (0x00158d000233f81f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-09-28 18:51:48: MQTT publish: topic 'hamburg/homeassistant/register/sensor/0x00158d000233f81f/temperature/config', payload '{"availability_topic":"nürnberg/zigbee2mqtt/bridge/state","device":{"identifiers":["zigbee2mqtt_0x00158d000233f81f"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"wsdcgq11lm_a","sw_version":"Zigbee2MQTT 1.14.4"},"device_class":"temperature","json_attributes_topic":"nürnberg/zigbee2mqtt/wsdcgq11lm_a","name":"wsdcgq11lm_a_temperature","state_topic":"nürnberg/zigbee2mqtt/wsdcgq11lm_a","unique_id":"0x00158d000233f81f_temperature_nürnberg/zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:info  2020-09-28 18:51:48: MQTT publish: topic 'hamburg/homeassistant/register/sensor/0x00158d000233f81f/temperature/config', payload 'null'
```
The `null` publication causes Home Assistant to lose the device all together.

## What did you expect to happen

Only one config publication, as in 1.14.3:
```
Zigbee2MQTT:info  2020-09-28 19:00:10: wsdcgq11lm_a (0x00158d000233f81f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2020-09-28 19:00:11: MQTT publish: topic 'hamburg/homeassistant/register/sensor/0x00158d000233f81f/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"nürnberg/zigbee2mqtt/wsdcgq11lm_a","json_attributes_topic":"nürnberg/zigbee2mqtt/wsdcgq11lm_a","name":"wsdcgq11lm_a_temperature","unique_id":"0x00158d000233f81f_temperature_nürnberg/zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000233f81f"],"name":"wsdcgq11lm_a","sw_version":"Zigbee2MQTT 1.14.3","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"nürnberg/zigbee2mqtt/bridge/state"}'
```

## How to reproduce it (minimal and precise)

Switch the Docker image from 1.14.3 to 1.14.4. The behavior can be reversed by downgrading.

## Debug info
Zigbee2MQTT version: 1.14.4
Adapter hardware: CC2531
Adapter firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}`
Configuration: https://gist.github.com/cybe/1a470baf1b4f0dcb2332005b92e397d0

Happy to provide more detailed information.

Koenkk, 2020-09-28:
Since 1.14.4, Zigbee2MQTT will clear outdated configs (e.g. device that has been removed) but this seems to be bugged when using a non-standard Home Assistant discovery topic (`hamburg/homeassistant/register` in your case). 

This should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

cybe, 2020-09-28:
Much appreciated! Will test an report back.

cybe, 2020-09-29:
Issue is confirmed to be resolved. Thanks!

closed 2020-09-29 by cybe