Koenkk/zigbee2mqtt #4485: Aqara wireless relay controller LLKZMK11LM reports interlock true when false after a reset

McGiverGim, 2020-09-29:
## What happened
My zigbee network become unstable. The culprit seemed two Aqara relay (they report an error when trying to get the network map), so I ended rejoining this two nodes. After that, all seemed to work ok, but it seems that rejoining it implied a reset of the devices and the interlock that I have enabled was disabled.
But the zigbee messages in the log showed interlock true.

## What did you expect to happen
If after reset the device the interlock is false, the zigbee messages must show them as false.

## How to reproduce it (minimal and precise)
- Enable interlock
- Reset and rejoin the device
- Look at the logs, the interlock appears as true but it does not work (I can enable/disable both switch at the same time).

## Debug info
Zigbee2MQTT version: 1.14.4.2
Adapter hardware: CC2531
Adapter firmware version: zStack12.20190608

McGiverGim, 2020-10-30:
No response so I suppose that this behavior was expected and not a bug?

Koenkk, 2020-10-30:
Can you provide the debug log when this reset happens?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

McGiverGim, 2020-11-02:
Thanks @Koenkk for looking at this. I have added some "comments" of my actions. I enabled the joining twice by error but for the rest is the same:
- Played with the switches, interlock working
- Enabled debug and join to the network
- Join again the relay
- Play with the switches, interlock is not working but reports true

Here is the log, filtered by the entity `rele_comedor_centro` that is the one who I used for this:

```txt
<!-- HERE I',M PLAYING WITH INTERLOCK ENABLING/DISABLING ONE SWITCH, IT WORKS, IT SWITCHS OFF THE OPPOSITE -->

info  2020-11-02 16:09:33: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":0.02,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:53: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":0.02,"state":"ON","state_l1":"ON","state_l2":"ON","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:53: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":0.02,"state":"ON","state_l1":"OFF","state_l2":"ON","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:53: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":0.02,"state":"ON","state_l1":"OFF","state_l2":"ON","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:56: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":13.48,"state":"ON","state_l1":"OFF","state_l2":"ON","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:58: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":13.48,"state":"ON","state_l1":"ON","state_l2":"ON","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:58: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":13.48,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":33,"voltage":224.3}'
info  2020-11-02 16:09:58: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":13.48,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":33,"voltage":224.3}'
info  2020-11-02 16:10:00: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":107,"power":0.02,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":33,"voltage":224.3}'

<!-- SWITCH TO DEBUG AND PERMIT JOIN -->

info  2020-11-02 16:10:32: Switching log level to 'debug'
info  2020-11-02 16:10:32: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.16.0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.16.0"}'
debug 2020-11-02 16:10:33: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
info  2020-11-02 16:10:33: Zigbee: allowing new devices to join.
info  2020-11-02 16:10:33: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.16.0","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.0"}'

<!-- REJOIN THE NETWORK WITH THE DEVICE -->

warn  2020-11-02 16:10:57: Device '0x00158d0003f453e3' left the network
info  2020-11-02 16:10:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0x00158d0003f453e3"},"type":"device_leave"}'
info  2020-11-02 16:10:57: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":[],"output":["ssIasZone"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b0018e86d21","interview_completed":false,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"definition":{"description":"Aqara human body movement and illuminance sensor","exposes":[{"access":"r","name":"battery","property":"battery","type":"numeric","unit":"%"},{"access":"r","name":"occupancy","property":"occupancy","type":"binary","value_off":false,"value_on":true},{"access":"r","name":"illuminance","property":"illuminance","type":"numeric","unit":"lx"},{"access":"r","name":"linkquality","property":"linkquality","type":"numeric" [... cut ...]
debug 2020-11-02 16:12:14: Received Zigbee message from 'rele_comedor_centro', type 'attributeReport', cluster 'genOnOff', data '{"61440":117440683,"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-02 16:12:14: MQTT publish: topic 'zigbee2mqtt/rele_comedor_centro', payload '{"consumption":0.83,"energy":0.83,"interlock":true,"linkquality":110,"power":0.98,"state":"ON","state_l1":"ON","state_l2":"OFF","temperature":33,"voltage":224.3}'

<!-- AGAIN LOG TO INFO -->

debug 2020-11-02 16:12:26: Received MQTT message on 'zigbee2mqtt/bridge/config/log_level' with data 'info'
info  2020-11-02 16:12:26: Switching log level to 'info'
```

McGiverGim, 2020-11-02:
Thanks for fixing! I'm using the Home Assistant addon, can I go to the edge version and return to the stable without losing data or needing to repair the network?

Koenkk, 2020-11-03:
I don't know, @ciotlosm do you know if this is possible? (I expect yes)

McGiverGim, 2020-11-03:
If not I think we can wait until published as stable. It is not an urgent issue, only happens when repairing.

ciotlosm, 2020-11-03:
@Koenkk @McGiverGim you can switch between `edge` and `stable` at will with no data loss, as long as you SAVE your add-on config and restore it after install of the other one. 

You can also have both installed and just put the same config in both, just make sure that just one is set to start on boot.

closed 2020-12-13 by github-actions[bot]