Koenkk/zigbee2mqtt #4486: Weekly schedule for thermostats json format

Hacker-CB, 2020-09-29:
I'm implementing support for tuya thermostat with custom schedule and custom cluster.

I think that it will be better if json payload for schedule will be same as in `hvacThermostat:getWeeklyScheduleRsp` and `hvacThermostat:setWeeklySchedule` .

So, I found interface:

`interface ThermoTransition {transitionTime: number; heatSetpoint?: number; coolSetpoint?: number}`

And tests:
```
const payloadHeat = {numoftrans:6, dayofweek:64, mode:1, transitions: [{transitionTime:360,heatSetpointitionTime:570,heatSetpoint:2200},{transitionTime:720,heatSetpoint:2150},{transitionTime:840,heatSetpoinansitionTime:990,heatSetpoint:2300},{transitionTime:1380,heatSetpoint:2100}]};
const frameHeat = Zcl.ZclFrame.create(FrameType.SPECIFIC, Direction.CLIENT_TO_SERVER, false, null, 8, chedule', 513, payloadHeat);
expect(frameHeat.toBuffer()).toStrictEqual(([1,8,1,6,64,1,104,1,23,0,58,2,152,8,208,2,102,8,72,3,102,8,222,3,252,8,100,5,52,8]));

const payloadCool = {numoftrans:6, dayofweek:64, mode:2, transitions: [{transitionTime:360,coolSetpointnsitionTime:570,coolSetpoint:2200},{transitionTime:720,coolSetpoint:2150},{transitionTime:840,coolSetpotransitionTime:990,coolSetpoint:2300},{transitionTime:1380,coolSetpoint:2100}]};
const frameCool = Zcl.ZclFrame.create(FrameType.SPECIFIC, Direction.CLIENT_TO_SERVER, false, null, 8, chedule', 513, payloadCool);
expect(frameCool.toBuffer()).toStrictEqual(([1,8,1,6,64,2,104,1,252,8,58,2,152,8,208,2,102,8,72,3,102,8,222,3,252,8,100,5,52,8]));

const payloadHeatAndCool = {numoftrans:6, dayofweek:64, mode:2, transitions: nTime:360,coolSetpoint:570, heatSetpoint: 2300}]};
const frameHeatAndCool = Zcl.ZclFrame.create(FrameType.SPECIFIC, Direction.CLIENT_TO_SERVER, false, etWeeklySchedule', 513, payloadHeatAndCool);
expect(frameHeatAndCool.toBuffer()).toStrictEqual(Buffer.from([1,8,1,6,64,2,104,1,252,8,58,2]));
```

Seems that format should be the next:

```json
{
  "weekly_schedule": {
    "1": {
      "dayofweek": 1,
      "numoftrans": 2,
      "mode": 1,
      "transitions": [
        {
          "transitionTime": 360,
          "heatSetpoint": 23
        },
        {
          "transitionTime": 570,
          "heatSetpoint": 2200
        }
      ]
    }
  }
}
```

I have some questions:

1. In which units is `transitionTime`? Is it minutes since midnight?
2. What is `dayofweek:64`?  Which format of the field (int or string)?
3. What means `mode:1`?
4. In the first transistion (which I found in tests) `heatSetpoint` is `23`. I'm a little confused. So, it means `0.23°C`, right?
5. `dayofweek` - first day is Sunday, right?
5. For example, If user writes to topic `deviceId/set/weekly_schedule` with key `2`, this should overwrite schedule only for day 2, all other days should leave unchanged?

@Koenkk could you point me to right way please.

Hacker-CB, 2020-09-29:
7. Seems that heatSetpoint should be in float format in json. `2201` -> `22.01`?

Hacker-CB, 2020-09-29:
3. I found from TI clusters datasheet:

[image]

Koenkk, 2020-09-29:
Most of it is described in the ZCL: https://github.com/Koenkk/zigbee-herdsman/blob/master/docs/Zigbee%20Cluster%20Library%20Specification%20v7.pdf

1. see zcl 6.3.2.3.2.5
2. uint8 (https://github.com/Koenkk/zigbee-herdsman/blob/eca0a5198077162aa05575bd88e16c162d588604/src/zcl/definition/cluster.ts#L1905), so a number from javascript.
4. the stepsize is 0.01 indeed, so 2300 = 23c
5. yes, see zcl 6.3.2.3.2.3
6. yes
7. yes

Hacker-CB, 2020-09-29:
Thank you for the respones. I already submitted PR.
Seems that all clear now.

Final JSON which can be readed/written to ETOP-08 thermostat:

```json
{
  "weekly_schedule": {
    "1": {
      "dayofweek": 1,
      "mode": 1,
      "numoftrans": 4,
      "transitions": [
        {
          "heatSetpoint": "11.5",
          "transitionTime": 0
        },
        {
          "heatSetpoint": "9.0",
          "transitionTime": 420
        },
        {
          "heatSetpoint": "25.0",
          "transitionTime": 600
        },
        {
          "heatSetpoint": "25.0",
          "transitionTime": 600
        }
      ]
    },
    
  }
}
```

closed 2020-09-29 by Hacker-CB