Koenkk/zigbee2mqtt #4490: Moes HY369RT (AKA TuYa TS0601) Unavailable In Home Assistant

danieledwardgeorgehitchcock, 2020-09-29:
## What happened
Added the TRV into Zigbee2MQTT - no errors or warnings. Can see states appear in Zigbee2MQTT topics. All entities in Home Assistant are showing as unavailable.

I have read here #4460 that the firmware needs to be updated on the device so, I unbound the device, added to the Tuya hub, and upgraded to the latest available.

## What did you expect to happen
Climate and sensor states to be available in Home Assistant

## How to reproduce it (minimal and precise)
Add Moes HY369RT to Zigbee2MQTT with Home Assistant enabled and Home Assistant configured to automatically add MQTT entries

Attempt to control auto discovered device via Home Assistant

## Debug info
Zigbee2MQTT version: 1.14.4 (dev)
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200805.zip

Koenkk, 2020-09-29:
Do you use the availability feature? https://www.zigbee2mqtt.io/information/availability.html

danieledwardgeorgehitchcock, 2020-09-29:
Hi,

Thanks for your reply. I have managed to fix my issue now.

I had to pull the latest homeassistant.js file and map it to my container.

closed 2020-09-29 by danieledwardgeorgehitchcock