Koenkk/zigbee2mqtt #4493: SNZB-03 is not supported

dedors, 2020-09-29:
## Information about the device + link
While trying to join a Sonoff SNZB-03 (Motion Sensor - already supported), I get following:

```
Zigbee2MQTT:info  2020-09-29 19:00:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b002228b7e5"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-09-29 19:00:20: Successfully interviewed '0x00124b002228b7e5', device has successfully been paired
Zigbee2MQTT:warn  2020-09-29 19:00:20: Device '0x00124b002228b7e5' with Zigbee model 'MSO1' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Current version: 1.14.4.2

## data/database.db entry of the device
`{"id":2,"type":"EndDevice","ieeeAddr":"0x00124b002228b7e5","nwkAddr":2467,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"MSO1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"MSO1","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":3,"hwVersion":1,"dateCode":"20191107"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001ca60eb2","zoneState":1}}},"binds":[]}},"appVersion":3,"hwVersion":1,"dateCode":"20191107","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1601398820687}`

Koenkk, 2020-09-29:
This variant is already supported in the latest dev branch.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-09-29 by Koenkk

MegoVaTx, 2023-12-04:
> This variant is already supported in the latest dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Friendly name
0xa4c1380c6f1d632f
Description
Last seen
N/A
Availability

[Disabled](https://www.zigbee2mqtt.io/guide/configuration/device-availability.html#availability-advanced-configuration)
Device type
EndDevice
Zigbee Model
SNZB-03
Zigbee Manufacturer
eWeLink
Support status
Not supported

IEEE Address
0xa4c1380c6f1d632f
Network address
0xD5EF
Firmware version
0122052017
Power
Battery 
Interview completed
True

Zigbee2MQTT version
1.29.2-dev commit: [82fd062](https://github.com/Koenkk/zigbee2mqtt/commit/82fd062)
Coordinator type
EZSP v8
Coordinator revision
6.7.8.0 build 373
Coordinator IEEE Address
0x000d6f001740c949
Frontend version
0.6.124

Im using Zigbee2MQTT Edge and its staing that's its unsupported even though it suppose to be.

Received message from unsupported device with Zigbee model 'SNZB-03' and manufacturer name 'eWeLink'
Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html