Koenkk/zigbee2mqtt #4494: Zemismart ZM25TQ new device

MSITETOP, 2020-09-29:
Good afternoon! 
I purchased a Zemismart ZM25TQ. 
Connected but got an error when pairing 
`NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html` 
as a result, I managed to perform the pairing: I have modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'. I added it to the config
`{
        zigbeeModel: ['owvfni3'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
        ],
        model: 'mcdj3aq',
        vendor: 'TuYa',
        description: 'Tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
}`
After that, the device was added and even managed (open, close, stop) but the position always returns 100. it turns without stopping, how can this be corrected?

`{"type":"commandGetData","device":{"ID":2,"_type":"Router","_ieeeAddr":"0xbc33acfffe457ad8","_networkAddress":50941,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,10,4,5,61184],"outputClusters":[25],"deviceNetworkAddress":50941,"deviceIeeeAddress":"0xbc33acfffe457ad8","clusters":{"genBasic":{"attributes":{"65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TZE200_wmcdj3aq","_powerSource":"Mains (single phase)","_modelID":"TS0601","_applicationVersion":82,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"meta":{},"_lastSeen":1601399433685},"endpoint":{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,10,4,5,61184],"outputClusters":[25],"deviceNetworkAddress":50941,"deviceIeeeAddress":"0xbc33acfffe457ad8","clusters":{"genBasic":{"attributes":{"65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[],"meta":{}},"data":{"status":0,"transid":138,"dp":1025,"fn":0,"data":{"type":"Buffer","data":[2]}},"linkquality":13,"groupID":0,"cluster":"manuSpecificTuyaDimmer","meta":{"zclTransactionSequenceNumber":2,"manufacturerCode":null,"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0}}}`

`{"type":"commandSetDataResponse","device":{"ID":2,"_type":"Router","_ieeeAddr":"0xbc33acfffe457ad8","_networkAddress":50941,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,10,4,5,61184],"outputClusters":[25],"deviceNetworkAddress":50941,"deviceIeeeAddress":"0xbc33acfffe457ad8","clusters":{"genBasic":{"attributes":{"65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[],"meta":{}}],"_manufacturerName":"_TZE200_wmcdj3aq","_powerSource":"Mains (single phase)","_modelID":"TS0601","_applicationVersion":82,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"meta":{},"_lastSeen":1601399436997},"endpoint":{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,10,4,5,61184],"outputClusters":[25],"deviceNetworkAddress":50941,"deviceIeeeAddress":"0xbc33acfffe457ad8","clusters":{"genBasic":{"attributes":{"65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[],"meta":{}},"data":{"status":0,"transid":1,"dp":515,"fn":0,"data":{"type":"Buffer","data":[0,0,0,0]}},"linkquality":18,"groupID":0,"cluster":"manuSpecificTuyaDimmer","meta":{"zclTransactionSequenceNumber":3,"manufacturerCode":null,"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0}}}`

Dylantje, 2020-09-30:
Is already working.. only i have a other problem perhaps you a solution?
[](https://github.com/Koenkk/zigbee2mqtt/issues/4465)

MSITETOP, 2020-10-01:
works for another manufacturerName I have '_TZE200_wmcdj3aq' which Was not previously available.

I can't help you with your problem, since I have only one motor

nogueirasa, 2020-10-01:
> works for another manufacturerName I have '_TZE200_wmcdj3aq' which Was not previously available.
> 
> I can't help you with your problem, since I have only one motor

Trying to connect mine ZM25TQ but the ID is different, says:"modelId":"TS0601","manufacturerName":"_TZE200_wmcdj3aq" on database and "Received message from unsupported device with Zigbee model 'TS0601'," on log.

How you did it?

MSITETOP, 2020-10-01:
I searched on google zm25tq modelId: TS0601 and came across this repository https://github.com/iquix/Smartthings/blob/master/devicetypes/iquix/tuya-window-shade.src/tuya-window-shade.groovy

found a wider list of models in it, one of them turned out to be mine

> 		fingerprint profileId: "0104", inClusters: "0000, 000A, 0004, 0005, 00EF", outClusters: "0019", manufacturer: "_TZE200_cowvfni3", model: "TS0601", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 000A, 0004, 0005, 00EF", outClusters: "0019", manufacturer: "**_TZE200_wmcdj3aq**", model: "TS0601", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 000A, 0004, 0005, 00EF", outClusters: "0019", manufacturer: "_TZE200_nogaemzt", model: "TS0601", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 000A, 0004, 0005, 00EF", outClusters: "0019", manufacturer: "_TZE200_5zbp6j0u", model: "TS0601", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 0003, 0004, 0005, 0006", outClusters: "0019", manufacturer: "_TYST11_cowvfni3", model: "owvfni3", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 0003, 0004, 0005, 0006", outClusters: "0019", manufacturer: "_TYST11_wmcdj3aq", model: "mcdj3aq", deviceJoinName: "Tuya Window Treatment"
> 		fingerprint profileId: "0104", inClusters: "0000, 0003, 0004, 0005, 0006", outClusters: "0019", manufacturer: "_TYST11_fdtjuw7u", model: "dtjuw7u", deviceJoinName: "Tuya Window Treatment"

Dylantje, 2020-10-01:
Perhaps that be the problem with my three shutters.. Samen name only diffirent id or something..

nogueirasa, 2020-10-01:
@MSITETOP , woring great. Thanks.

Just add:
fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
],

nogueirasa, 2020-10-01:
@Dylantje works fine with the first one, when I try to configure de second one I have the same problem as you

nogueirasa, 2020-10-01:
Worked for 2 motors. Copied the device.js configuration twice:

    {
        zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja', 'u1rkty3'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_xuzcvlku'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_nkoabg8w'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
        ],
        model: 'TS0601_curtain',
        vendor: 'TuYa',
        description: 'Curtain motor',
        whiteLabel: [
            {vendor: 'Yushun', model: 'YS-MT750'},
            {vendor: 'Zemismart', model: 'ZM79E-DT'},
            {vendor: 'Binthen', model: 'BCM100D'},
            {vendor: 'Binthen', model: 'CV01A'},
            {vendor: 'Zemismart', model: 'M515EGB'},
            {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },

and then again:

    {
        zigbeeModel: ['mcdj3aq', 'mcdj3aq\u0000'],
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
        ],
        model: 'mcdj3aq',
        vendor: 'TuYa',
        description: 'Tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
    },

Dylantje, 2020-10-02:
only in Ha? Or is this in a z2m place

MSITETOP, 2020-10-02:
I think it should work with z2m

Dylantje, 2020-10-02:
Please tell me what the do..
Or where i need to place the copyed data

nogueirasa, 2020-10-07:
Hello @Dylantje ,
Working on 1.15.0 with no problem.

Do you aready now how to edit "device.js"?

If not, look this link: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

If yes, just edit this device including the line in bold:

```
{
    zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja', 'u1rkty3'],
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_xuzcvlku'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_nkoabg8w'},
        _**{modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}**_
    ],
    model: 'TS0601_curtain',
    vendor: 'TuYa',
    description: 'Curtain motor',
    whiteLabel: [
        {vendor: 'Yushun', model: 'YS-MT750'},
        {vendor: 'Zemismart', model: 'ZM79E-DT'},
        {vendor: 'Binthen', model: 'BCM100D'},
        {vendor: 'Binthen', model: 'CV01A'},
        {vendor: 'Zemismart', model: 'M515EGB'},
        {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
    ],
    supports: 'open, close, stop, position',
    fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
    toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
},
```

Koenkk, 2020-10-07:
@nogueirasa added the new fingerprint. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Dylantje, 2020-10-07:
so when i update[ after the a few hours] . all the device in this database should working? no matter what id or name?

Koenkk, 2020-10-07:
@Dylantje I don't have a list of all your device in this database, but the entry of https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-704832418 should work.

Dylantje, 2020-10-07:
> @Dylantje I don't have a list of all your device in this database, but the entry of [#4494 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-704832418) should work.

I do not have a list the datbase you says, when that is working GREAT!!!

Thanks for the work again .. Love you Dutchmen..xxx

garagha, 2020-10-07:
hello. thx for you works!
there is a strange thing: the curtain goes down. arrives. 100% close status. and then the 0% status immediately comes. what could it be ?

Zigbee2MQTT:debug 2020-10-07 22:12:45: Received Zigbee message from '0xbc33acfffe4a09d1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":**[0,0,0,0]**,"type":"Buffer"},"dp":515,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-07 22:12:45: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe4a09d1', payload '{"linkquality":21,"**position":100**,"running":false}'
Zigbee2MQTT:debug 2020-10-07 22:12:45: Received Zigbee message from '0xbc33acfffe4a09d1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":**[0,0,0,100**],"type":"Buffer"},"dp":515,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-07 22:12:45: MQTT publish: topic 'zigbee2mqtt/0xbc33acfffe4a09d1', payload '{"linkquality":21,"position":**0**,"running":false}'

nogueirasa, 2020-10-07:
> hello. thx for you works!
> there is a strange thing: the curtain goes down. arrives. 100% close status. and then the 0% status immediately comes. what could it be ?
>
> Zigbee2MQTT:debug 2020-10-07 [...]

Hello @garagha ,
Did you configured the opened (100%) and closed (0%) position?

garagha, 2020-10-07:
configured  with 433 pult.

garagha, 2020-10-08:
but 100% closed. and 0% open, if it matters

nogueirasa, 2020-10-08:
Doesn't matters

Dylantje, 2020-10-08:
@Koenkk looks there is no thing changed on git?
When i try to get the other 2 shutters to learn they are not on teh list

So is the rebuild of z2m with the new database for the other shutters  finisht and ok?

Ps did you get your beers on paypal?

Koenkk, 2020-10-08:
@Dylantje can you share the data/database.db of those 2 non-working shutters?

I got my beers, big thanks :smile:

Dylantje, 2020-10-08:
mmm my nick is domb71 so i do not have a db is was thinking this part was to be added to your database:
https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-701935438
Or:
https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-704832418

So i do not have other data.. only 2 shutter from the same manufactury only they will not learn and nogueirasa has a solution

So i was thinking when you this part to z2m they will work :-)
Sorrie of zo ik ben niet de slimste hahaha

Koenkk, 2020-10-08:
@Dylantje did you switch to the dev branch? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html (the change is not in a release version yet).

Dylantje, 2020-10-08:
> @Dylantje did you switch to the dev branch? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html (the change is not in a release version yet).

aha thats why my nick is dombo71

is there no problem switching?
And everthing in the cli to do?

Koenkk, 2020-10-08:
It's safe to switch, make sure to carefully follow the instructions.

Dylantje, 2020-10-08:
> It's safe to switch, make sure to carefully follow the instructions.

after that i can go later back to master?

Koenkk, 2020-10-08:
You have to wait until the next release (1 november)

Dylantje, 2020-10-11:
@Koenkk  I did switch to the dev branche.
Did a git pull.
Only cannot learn the 2 other rollers.
Now i look into the file that is @nogueirasa let me see

And i see only 1 of the shutters..
[image]

So wht is the best way to  get the shutters into my system?
Can i remove the devices file and do new gitpul?
Or do i set the new devices into the fil [ like a risk for errors to me? ]

Or can i better remove the whole map zigbee herdsman converters?
And do i need to do more?

nogueirasa, 2020-10-11:
> @Koenkk I did switch to the dev branche.
> Did a git pull.
> Only cannot learn the 2 other rollers.
> Now i look into the file that is @nogueirasa let me see
>
> And i see only 1 of the shutters..
> [image] [...]

Hello @Dylantje ,
Almost there, but don´t change this device, dosen't works for me, looks for that one:

`{
    zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja', 'u1rkty3'],
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_xuzcvlku'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_nkoabg8w'},
    ],
    model: 'TS0601_curtain',
    vendor: 'TuYa',
    description: 'Curtain motor',
    whiteLabel: [
        {vendor: 'Yushun', model: 'YS-MT750'},
        {vendor: 'Zemismart', model: 'ZM79E-DT'},
        {vendor: 'Binthen', model: 'BCM100D'},
        {vendor: 'Binthen', model: 'CV01A'},
        {vendor: 'Zemismart', model: 'M515EGB'},
        {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
    ],
    supports: 'open, close, stop, position',
    fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
    toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
},`

And insert: "        {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}"

At teh end will looks like this:

`{
    zigbeeModel: ['owvfni3\u0000', 'owvfni3', 'aabybja', 'u1rkty3'],
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_5zbp6j0u'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_xuzcvlku'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_nkoabg8w'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_wmcdj3aq'}
    ],
    model: 'TS0601_curtain',
    vendor: 'TuYa',
    description: 'Curtain motor',
    whiteLabel: [
        {vendor: 'Yushun', model: 'YS-MT750'},
        {vendor: 'Zemismart', model: 'ZM79E-DT'},
        {vendor: 'Binthen', model: 'BCM100D'},
        {vendor: 'Binthen', model: 'CV01A'},
        {vendor: 'Zemismart', model: 'M515EGB'},
        {vendor: 'Tuya', model: 'DT82LEMA-1.2N'},
    ],
    supports: 'open, close, stop, position',
    fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
    toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
},`

Koenkk, 2020-10-14:
@Dylantje it's in the latest dev branch. But maybe your device is slightly different. Please share the `data/database.db` entry of that device.

Dylantje, 2020-10-14:
> @Dylantje it's in the latest dev branch. But maybe your device is slightly different. Please share the `data/database.db` entry of that device.

I typet this part..
Looks my file is different between the file in git:
https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-706745948
I will not be overwritten looks..

I can see teh device in the database because they are not learned to z2m.

Or where do i have to look?

Koenkk, 2020-10-14:
@Dylantje does anything appear in the Zigbee2MQTT log when you factory reset the device while having joining enabled?

Dylantje, 2020-11-01:
hi.. did the upgrade to 1.16  only still no options to learn the other 2 rollershutters.. No thing to see in the log of z2m. And only 1 of the 3 is working.
Inthe devices document there is still 1 tya shuttter: `      exposes: [e.switch(), e.power(), e.current(), e.voltage(), e.energy()],
    },
    {
        zigbeeModel: ['mcdj3aq', 'mcdj3aq\u0000'],
        model: 'mcdj3aq',
        vendor: 'TuYa',
        description: 'Tubular motor',
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZM25TQ'},
        ],
        supports: 'open, close, stop, position',
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options],
        exposes: [e.cover_position()],
    },
    {`

Other options?

Koenkk, 2020-11-02:
@Dylantje if I understand correctly, the problem is not that the device shows as unsupported but that you cannot pair the device at all. For this check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair and also try to pair the device very close to the coordinator.

Dylantje, 2020-11-02:
@Koenkk No the problem is thta in devices.js there is only 1 of the three shutters.
And i did the upgrade to 1.16 and the new devices are not in the devices.js document

Koenkk, 2020-11-03:
@Dylantje 

- What did you expect to see added in devices.js?
- can provide the `data/database.db` entry of these devices?

Dylantje, 2020-11-04:
Koen, perhaps we talk in dutch with dm or something?

I was hoping that the other brands/type from this rollershutter where in to the devices.js from the new version of z2m.
The problem is i see no devices into z2m when i try to learn the other brands/type
I do not now what to see in the devices.js only  in this topic is see other devices,js where other did places some more brand/types. I was thinking in the new version all the rollers where into z2m.

Koenkk, 2020-11-04:
@Dylantje wat krijg je te zien als je een device probeert te pairen? (zie je dan uberhaupt iets? zo niet kan je dan het device proberen te pairen dichtbij de coordinator?)

Dylantje, 2020-11-14:
@Koenkk 
[image]
There is stil only 1 shutter.
When i try to pair i so no thing @all.
And when i look in to the device.js there is also no other tuya shutter like the onse here in this topic.
[](https://github.com/Koenkk/zigbee2mqtt/issues/4494#issuecomment-706767998
)
Or are these other device not into the device.js file needed?

Koenkk, 2020-11-14:
@Dylantje devices not pairing is not related to `devices.js`. TuYa devices are sometimes hard to pair indeed, sometimes it helps to move the really close the the coordinator and pair there. Also check: https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

Dylantje, 2020-11-16:
Hi... Did everything you say and what is write down in the wiki..
Only the new flashing option i did not [ beetje bang dat niets meer doet ]

[image]
Could this be the problem?

Perhaps i can buy a other stick with the newest firmware to test? What is the best type @ the moment? read also there new brands that are working fine?

Or are there other options?

Dylantje, 2020-11-18:
Hi...
Thanks, 

is it possible to buy  new stick and then use the new one?
And save this one for just in case?

Is there a wiki how to use new stick with old paired devices?

Koenkk, 2020-11-18:
For zstack 1.2 you can just swap the sticks (no need to repair)

Dylantje, 2020-11-18:
Do you have a advice for the best stick that is working with z2m at the  moment?

Koenkk, 2020-11-18:
Check: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter (zig-a-zig-ah! (zzh!)” adapter by Electrolama OR the “CC2652RB stick” by slaesh)

closed 2020-12-26 by github-actions[bot]