Koenkk/zigbee2mqtt #4495: One Ikea button automatically binds to GroupID 0, along with other devices

d-edge82, 2020-09-29:
First of all thank you for this awesome work.

## What happened

When I attach an ikea EE1810, it gets a groupID 0, which does not exist in the configuration.
Then I find all devices that were automatically added to the same group, and therefore the new remote controls it.

This makes a remote that toggles two curtains, and three lights in the middle of the house

## What did you expect to happen

The button to be attached, without being added to a group - so I can control what it does.

## How to reproduce it (minimal and precise)

I wish I knew

## Debug info
Zigbee2MQTT version:
Adapter hardware: Conbee II
Adapter firmware version:

log:

info  2020-09-29 21:11:57: Switching log level to 'debug'
info  2020-09-29 21:11:57: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"1.14.4","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2"},"log_level":"debug","network":{"channel":20,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.14.4"}'
debug 2020-09-29 21:11:58: Received Zigbee message from 'Noamroom Remote 2', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-09-29 21:11:58: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote 2', payload '{"action":"toggle","battery":74,"linkquality":141,"update_available":false}'
info  2020-09-29 21:11:58: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote 2', payload '{"action":"","battery":74,"linkquality":141,"update_available":false}'
info  2020-09-29 21:11:58: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote 2/action', payload 'toggle'
debug 2020-09-29 21:11:59: Received Zigbee message from '21 Lights 17', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-09-29 21:11:59: MQTT publish: topic 'zigbee2mqtt/21 Lights 17', payload '{"brightness":0,"linkquality":141,"state":"OFF","update_available":false}'
debug 2020-09-29 21:11:59: Received Zigbee message from '21 Lights 18', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2020-09-29 21:11:59: MQTT publish: topic 'zigbee2mqtt/21 Lights 18', payload '{"brightness":0,"linkquality":141,"state":"OFF","update_available":false}'
debug 2020-09-29 21:11:59: Received Zigbee message from '21 Lights 18', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2020-09-29 21:11:59: MQTT publish: topic 'zigbee2mqtt/21 Lights 18', payload '{"brightness":0,"linkquality":0,"state":"OFF","update_available":false}'
debug 2020-09-29 21:11:59: Received Zigbee message from '21 Lights 19', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2020-09-29 21:11:59: MQTT publish: topic 'zigbee2mqtt/21 Lights 19', payload '{"brightness":0,"linkquality":141,"state":"OFF","update_available":false}'
debug 2020-09-29 21:11:59: Received Zigbee message from 'Living Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID null

Koenkk, 2020-09-29:
Can you try to force configure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , make sure to wakeup the device right before doing so.

d-edge82, 2020-09-30:
Thanks for the support @Koenkk 

Unfortunately, same result after the force configuration:

info  2020-09-30 16:52:07: Configuring 'Noamroom Remote'
debug 2020-09-30 16:52:23: Received Zigbee message from 'Living Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"3":47,"5":11,"7":"0x0000000000000000","8":4384,"9":4615}}' from endpoint 1 with groupID null
debug 2020-09-30 16:52:35: Device 'Noamroom Remote' announced itself
info  2020-09-30 16:52:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Noamroom Remote"},"type":"device_announced"}'
debug 2020-09-30 16:52:38: Received Zigbee message from 'Noamroom Remote', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote', payload '{"action":"toggle","battery":74,"linkquality":233,"update_available":false}'
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote', payload '{"action":"","battery":74,"linkquality":233,"update_available":false}'
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote/action', payload 'toggle'
debug 2020-09-30 16:52:38: Received Zigbee message from '21 Lights 17', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/z2m_kitchen_light', payload '{"brightness":0,"state":"ON"}'
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/21 Lights 17', payload '{"brightness":0,"linkquality":233,"state":"ON","update_available":false}'
debug 2020-09-30 16:52:38: Received Zigbee message from '21 Lights 17', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/z2m_kitchen_light', payload '{"brightness":0,"state":"ON"}'
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/21 Lights 17', payload '{"brightness":0,"linkquality":233,"state":"ON","update_available":false}'
debug 2020-09-30 16:52:38: Received Zigbee message from 'Top Kitchen SmartPlug', type 'attributeReport', cluster 'genOnOff', data '{"61440":121511469,"onOff":0}' from endpoint 1 with groupID null
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/Top Kitchen SmartPlug', payload '{"consumption":0.86,"linkquality":233,"power":0,"state":"OFF","temperature":45}'
debug 2020-09-30 16:52:38: Received Zigbee message from 'Bedroom curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID null
debug 2020-09-30 16:52:38: Received Zigbee message from 'Bedroom curtain', type 'attributeReport', cluster 'genAnalogOutput', data '{"61440":54457600,"presentValue":1}' from endpoint 1 with groupID null
info  2020-09-30 16:52:38: MQTT publish: topic 'zigbee2mqtt/Bedroom curtain', payload '{"linkquality":233,"position":1,"running":true}'
debug 2020-09-30 16:52:38: Received Zigbee message from 'Living Room Curtain', type 'attributeReport', cluster 'genBasic', data '{"1028":1}' from endpoint 1 with groupID null
debug 2020-09-30 16:52:42: Received Zigbee message from 'Noamroom Remote', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2020-09-30 16:52:42: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote', payload '{"action":"toggle","battery":74,"linkquality":233,"update_available":false}'
info  2020-09-30 16:52:42: MQTT publish: topic 'zigbee2mqtt/Noamroom Remote', payload '{"action":"","battery":74,"linkquality":233,"update_available":false}'

Yes, now it also controls a freshly installed xiaomi smart plug.. 😆

d-edge82, 2020-09-30:
OK just found another interesting phenomenon:

The remote keeps controlling those devices when HA host is off..
I guess it's was bind to those devices somehow.

Do you think this still can be z2m integration related?

Koenkk, 2020-09-30:
@d-edge82 it seems configure doesn't complete (you should see either `Failed to configure..` or `Successfully configured ..` in the log.

- Right before `info 2020-09-30 16:52:07: Configuring 'Noamroom Remote'` I don't see any commands from the device, I expect this because you need to wakeup the device by clicking buttons on it like I mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/4495#issuecomment-700954038
- Could be that this is a bug in the deconz integration, try updating to the latest dev, reconfigure and see if it solves things: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

d-edge82, 2020-09-30:
I think this is what I am witnessing:

https://github.com/Koenkk/zigbee2mqtt/issues/2325
and this:
https://community.home-assistant.io/t/please-help-automation-for-ikea-tradfri-remote-control-e1524-e1810/142707/20

I will try to force config it again and update.
Thanks

d-edge82, 2020-09-30:
> @d-edge82 it seems configure doesn't complete (you should see either `Failed to configure..` or `Successfully configured ..` in the log.
> 
> * Right before `info 2020-09-30 16:52:07: Configuring 'Noamroom Remote'` I don't see any commands from the device, I expect this because you need to wakeup the device by clicking buttons on it like I mentioned in [#4495 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4495#issuecomment-700954038)
> * Could be that this is a bug in the deconz integration, try updating to the latest dev, reconfigure and see if it solves things: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Update fails:
error 2020-09-30 20:14:43: Failed to configure 'Noamroom Remote 2', attempt 1 (TypeError: Cannot read property 'message' of undefined

I am afraid to update the conbee to the latest, as I have noticed it is having issues adding devices afterwards
any chance to resolve this without updating?

Koenkk, 2020-09-30:
Can you update to the latest dev, I know there are some conbee fixes in there.

d-edge82, 2020-10-01:
> Can you update to the latest dev, I know there are some conbee fixes in there.

You mean to use the edge version of Z2M?
Already tried that, same results.

Koenkk, 2020-10-02:
In that case please provide the herdsman debug logging of the configure.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

d-edge82, 2020-10-02:
I think I enabled the herdsmen debug logging now.

I tried once to manually config the remote after a restart and it worked, it got configured and worked properly.
When I tried to reset it next to it's place (not near the coordinator) - it started controlling different devices in my house again.
I suspect this has to do with the "Group 0" thing.

Here is the log, thanks for the support!

info  2020-10-03 02:13:46: Configuring 'Noamroom Remote 2'
debug 2020-10-03 02:13:47: Received Zigbee message from 'Top Kitchen SmartPlug', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"149":0.8635027408599854,"152":0,"154":0,"3":41,"5":23,"7":"0x0000000000000000","8":4898,"9":3074}}' from endpoint 1 with groupID null
info  2020-10-03 02:13:47: MQTT publish: topic 'zigbee2mqtt/Top Kitchen SmartPlug', payload '{"consumption":0.86,"linkquality":238,"power":0,"state":"OFF","temperature":41}'
debug 2020-10-03 02:13:51: Received Zigbee message from 'Noam Smart Plug', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID null
debug 2020-10-03 02:13:52: Received Zigbee message from 'Top Kitchen SmartPlug', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID null
debug 2020-10-03 02:14:04: Received Zigbee message from 'Playroom Desk Lamp', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2020-10-03 02:14:04: MQTT publish: topic 'zigbee2mqtt/Playroom Desk Lamp', payload '{"brightness":254,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2020-10-03 02:14:11: Received Zigbee message from 'Parents Bath Mirror lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
info  2020-10-03 02:14:11: MQTT publish: topic 'zigbee2mqtt/Parents Bath Mirror lamp', payload '{"brightness":254,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2020-10-03 02:14:15: Received Zigbee message from 'Playroom Desk Lamp', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
info  2020-10-03 02:14:15: MQTT publish: topic 'zigbee2mqtt/Playroom Desk Lamp', payload '{"brightness":254,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2020-10-03 02:14:38: Received Zigbee message from 'Bedroom curtain', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":100,"3":33,"5":8,"7":"0x0000000000000000","8":4384,"9":4871}}' from endpoint 1 with groupID null
debug 2020-10-03 02:14:41: Received Zigbee message from 'Under Stairs Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2020-10-03 02:14:41: MQTT publish: topic 'zigbee2mqtt/Under Stairs Light', payload '{"brightness":254,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2020-10-03 02:14:42: Received Zigbee message from 'Noam Smart Plug', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID null
debug 2020-10-03 02:14:43: Received Zigbee message from 'Top Kitchen SmartPlug', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID null
debug 2020-10-03 02:14:44: Received Zigbee message from 'Under Stairs Light', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID null
info  2020-10-03 02:14:44: MQTT publish: topic 'zigbee2mqtt/Under Stairs Light', payload '{"brightness":254,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'
error 2020-10-03 02:14:46: Failed to configure 'Noamroom Remote 2', attempt 2 (Error: Bind 0x680ae2fffe81fbb7/1 genOnOff from '901' failed (waiting for response TIMEOUT)
    at DeconzAdapter.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:656:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65))
debug 2020-10-03 02:14:53: Received Zigbee message from 'Amud Light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
info  2020-10-03 02:14:53: MQTT publish: topic 'zigbee2mqtt/Amud Light', payload '{"brightness":120,"linkquality":238,"state":"OFF","update":{"state":"idle"},"update_available":false}'

d-edge82, 2020-10-02:
I kept trying to configure the remote, while checking the logs I see this:

debug 2020-10-03 02:27:05: Received Zigbee message from 'Noamroom Remote 2', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":0}' from endpoint 1 with groupID 0

This came right after I clicked the middle button to wake up the device.
I think the "Group 0" is the issue..

Any chance to clear that setting?

Koenkk, 2020-10-03:
Group 0 is indeed the issue, which will be fixed when the configure succeeds. I still don't see commands from the remote in your log indicating the device woke up (point 1 of https://github.com/Koenkk/zigbee2mqtt/issues/4495#issuecomment-701439306). Make sure to click a button right before configuring and keep clicking every second.

d-edge82, 2020-10-04:
Thanks @Koenkk, 
1. I managed to properly configure the remote, after it was done the remote didn't respond
I had to re-pair it to the system and again it started controlling the other devices..
Is this an error with my remote of have I encountered a bug?

I would like to have the option to reset the remote without it losing it's connection everytime.

2. any tip on how to enable herdsman logging? I tried adding it to the addon configuration and restart the addon but nothing has changed.

d-edge82, 2020-10-04:
After the force setup the remote now works.

It gets messed up again if I reset it, so it will be great if you can check
But I don't have the issue anymore, so no further support is needed.

Thank you!

closed 2020-10-04 by d-edge82