Koenkk/zigbee2mqtt #4496: Errors when binding Trådfri switches to groups on ZZH

esbenr, 2020-09-29:
## What happened
Trying to bind Trådfri switches to Z2M groups on ZZH. The bindings that succeed seem to "fall" off after Z2M restart.
Having major issues with binding too.

## What did you expect to happen
I have had the same groups and devices with CC2531. There binding worked fine and did not "dissapear".

## How to reproduce it (minimal and precise)
Create a group with multiple Trådfri single color bulbs.
Try to pair a Trådfri switch with the group.
Z2M logs error:
```
Zigbee2MQTT:info  2020-09-29 22:38:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"Living room Switch above stairs top","to":"light_living_room"},"type":"device_bind"}'
Zigbee2MQTT:error 2020-09-29 22:38:45: Failed to bind cluster 'genLevelCtrl' from 'Living room Switch above stairs top' to 'light_living_room' (Error: Bind 0x14b457fffec8dfbd/1 genLevelCtrl from '22' failed (AREQ - ZDO - bindRsp after 10000ms))
```

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2652R1 (ZZH) - worked like a charm on CC2531 with sourcing firmware
Adapter firmware version: CC26X2R1_20200805

Koenkk, 2020-09-30:
This happens when the switch is sleeping while the command is executed, wake it up by pressing a button on it right before sending the command

esbenr, 2020-10-01:
Hi. That wasen't the problem. I had to remove the device from Z2M and then from HA. Restart Z2M and pair the device from scratch. I have experienced that with almost all my Trådfri switches in the house.

It's a problem that arises because the devices was already paired via the CC2531. Then I switched to ZZH, changed pan_id and the devices automatically re-paired with this issue.
Once they have been completely removed as described above and repaired, they bind as expected.

closed 2020-10-01 by esbenr