Koenkk/zigbee2mqtt #4497: modelID: 'TS0601', manufacturerName: '_TZE200_kyfqmmyl'

nicktkd, 2020-09-29:
## Information about the device + link

3 gang swich

https://aliexpress.ru/item/4001096045589.html?spm=a2g0s.9042311.0.0.264d33edIiLJs1&_ga=2.72857041.359820516.1601298120-1311678371.1600341026&_gac=1.216930660.1601106719.Cj0KCQjwzbv7BRDIARIsAM-A6-1kQ8dE5Jv9N7sPus14koSHcYFrSgkHACVmBCMvCjhjPS5vVyOYajQaAgdLEALw_wcB
..

## data/database.db entry of the device

https://github.com/Koenkk/zigbee-herdsman-converters/blob/1638ae02ad15e780fd6e85bbe9d0becb035430d4/devices.js#L1227

This line need remove

This code need to past into  #L1274
```
 {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_kyfqmmyl'},
        ],
        model: 'TS0601_switch_3_gang',
        vendor: 'TuYa',
        description: '3 gang switch',
        supports: 'on/off',
        fromZigbee: [fz.tuya_switch, fz.ignore_basic_report, fz.tuya_switch2],
        toZigbee: [tz.tuya_switch_state],
        meta: {configureKey: 1, multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Norklmes', model: 'MKS-CM-W5'},
            {vendor: 'Somgoms', model: 'ZSQB-SMB-ZB'},
            {vendor: 'Moes', model: 'WS-EUB1-ZG'},
        ],
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(2)) await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(4)) await bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
        },
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 1, 'l3': 1};
        },
 },
```

nicktkd, 2020-09-30:
UPDATE 

I found problem, that I can't run a switch from web

I manage code tn that way, but I have a delay in working 

```
  {
        fingerprint: [
            {modelID: 'TS0601', manufacturerName: '_TZE200_kyfqmmyl'},
        ],
        model: 'TS0601_switch_3_sensor_gang',
        vendor: 'TuYa',
        description: '3 sensor gang switch',
        supports: 'on/off',
        fromZigbee: [fz.tuya_switch, fz.ignore_basic_report, fz.tuya_switch2],
        toZigbee: [tz.tuya_switch_state],
        meta: {configureKey: 1, multiEndpoint: true},
        whiteLabel: [
            {vendor: 'Norklmes', model: 'MKS-CM-W5'},
            {vendor: 'Somgoms', model: 'ZSQB-SMB-ZB'},
            {vendor: 'Moes', model: 'WS-EUB1-ZG'},
        ],
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(2)) await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            if (device.getEndpoint(3)) await bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        },
        endpoint: (device) => {
            // Endpoint selection is made in tuya_switch_state
            return {'l1': 1, 'l2': 1, 'l3': 1};
        },
    },

```

Koenkk, 2020-09-30:
Does it work correctly with this config? In that case I can add it

nicktkd, 2020-09-30:
Yes, it works, but with some delay. I will continue work on it, but it would be great if you add  it

Koenkk, 2020-09-30:
is this a router or end device?

nicktkd, 2020-09-30:
Right now HA recognizes it as a router https://gyazo.com/501938ea528be7a53810f89366752410

Koenkk, 2020-09-30:
Thanks, added the device.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-07 by github-actions[bot]