Koenkk/zigbee2mqtt #4499: newbie adding new device

benchdog93, 2020-09-29:
[Ecolink contact sensor.txt](https://github.com/Koenkk/zigbee2mqtt/files/5302268/Ecolink.contact.sensor.txt)

After following the directions and reading the log files I have somewhat successfully added an unsupported contact to to my devise,js file and my zigbee to mqtt log now accepts it as supported and I can see that it changes state but does not move into MQTT db. 

I played with the device config a lot with the same results. I also add one in ZHA which did add it to MQTT db but it does not change state, but does show temp.

I attached a txt file with the log data, config, device sig from ZHA, entities from ZHA, and my setup.

Did I configure the device correctly?
I am assuming it needs a converter config but I am not sure how to properly configure it. I have an example in the file I made after reviewing other devices but I am sure it is not right. I did not see one with temp. 

Hopefully I am not out of line and posting this in the right place.
Thank you for your support

Koenkk, 2020-09-30:
With the config shown in the log, does the device correctly publish contact: true and false over MQTT? If yes I can add this device and make sure this device is correctly discovered in Home Assistant.

benchdog93, 2020-09-30:
The device only publish contact true and false over zigbee2mqtt addon. It does not move into MQTT Mosquitto broker. Other devices I have added after this device have moved successfully. Hopefully I have made sense, Thank you

Koenkk, 2020-09-30:
Your config looks good so I've added it, it will be included in the next Zigbee2MQTT release tomorrow (1.15.0). 

About the MQTT part, not sure what the issue is here, lets see if it works in 1.15.0

benchdog93, 2020-10-07:
Wanted to let you know after I updated this evening contacts are working. Thank you again

closed 2020-10-07 by benchdog93