Koenkk/zigbee2mqtt #4500: Kwikset 66492-001 - No network route

krisoshau, 2020-09-30:
## What happened
Constantly unable to push/pull data from Kwikset door lock. When using lovelace to lock/unlock the button stays lit and nothing happens. Sometimes this works as intended. It also fails to get the current status and says its unlocked when it is locked.
https://www.zigbee2mqtt.io/devices/66492-001.html

Z2MQTT Version: 1.14.4.2 
Mosquito Version: 5.1
Supervisor Version:245
HassIO Version: 4.13

## What did you expect to happen
The device should lock/unlock with lovelace

## How to reproduce it (minimal and precise)
Issue the lock/unlock via lovelace. 

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
CC2531
Adapter firmware version:  CC2531_DEFAULT_20190608

Zigbee2MQTT:debug 2020-09-30 12:14:10: Received MQTT message on 'zigbee2mqtt/0x002446fffd03731e/set' with data 'LOCK'
Zigbee2MQTT:debug 2020-09-30 12:14:10: Publishing 'set' 'state' to '0x002446fffd03731e'
Zigbee2MQTT:error 2020-09-30 12:14:36: Publish 'set' 'state' to '0x002446fffd03731e' failed: 'Error: Command 0x002446fffd03731e/2 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'
Zigbee2MQTT:debug 2020-09-30 12:14:36: Error: Command 0x002446fffd03731e/2 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:311:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
Zigbee2MQTT:info  2020-09-30 12:14:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x002446fffd03731e' failed: 'Error: Command 0x002446fffd03731e/2 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"0x002446fffd03731e"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:debug 2020-09-30 12:14:43: Received Zigbee message from '0x00158d00035a11bb', type 'attributeReport', cluster 'genBasic', data '{"65282":[{"elmType":16,"elmVal":1},{"elmType":33,"elmVal":2972},{"elmType":33,"elmVal":9128},{"elmType":36,"elmVal":[0,1]},{"elmType":33,"elmVal":1362},{"elmType":32,"elmVal":93}]}' from endpoint 1 with groupID 0

I have also tried restarting all the components, and rebuilding my entire haasio from the ground up.

Koenkk, 2020-09-30:
Does it work fine again when repairing the device?

krisoshau, 2020-10-01:
Unfortunately it still occurs. I brought it of amazon too so I will try get a replacement sent to me so I can ensure it's not the physical unit. Any other ideas?

Koenkk, 2020-10-01:
If that also doesn't work we have to sniff the traffic to see what is going on: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

closed 2020-11-08 by github-actions[bot]