Koenkk/zigbee2mqtt #4501: Issues with adding homeassistant.js

atxbyea, 2020-09-30:
## What happened

Error when trying to add "homeassistant.js" to the container to create a new device in "devices.js"

https://pastebin.com/58KY169y

## What did you expect to happen
Container to boot up and be able to see the device in HA so I can trigger it and be able to create a new device support

## How to reproduce it (minimal and precise)

`docker create -it --name neji -v /docker/neji:/app/data -v /opt/zigbee-herdsman-converters:/app/node_modules/zigbee-herdsman-converters -v /opt/homeassistant.js:/app/lib/extension/homeassistant.js --device=/dev/ttyACM0 -e TZ=Europe/Oslo -v /run/udev:/run/udev:ro --privileged=true koenkk/zigbee2mqtt:latest-dev`

## Debug info
Zigbee2MQTT version: 1.14.4-dev
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417

Koenkk, 2020-09-30:
Your `homeassistant.js` is outdated (and therefore not compatible with your Zigbee2MQTT version). Make sure to update the latest version, so if you are on the latest-dev, take https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js

closed 2020-09-30 by Koenkk

atxbyea, 2020-10-01:
Thanks @Koenkk that worked wonders.

Should we perhaps add it to the documentation for adding new device support that you have to remember to use the right `homeassistant.js` ?

Koenkk, 2020-10-01:
@atxbyea we can, but in the coming time we will make changes to homeassisant.js unnecessary since we will move the discovery payloads to `devices.js`.