Koenkk/zigbee2mqtt #4503: Fixing ZigBee3.0 dual USB wireless socket plug

pedromargarido, 2020-09-30:
## What happened
This device has one power outlet and two usb ports. The mqtt set command is controling just the power outlet.

## What did you expect to happen
It should have two independant "switches" to control the usb and the outlet separately.

## How to reproduce it (minimal and precise)
I tied fixing it on devices.js chaging the original device:
`    {
        zigbeeModel: ['TS0112'],
        model: 'ZK-EU-2U',
        vendor: 'Moes',
        description: 'ZigBee3.0 dual USB wireless socket plug',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
    },`

To this:
`    {
        zigbeeModel: ['TS0112'],
        model: 'ZK-EU-2U',
        vendor: 'Moes',
        description: 'ZigBee3.0 dual USB wireless socket plug',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true, configureKey: 1},
        endpoint: (device) => {
            return {Socket: 1, USB: 2};
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint2);
        },
    },`

After this modification I'm able to control the outlet and the usb port publishing to the following topics:
`zigbee2mqtt/device/1/set
zigbee2mqtt/device/2/set`

The response I get is the following:
'{"linkquality":57,"state":"OFF","state_Socket":"ON","state_USB":"ON"}'

## The problem
1- I'm getting errors on zigbee2mqtt log:
`Zigbee2MQTT:error 2020-09-30 08:13:04: Failed to configure '0xec1bbdfffe66b673', attempt 1 (Error: ConfigureReporting 0xec1bbdfffe66b673/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 15031 - 1 - 3 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))`

`Zigbee2MQTT:error 2020-09-30 08:13:41: Failed to configure '0xec1bbdfffe66b673', attempt 2 (Error: ConfigureReporting 0xec1bbdfffe66b673/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 15031 - 1 - 6 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))`

2 - I though I would be able to control publishing on:
`zigbee2mqtt/device/USB/set`
`zigbee2mqtt/device/Socket/set`

Because I did:
`        endpoint: (device) => {
            return {Socket: 1, USB: 2};
        }`

But these topics are not being recognized by Zigbee2mqtt.

3 - It's being published:
`{"linkquality":55,"state":"ON","state_Socket":"OFF","state_USB":"ON"}`
"state_Socket" and "state_USB" are ok, but what is "state"??

4 - The device is not being recognized by HA Autodiscovery as a 2 switches device, it recognizes only one switch controling the "state_socket" and the "state". 

## database.db
`            2820
         ],
         "outClusterList":[
            25
         ],
         "clusters":{
            "genBasic":{
               "attributes":{
                  
               }
            }
         },
         "binds":[
            
         ]
      }
   },
   "appVersion":67,
   "zclVersion":3,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1601242259576`

Koenkk, 2020-09-30:
1. Why did you add the `configure()` in the first place? This is only needed when e.g. the switch can also controlled manually and without config no updates are published.
2. You can only use endpoint names defined here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/utils.js#L31
3. Old attribute, clear your `data/state.json`
4. I will take care of this once config is updated

pedromargarido, 2020-09-30:
1. Because I really don't know what I'm doing, I just copied from some other device that has 2 plugs. 

I commented both 'await configureReporting.onOff();' is that what you mean?

It's like that now:
```
        zigbeeModel: ['TS0112'],
        model: 'ZK-EU-2U',
        vendor: 'Moes',
        description: 'ZigBee3.0 dual USB wireless socket plug',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true, configureKey: 1},
        endpoint: (device) => {
            return {l1: 1, l2: 2};
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            //await configureReporting.onOff(endpoint1);
            const endpoint2 = device.getEndpoint(2);
            await bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
            //await configureReporting.onOff(endpoint2);
        },
    },
```

Well I did that, the errors are gone, but now i'm getting this:
``` 
Zigbee2MQTT:debug 2020-09-30 15:24:11: No converter available for 'ZK-EU-2U' with cluster 'genBasic' and type 'attributeReport' and data '{"appVersion":67}'
```
The mqtt topics are working. I'm able to control both outputs and to check their states.

You said `This is only needed when e.g. the switch can also controlled manually and without config no updates are published.` this device has a physical switch that control both outputs, at the same time, based the outlet state (if the outlet is off, when it's pressed, it will turn the usb and outled on, if the outlet is off, it will turn both on).

2. Sorry, I didn't know that, I fixed it.

3. Great, it's fixed.

4. Ok!

Thank you very much for your help!

sjorge, 2020-09-30:
You can add fz.ignore_basic_report to fromZigbee to make the debug go away

Koenkk, 2020-10-02:
1. > You said This is only needed when e.g. the switch can also controlled manually and without config no updates are published. this device has a physical switch that control both outputs, at the same time, based the outlet state (if the outlet is off, when it's pressed, it will turn the usb and outled on, if the outlet is off, it will turn both on).

So when now controlling the device with the physical switch, does the state in MQTT update? If yes I will update the existing definition and this can be closed.

pedromargarido, 2020-10-02:
Yes, the status is being updated when the switch is pressed and both outputs are being controled and reported, you can update the existing definition if you want. 

The only thing I miss is a periodical report of the link quality, it's only being reported when one of the switches changes its state. Is there something we can do about that or it's the device behavior?

One last question, what about the HA Auto Discovery not working?

This is how it's being showed, only one switch:
[image]

Koenkk, 2020-10-02:
I've updated the definition. Linkquality is indeed reported when the device sends a message, this is normal and applies to all devices.

HA discovery will be fixed automatically.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-02 by Koenkk