Koenkk/zigbee2mqtt #4504: TRV Danfoss ally

zhapwey, 2020-09-30:
hello:

I have a danfoss trv by zigbee protocol.
I have seen that it is very similar to that of hive (the one that I do not have to test). What happens to me is that it does not accept system_mode: off only accepts heat.
[image]

The problem is that if I use the manual TRV (playing with the hand) it works fine, but if I write the command to it by zigbee it seems that it does not eliminate the calculations of its algorithm for the previous command and it does very strange things.

I am trying to create a converter to tell open window and that the trv turns off but I am not able, I have this info:
[image]

but I am unable to find it, if someone can give me a hand I would appreciate it.

thanks

zhapwey, 2020-09-30:
I've been looking because I'd love to understand how this is done.
Could be:

**1**.In _/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js_ edit the cluster  _hvacThermostat_ 
and add a line for each new attribute to create. For example for:
[image]
 MountingModeActive: { ID: 0x4013, type: dataType_1.default.boolean },

**2**.In _/opt/zigbee/node_modules/zigbee-herdsman-converters/converters/toZigbee.js_ create a converter for this attribute(
This I don't quite understand how to do it). For example:
    _thermostat_MountingModeActive: {
        key: ['MountingModeActive'],
        convertSet: async (entity, key, value, meta) => {
            const MountingModeActive = value;
            await entity.write('hvacThermostat', {MountingModeActive});
            return {state: {MountingModeActive: value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('hvacThermostat', ['MountingModeActive']);
            return {state: {MountingModeActive: value}};
        },
    },_
**3**. In devices.js add in toZigbee devices tz.MountingModeActive

Does something I think make sense or am I very lost ??? 

Thanks,

brozikcz, 2020-10-13:
Hi,

thank you for your investigation. Do you have any progress? :)

zhapwey, 2020-11-04:
I am not afraid that for this trv to work well, it is necessary to use commands that I am not clear about how they work, without the help of people who know I am not sure how to continue, sorry.

sjorge, 2020-11-04:
Did you get a list of the zigbee attribtutes it supports? If so we can probably work on this, if not it's guess work which is difficult without a testing unit.

zhapwey, 2020-11-05:
hello @sjorge:
If I think I have the updated mapping
[Danfoss trv.xlsx](https://github.com/Koenkk/zigbee2mqtt/files/5493076/Danfoss.trv.xlsx)

I have already achieved that it is quite similar to hive's trv, write and read practically everything to it, the problem is that to write the setpoint, if you write directly through the OccupiedHeating Setpoint attribute, the trv takes a long time to answer, the correct way to do it seems to be through the command 0x40 giving it the command and the exchange rate of the same to make it faster or slower, but I am unable.

And my other problem is that I cannot read or write in the attributes that start with 0x4XXX, to add the open window functionality or external measured room...

I am at your disposal for any test that I can do with it, and regarding a test unit, I have no problem sending mine to someone if you are interested in working with it. Because I think that this trv is of quite a quality and quite reliable, if it worked well I would surely mount it at home, the rest of the options that we have not quite convinced me at all

External   Measured Room Sensor
--

External   Measured Room Sensor
--

sjorge, 2020-11-05:
Ah you've been reverse engineering by writing to it! Impressive.

Perhaps if there are custom attributes involved we need to set the manufacturing specific options, which requires a change to zigbee-herdsman to teach it about them I think.

zhapwey, 2020-11-05:
If I have achieved something but I am blocked with the issue of the command 0x40, which also the valve reports it, that is.
If you change the setpoint from zigbee z2m/0x14b457fffeb6915c {"linkquality":123,"occupied_heating_setpoint":24.5,"setpoint_change_source":2}

If you manually change the setpoint (from the valve) z2m/0x14b457fffeb6915c {"linkquality":123,"occupied_heating_setpoint":22.5,"setpoint_change_source":0}

The manual change is done instantly and the one you do by zigbee takes a long time to react.

If you see possibilities to make it work I am at your disposal, do you think we can try to do something with it?

Best regards,

Zeddy-, 2020-11-10:
Hi, 

just wanted to jump in here as i am struggling with exactly the same thing. I have not seen any other device implement COMMANDS. just reading and writing attributes.

[image]

zhapwey, 2020-11-11:
hi:
We continue at the same point, I do not know if there is no one who can give us a hand or this would be more to make a query in converts rather than in z2m, but I do not have the knowledge to advance here.

I have tried to write the commandtype setpoint separately and it won't let me, it tells me the attribute is read-only. So it seems that the only way to make the TRV work correctly is through that command.

Best Regards,

GrahanF, 2020-11-14:
Hi Guys.

Just joined github as I came across this thread when I was looking for information on problems with changing setpoints on the Hive trv and I have information that may be pertinent to your issues.

I have just recently purchased 10 Hive trvs and they are intermittently failing to update their setpoint on a schedule change in the hive app. They can be updated manually within the app or by turning the wheel on the valve. Control always appears to follow the setpoint displayed on the valve. Hive firmware is V 21B. I am unaware of how much of the firmware is shared with the Danfoss variant and whether it too has this issue. Both appear to share some but not all features, however there is very little info out there. Tech support have been in denial for a month. The new firmware was pushed out on 23 September and was intended to cure a problem with slow response to heating demand.

As you guys are clearly interfacing with the valves  at a level that is far more informed than mere "users", any  insight you can share that might help to define the problem would be gratefully received,

The failure mode is unknown  - random and intermittent but pointing perhaps to some internal memory corruption rather than a failure of comms or an issue with control logic.

The issue is causing major frustration within the Hive user community.

I hope this information is helpful and apologies if, through ignorance,  I have breached protocols by interjecting  in this conversation.

Regards.

Graham

Graham

Zeddy-, 2020-11-14:
We updated the Ally with the latest firmware and so far what we noticed is external open window works, and setpoint seems to actually move the valve pretty fast. Open window  closes the valve within 30 seconds, which is decent. But i think the main issue is the internal PID is the reason why these are so slow.

But something i still find absolutely amazing is how zigbee commands are never talked about, anywhere? Have i misunderstood them or something? Why doesn't zigbee2mqtt have a single custom command integrated?

sjorge, 2020-11-14:
> Why doesn't zigbee2mqtt have a single custom command integrated?

Because the manufacturing specific cluster stuff is defined in zigbee-herdsman and they are used in zigzagged-herdsman-convertors?

e.g. https://github.com/Koenkk/zigbee-herdsman/blob/60418017ea739a7c0d96aca2814d82e79de01901/src/zcl/definition/cluster.ts#L446 which is then used here https://github.com/Koenkk/zigbee-herdsman-converters/blob/84b47e269be2fc50653854234fb970f7dc8c8f7d/converters/toZigbee.js#L3728

AlexThePotter, 2020-11-27:
I get a similar issue with Hive TRV001. 

If I use it manually, it reacts instantly. If I change set point over z2m the setpoint is changed - the screen shows new setpoint, but the device never changes the opened amount of the TRV. even if massively different to the current temp.

zhapwey, 2020-11-30:
hello:

I did some tests but my knowledge of this is very limited, in fact that is why I asked for help but for now we have had no luck. In case they help someone

Try creating a command looking at setpoint_raise_lower.
**In nano /opt/z2m/node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js**
            setpointCommand: {
                ID: 64,
                parameters: [
                    { name: 'setpointType', type: dataType_1.default.enum8 },
                    { name: 'HeatingSetpoint', type: dataType_1.default.int16 },
                ],
            },
**And tozigbee.js:**
    thermostat_setpointCommand: {
        key: ['setpointCommand'],
        convertSet: async (entity, key, value, meta) => {
            const payload = {setpointType: value.setpointType, HeatingSetpoint: Math.round(value.HeatingSetpoint)};
            await entity.command('hvacThermostat', 'setpointCommand', payload, getOptions(meta.mapped, entity));
            return {readAfterWriteTime: 250, setpoint: {setpointcommand: value}};
        },
    },

**Call the converter from the device.js like this:**

mosquitto_pub -t 'z2m/0x14b457fffeb6915c/set' -m '{"occupied_heating_setpoint":20,"Setpoint_Change_Source":1}'

And it does not give me a command error in the debug but it does not change the command either.

I have been confirmed from various sources that these trv, danfoss, viessman, hive (it is the same valve) change the setpoint differently if you change from the command with the setpoint_change_source to one value or another, and by default without writing there is very very slow

I have the specifications of the command but I don't know how to work in case they are worth it to someone:

[image]

[image]

I have stuck with her and I have not achieved anything, good luck for those who try.

BR

ma-zal, 2020-12-14:
For this issue I opened topic [Danfoss Ally: Slow actuator response](https://community.home-assistant.io/t/danfoss-ally-slow-actuator-response/254939) in Community HA. I hope other people will add experiences and knowledge, which it helps to solve the issue.

randomuser389, 2020-12-19:
> hello:
> 
> I did some tests but my knowledge of this is very limited, in fact that is why I asked for help but for now we have had no luck. In case they help someone
> 
> Try creating a command looking at setpoint_raise_lower.
> **In nano /opt/z2m/node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js**
> setpointCommand: {
> ID: 64,
> parameters: [
> { name: 'setpointType', type: dataType_1.default.enum8 },
> { name: 'HeatingSetpoint', type: dataType_1.default.int16 },
> ],
> },
> **And tozigbee.js:**
> thermostat_setpointCommand: {
> key: ['setpointCommand'],
> convertSet: async (entity, key, value, meta) => {
> const payload = {setpointType: value.setpointType, HeatingSetpoint: Math.round(value.HeatingSetpoint)};
> await entity.command('hvacThermostat', 'setpointCommand', payload, getOptions(meta.mapped, entity));
> return {readAfterWriteTime: 250, setpoint: {setpointcommand: value}};
> },
> },
> 
> **Call the converter from the device.js like this:**
> 
> mosquitto_pub -t 'z2m/0x14b457fffeb6915c/set' -m '{"occupied_heating_setpoint":20,"Setpoint_Change_Source":1}'
> 
> And it does not give me a command error in the debug but it does not change the command either.
> 
> I have been confirmed from various sources that these trv, danfoss, viessman, hive (it is the same valve) change the setpoint differently if you change from the command with the setpoint_change_source to one value or another, and by default without writing there is very very slow
> 
> I have the specifications of the command but I don't know how to work in case they are worth it to someone:
> 
> [image]
> 
> [image]
> 
> I have stuck with her and I have not achieved anything, good luck for those who try.
> 
> If someone wants to do some test, happy to help you.
> BR

note that the setpoint command is a danfoss manufacturer specific command so manufacturer needs to be set at 0x1246 in the command and the flag that indicate manufactuer needs to set true.

gielfeldt, 2021-01-20:
I think the payload published to mqtt need to be: 
```json
{
  "setpointCommand": {
     "setpointType": 1,
     "HeatingSetpoint": 20
  }
}
```

And the "thermostat_setpointCommand" need to be added to devices.js for the device in question.

If I'm reading the code right. Which I'll admit, I'm not sure I am :-)

closed 2021-01-28 by github-actions[bot]

marcinkowalczyk, 2021-02-19:
Seems this issue has been fixed in Deconz, ia there a way to get it fixed here too?

robertalexa, 2021-05-19:
For everyone interested in that command, plus a lot more features, keep an eye on the PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/2592

This handles a lot of features. This is what I've managed to get.
[image]

Rob

zhapwey, 2021-05-19:
hello:

Great news, congratulations and great work!!!
I have tried to test your set_point but it has been impossible I add the converter in tozigbee.js and in devices the tozigbee but it give.s me an error, I will have to wait for the PR.

zigbee2mqtt/0x14b457fffebada95/set {"occupied_heating_setpoint":"24"}
zigbee2mqtt/bridge/logging {"level":"error","message":"Publish 'set' 'occupied_heating_setpoint' to '0x14b457fffebada95' failed: 'Error: Cluster 'hvacThermostat' has no command 'danfossSetpointCommand''"}
zigbee2mqtt/bridge/log {"message":"Publish 'set' 'occupied_heating_setpoint' to '0x14b457fffebada95' failed: 'Error: Cluster 'hvacThermostat' has no command 'danfossSetpointCommand''","meta":{"friendly_name":"0x14b457fffebada95"},"type":"zigbee_publish_error"}

I think I need to include something but I don't know what it is.

Best Regards,

robertalexa, 2021-05-20:
@zhapwey 

You are missing some code for zigbee-herdsman module.

The easiest way to get this is to run your tests on zigbee2mqtt dev branch (this will include the latest version of zigbee-herdsman with the danfossSetpointCommand).

So:
1: checkout zigbee2mqtt to dev branch
2: replace your danfoss.js with the on in PR
3. replace toZigbee.js
4. replace fromZigbee.js
5. test

Would you be kind  to help me with some info about the Danfoss Ally? I do not have this device but I am trying to support it properly. Could you please tell me when you physically turn the temperature on the TRV, what is the range that the device allows? Is it 6 to 28 degrees Celsius or 5 to 32? 

Thanks

sjorge, 2021-05-20:
> Would you be kind to help me with some info about the Danfoss Ally? I do not have this device but I am trying to support it properly. Could you please tell me when you physically turn the temperature on the TRV, what is the range that the device allows? Is it 6 to 28 degrees Celsius or 5 to 32?

It might be possible to query AbsMin/Max from the attributes, they are marked optional but I believe the viessmann had those, so there is a high chance the Danfoss and Hive have them too.

[image]

robertalexa, 2021-05-20:
@sjorge I know them for Hive TRV, but because it uses a different firmware to Danfoss I just want to double check.

The original Danfoss code was set to 6 to 28 degrees and my Hive TRV physically goes 5 to 32. So I have corrected the code for Hive to the latter.

If Danfoss is the same, then I will change the code for Danfoss Ally as part of my PR.

Rob

zhapwey, 2021-05-20:
good:
thanks a lot tomorrow i try your pr. Well, I have 3 units, two came to 6-28 and another to 5-35. Another issue that I have seen in these valves is that the first days they open at an hour and close, it seems that they are looking for the opening point of the valve or something like that, suddenly they stop doing it. It has happened to yours, I tell you after a reset or at the beginning, it seems that they have some register that puts them in regulation mode for this. regards,

robertalexa, 2021-05-20:
@zhapwey I think you might be referring to the calibration process? Where the PID learns how long it takes for temperature to be reached depending on valve movement.
If this is what you mean, I am not sure what is the logic behind it making it happen, but my guess is that it would be an internal process of the PID. If anyone knows any better please shed some light.

sjorge, 2021-05-20:
Interesting, is the one running a different firmware? If it is, we probably need to do something like we do for color_temp and read the min/max per device and store it.

robertalexa, 2021-05-22:
@zhapwey The PR is now merged in the dev branch of z2m, if you test it there everything should work out of the box.

There is however another PR in progress for a couple of improvements
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2614

zhapwey, 2021-05-22:
It is great news I hope to give you my opinion on the operation shortly. Regarding the pr you are doing, if you need any test, do not hesitate to tell me that I will help you with whatever you need, Regarding what I told you the other day, more than a pid adjustment, it seems that the valve wants to learn its point where it begins to open. because the heat_requiered is set to true when piheatingdemand> 0 and the valve does not really start to open until it is at 30%, anyway I think I have all the zigbee clusters of this valve would they help?

robertalexa, 2021-05-22:
Everything about this devices has been fixed or added if it was missing.

Changing the temperature will now instantly occur and you will see pi changing fast, unlike before.

Try to use the dev brach and see if everything works as expected

zhapwey, 2021-05-24:
It's wonderful, it's a shame that winter is over but the TRV's response is much faster now.

Great job, @robertalexa congratulations !!!!

rindlerblabla, 2021-08-09:
@robertalexa Have ordered some thermostats, but haven't received them yet. However, is it possible to use an external sensor, like https://community.home-assistant.io/t/danfoss-ally-trv-working-with-remote-temp-sensor/276686 / https://community.home-assistant.io/t/danfoss-ally-thermostat-firmware-update-ota-v1-08-open-window-external-temp-zha-deconz/261951/11 ?

robertalexa, 2021-08-09:
I am guessing you mean an external temperature sensor. No, it is not possible currently using z2m. The device does not have converters for attribute 16405 (0x4015)
[image]

I will add this on my TODO list and try to have a look at implementing it, but I am not making any promises.

rindlerblabla, 2021-08-09:
> I am guessing you mean an external temperature sensor. No, it is not possible currently using z2m. The device does not have converters for attribute 16405 (0x4015)
> [image]
> 
> I will add this on my TODO list and try to have a look at implementing it, but I am not making any promises.

Cool, thanks!

robertalexa, 2021-08-09:
@rindlerblabla PR created https://github.com/Koenkk/zigbee-herdsman-converters/pull/2890

Hopefully this will be available in dev in a few days and in stable on 1st of Sept.

Hope this helps you!

rindlerblabla, 2021-08-10:
> @rindlerblabla PR created [Koenkk/zigbee-herdsman-converters#2890](https://github.com/Koenkk/zigbee-herdsman-converters/pull/2890)
> 
> Hopefully this will be available in dev in a few days and in stable on 1st of Sept.
> 
> Hope this helps you!

Wow, thanks! A lot!

rindlerblabla, 2021-08-12:
@robertalexa Sorry, just didn't get this.

`Set at maximum 3 hours interval but not more often than every 30 minutes at every 100 value change.`

Should the value be multiplied by hundred, or can I just send the regular value, ie 23.2? According to this they multiply the value... https://community.home-assistant.io/t/danfoss-ally-trv-working-with-remote-temp-sensor/276686/

robertalexa, 2021-08-12:
The value would be in this format. 22.68 degrees C -> 2268.

As to the spec, unfortunately I can not explain this any further because personally i am not sure I understand it fully either. That is because the specs are a bit vague.
[image]

The way i interpret this is the following:
- you need to send a value more often than 3 hours. this is because it internally resets to standard operation mode every 3 hours. meaning it will use the internal temperature sensor instead.
- you should not update it more often than 30 mins
- you should not update it if the difference between the old value (30 mins ago) and the new value (now) is less than 100 points of value. so if last value was 2285 and new value is 2250, you should not send the update. However is the new value is 2150 you should.

I do not know the downside of not respecting these specs. The device will 100% accept the new value, but i do not know how the internal mechanism will react to it.

On a personal note, the way I would do it is to send the new value every 2h 30min, on whatever the difference is, even if there is no difference. I think the message you should take is do not send too small value differences too often. But if you send too small value differences, but closer to the 3h expiry time, that should be absolutely fine.

Hope this helps

rindlerblabla, 2021-08-12:
@robertalexa Thanks for your explanation! I created an automation which sends a new value every 30th minute if the difference is more than 100 compared to the external_measured_room_sensor. Hope it will bot drain the batteries.

rindlerblabla, 2021-08-12:
I use this as condition. 

`condition: template
value_template: >-
  {{ (states('sensor.REAL_EXTERNAL_SENSOR') | float * 100) -
  (states('sensor.ally_external_measured_room_sensor') | float) | abs
  >100 }}`

rindlerblabla, 2021-08-15:
By the way, do you know why `pi_heating_demand` isn't exposed to Home assistant? It's not among those inactivated entities either.

robertalexa, 2021-08-15:
`pi_heating_demand` is a property of the z2m Climate type, alongside `occupied_heating_setpoint`, `local_temperature` and `system_mode`. When a z2m Climate gets translated into a HA Climate, HA does not have a `pi_heating_demand`. This is not related to how this device is supported in z2m, this is valid for all climate devices, it is how z2m is coded.

In reality `pi_heating_demand` will not give you any information that is useful in your automation or decision making (not as far as i can think tbh). That only shows how much the valve is open, and that is decided by the internal PID of the TRV, based on the temperature difference between the local and demand.

For any decision making with regards to if the trv is in operation or not, you can safely use the attribute called `heat_required`. This is the PID saying if heat is needed to reach the desired temp or not.

Hope this helps.

wormiedk, 2021-08-21:
pi_heating_demand could be usefull in rooms with multiple radiators to see if one is much more active than another and therefore would benefit from an external temp sensor or similar

robertalexa, 2021-08-21:
You can still find that out with heat required, and not with how much the valve is open. But maybe i am over simplifying things.

As to external sensor, i have added this functionality in a recent PR and it should be available in the next release.

ahaafin, 2021-11-09:
Is there a way to disable Window Open feature on Danfoss Ally? Would it be possible with zigbee2mqtt? I don't have Danfoss Gateway.

randomuser389, 2021-11-10:
> Is there a way to disable Window Open feature on Danfoss Ally? Would it be possible with zigbee2mqtt? I don't have Danfoss Gateway.

Yes it's possible to disable Window Open feature from firmware 01.08 the attribute for this is Cluster 0x201 Attribute 0x4051.

See Zigbee Cluster Specifications:
https://assets.danfoss.com/documents/176987/AM375549618098en-000101.pdf

ahaafin, 2021-11-28:
> Yes it's possible to disable Window Open feature from firmware 01.08 the attribute for this is Cluster 0x201 Attribute 0x4051.
>
> See Zigbee Cluster Specifications: [...]

Thank you for the info. Any idea how to do this in practise? I'm new to zigbee. I have a zigbee2mqtt configured and Ally connected to it but couldn't find where I could manually enter the cluster and attribute. Google didn't help.

robertalexa, 2021-11-28:
It is currently not supported, there has been some work done, but to my knowledge it is not yet completed.

The attribute has received support in zigbee-herdsman, but not yet in zigbee-herdsman-converters.

@kennylevinsen is Window Open Feature still something you plan on adding or should I have a look at it?

kennylevinsen, 2021-11-28:
> @kennylevinsen is Window Open Feature still something you plan on adding or should I have a look at it?

It wasn't on my todo-list, so go ahead!

Mikk36, 2021-12-01:
>     * you should not update it if the difference between the old value (30 mins ago) and the new value (now) is less than 100 points of value. so if last value was 2285 and new value is 2250, you should not send the update. However is the new value is 2150 you should.

0.1K = 0.1C°
K -> Kelvin

robertalexa, 2021-12-01:
@Mikk36 
[image]

Here is where the confusion stems from :)

0.1K is never ever in a million years equal to 0.1C. That is because there is a difference of -273.15 between measurements on these 2 scales. However (and this is important), their scale is liniar. Meaning that a DIFFERENCE of 0.1K is equal to 0.1C

But thanks for raising this again. Now that my mind is thinking straight, and I have been away from it for some time, I realise that i have misinterpreted it in the past.

So as a result there is now a PR to update the description of the attribute https://github.com/Koenkk/zigbee-herdsman-converters/pull/3439

Mikk36, 2021-12-01:
That is exactly what I meant, that a difference of 0.1K is equal to a difference of 0.1C° in relative terms.

robertalexa, 2021-12-01:
> Yes it's possible to disable Window Open feature from firmware 01.08 the attribute for this is Cluster 0x201 Attribute 0x4051.
>
> See Zigbee Cluster Specifications: [...]

@randomuser389 So I have just had a look at this now using my Hive TRV (which is basically a Danfoss Ally but with their own firmware) and unfortunately I am getting an UNSUPPORTED_ATTRIBUTE error message. I suspect this is because of the Hive firmware vs the Danfoss official one, but it makes me reticent over adding this functionality for Danfoss blindly (e.g. without being able to confirm it works)

If anyone has a danfoss ally (and even the Popp 701721 TRV) and is willing to help me test the above, then at least I know how to move forward?

toZigbee
```
    danfoss_window_open_feature_enable: {
        key: ['window_open_feature_enable'],
        convertSet: async (entity, key, value, meta) => {
            await entity.write('hvacThermostat', {'danfossWindowOpenFeatureEnable': value}, manufacturerOptions.danfoss);
            return {readAfterWriteTime: 200, state: {'window_open_feature_enable': value}};
        },
        convertGet: async (entity, key, meta) => {
            await entity.read('hvacThermostat', ['danfossWindowOpenFeatureEnable'], manufacturerOptions.danfoss);
        },
    },
```

fromZigbee
```
if (msg.data.hasOwnProperty('danfossWindowOpenFeatureEnable')) {
                result[postfixWithEndpointName('window_open_feature_enable', msg, model)] = (msg.data['danfossWindowOpenFeatureEnable'] === 1);
            }
```

Then on the device add this to `toZigbee` section: `tz.danfoss_window_open_feature_enable` and to the exposes:
```
exposes.binary('window_open_feature_enable', ea.ALL, true, false)
                .withDescription('Whether or not the window open feature is enabled'),
```

ipeacocks, 2021-12-01:
@robertalexa thank you very much for your work. Previously I've bought one Moes HY368 TRV. It's quite strange thermostat, it works but very inertial and opens valve very slow even when temperature is 4 C less than target one.

Danfoss Ally at first point of view (bought it today) it's like BMW (or what do you like more?) in the world of TRVs. Good plastic, nice control, silent engine. Wow, sounds like a miracle! Yes, it costs a lot but it's more or less fair price.

Few question regarding it:

* Is it possible to turn it off? I can't see such option in zigbee2mqtt. How to turn it off for summer?
* Is it possible to output `pi_heating_demand` value (valve position). I like to see how often it tries to open/close valve.

robertalexa, 2021-12-02:
Hi @ipeacocks I have never tried other TRVs, so i can't make a good comparison, but what i can say is that i have never been dissappointed by my 8 Hive TRVs (undercover Danfoss Ally). So i personally holehatedly recommend them.

> * Is it possible to turn it off? I can't see such option in zigbee2mqtt. How to turn it off for summer?

Set the temperature to a low value, like 9C for example. So you end up with sth that looks like this:
[image]

> * Is it possible to output `pi_heating_demand` value (valve position). I like to see how often it tries to open/close valve.

It is as of the latest release of z2m (today, 1st of December) https://github.com/Koenkk/zigbee2mqtt/pull/9759

Hope this helps :)

ipeacocks, 2021-12-02:
> Set the temperature to a low value, like 9C for example
It's just not very logical as for me. But not critical in general. As I understood it's more vendor-specific thing.

> It is as of the latest release of z2m (today, 1st of December) #9759

Nice. Now I am fully happy.

robertalexa, 2021-12-02:
> It's just not very logical as for me. But not critical in general. As I understood it's more vendor-specific thing.

Unfortunately there is no better way. And this is not a limitation of z2m, but rather the device itself. Other thermostats will have an `off` `system_mode`. The Danfoss only has `heat` (on) and nothing else.

[image]

So i guess we have to make it work with what we've got :)

This is what I am doing in my setup and automations and I have personally not had any issues because of it (or weird quirks). :)

rindlerblabla, 2021-12-02:
Another question regarding the external temperature sensor. Would it be possible to bind the external sensor somehow directly to the thermostat without sending those values from eg Home Assistant? Does zigbee in general support these kind of bindings?

robertalexa, 2021-12-02:
Based on the specifications of the thermostat, my guess would be that it is not possible. The specs mention that the gateway should / will be sending the relevant information at certain time intervals and temperature differences. So based on that logic, a direct binding will not do the same.

As to the actual binding (based on my understanding of binding), that would not be possible. In binding you need 2 devices to communicate "about" the same cluster and attributes. A temperature sensor will not talk about the climate cluster, and a very specific attribute (which in this case is a danfoss specific one). Hopefully I am not misunderstanding it so that i am misinforming you.

Hope this helps

kennylevinsen, 2021-12-02:
> @robertalexa thank you very much for your work. Previously I've bought one Moes HY368 TRV. It's quite strange thermostat, it works but very inertial and opens valve very slow even when temperature is 4 C less than target one.

Danfoss uses a custom command for quick response, otherwise doing very slow response. Maybe it's the same for Moes?

ipeacocks, 2021-12-02:
> Unfortunately there is no better way. And this is not a limitation of z2m, but rather the device itself. Other thermostats will have an `off` `system_mode`. The Danfoss only has `heat` (on) and nothing else.

@robertalexa I am not blaming you of course. Just wished to ask that directly. I've more or less suspected that it's the nature of this valve. Just shared that it's a bit strange. Thank you for you integration because so far I am very happy with this valve.

[23 more replies]