Koenkk/zigbee2mqtt #4509: Lonsonho 2 gang switch not pairing

saabirsa, 2020-09-30:
Hello,

I am trying to pair a Lonsonho 2 gang switch (battery powered). For some reason, no matter what I try, it refuses to show up for interview in the zigbee2mqtt logs.

Separately, I have a Lonsonho 1 gang switch (battery powered) successfully paired already. I removed this to see if there was a conflict, and tried to pair the 2 gang switch only, but still no luck.

To make sure there wasn't a hardware issue with the 2 gang switch, I've attempted to connect the switch to my SmartThings hub, where it was picked up instantly.

The logs for the Lonsonho 1 gang switch shows as the following;
https://pastebin.com/qYMvs33F
(As a side, the 1 gang switch identifies as _TZ3000_peszejy7_ which I am also not sure is correct)

Some further background:
I have 6 devices already connected and working (including the Lonsonho 1 gang switch)
`permit_join` is set to true, logging is set to `debug`
I have attempted to pair straight after restarting zigbee2mqtt
The co-ordinator (a CC2531) is connected via a USB extension cable
I have tried to power down other hubs (SmartThings/Tradfri) while attempting to pair
I've replaced the battery
I've tried another of the switch (Lonsonho 2 gang) with the same issue

I was wondering if anyone had any ideas on how to get the device to pair? Happy to share more logs if required.

Koenkk, 2020-10-01:
Have you tried reflashing your CC2531 and pairing the device immediately after that?

saabirsa, 2020-10-01:
Oddly that seems to have done the trick.....many thanks!
Out of interest, is there any particular reason why this seems to work? Is it as simple as a failed flash first time?

closed 2020-10-01 by saabirsa

Koenkk, 2020-10-02:
@saabirsa if using the source routing fimrware (20190619) this can happen because the internal child table is full (it can only handle 5 direct connections). Reflashing will clear this table freeing up space.