Koenkk/zigbee2mqtt #4511: zigbee2mqtt/bridge/configure generates an error with TS0121_switch

morphelinho, 2020-10-01:
## What happened
Tried to configure a TS0121_switch device but it produces error message
## What did you expect to happen
Configuration is successful and no error message is produced
## How to reproduce it (minimal and precise)
Send MQTT message:
> topic: zigbee2mqtt/bridge/configure
payload: MySwitch_1

Generated log messages:
> error 2020-09-28 19:17:14: Failed to configure 'MySwitch_1', attempt 1 (TypeError: ConfigureReporting MySwitch_1/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable)
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:164:28)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:374:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))

## Debug info
Zigbee2MQTT version: 1.14.4 (commit #0cd4607)
Adapter hardware: Conbee II
Adapter firmware version: 26650700

Koenkk, 2020-10-02:
Can you try with 1.15.0? There are some Conbee related fixes in there.

morphelinho, 2020-10-02:
Thanks for your answer!

I updated the Zigbee2MQTT version to 1.15.0 (commit #1bccc5d).

Unfortunately I received nearly the same error message (with slightly different line numbers in stacktrace)

> error 2020-10-02 14:54:36: Failed to configure 'MySwitch_1', attempt 1 (TypeError: ConfigureReporting MySwitch_1/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable)
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:171:28)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:384:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))

I expect if the configuration is successful that the zigbee requests have a 10000ms timeout? Is this expectation correct?

Is it currently possible to specify a device specific timeout for zigbee requests so that a nearly 'fire and forget' behaviour is configurable with a accordingly low timeout? What is the default timeout for zigbee requests?

Koenkk, 2020-10-02:
Please provide the herdsman debug logging when reproducing this so at @chrishae can look further into this.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

morphelinho, 2020-10-02:
Herdsman debug logging:

>  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 41 +8s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +8s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 42 +10s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +10s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +6ms
Zigbee2MQTT:debug 2020-10-02 22:07:08: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'MySwitch_1'
Zigbee2MQTT:info  2020-10-02 22:07:08: Configuring 'MySwitch_1'
  zigbee-herdsman:controller:endpoint Bind MySwitch_1/1 genOnOff from 'MyCoordinator/1' +21s
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x12e7 EP:0 SeqNr. 43 request id: 6 +3s
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 6 status: 0 +3s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +4ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110 +92ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +92ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 44 +5ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x12e7 request id: 6 confirm status: 0 +8ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 45 +48ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 45 srcAddr: 0x12e7 destAddr: 0x0 profile id: 0x0 cluster id: 0x8021 +50ms
  zigbee-herdsman:deconz:frameParser response payload: 6,0 +0ms
  zigbee-herdsman:deconz:adapter resolve data request without a transSeq Nr. +21s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0x12e7 status: 0 +1ms
  zigbee-herdsman:controller:endpoint ConfigureReporting MySwitch_1/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +165ms
  zigbee-herdsman:deconz:adapter zclFrame.payload: +2ms
  zigbee-herdsman:deconz:adapter [
  zigbee-herdsman:deconz:adapter   {
  zigbee-herdsman:deconz:adapter     direction: 0,
  zigbee-herdsman:deconz:adapter     attrId: 0,
  zigbee-herdsman:deconz:adapter     dataType: 16,
  zigbee-herdsman:deconz:adapter     minRepIntval: 0,
  zigbee-herdsman:deconz:adapter     maxRepIntval: 3600,
  zigbee-herdsman:deconz:adapter     repChange: 0
  zigbee-herdsman:deconz:adapter   }
  zigbee-herdsman:deconz:adapter ] +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x12e7 EP:1 SeqNr. 46 request id: 7 +47ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 7 status: 0 +51ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +9ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +9ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 47 +39ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x12e7 request id: 7 confirm status: 0 +42ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 4 +99ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +26ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1 +26ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 48 +21ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 48 srcAddr: 0x12e7 destAddr: 0x0 profile id: 0x104 cluster id: 0x6 +23ms
  zigbee-herdsman:deconz:frameParser response payload: 8,4,11,6,130 +0ms
  zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 4 +50ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":6,"statusCode":130}},"address":4839,"endpoint":1,"linkquality":120,"groupID":null}' +21s
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +4ms
  zigbee-herdsman:deconz:adapter response received +2ms
[... cut ...] cEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable)
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:171:28)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:384:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))

ChrisHae, 2020-10-06:
> zigbee-herdsman:controller:log Received 'zcl' data [...]

It seems the configure reporting command is not supported by the device (StatusCode: 130)

Can anybody confirm that configure reporting of the onOff Cluster works with this device e.g. with the CC25xx stick?

Imaginous, 2020-11-03:
I can't reply that it is working, but I can reply that my QS-Zigbee-D02-TRIAC-2C-LN gives the same error.
I'm also using a ConbeeII stick. 

I'm using V1.16.0 of Zigbee2MQTT

Nov 03 08:05:01 domoticapi1 npm[28003]: Zigbee2MQTT:info  2020-11-03 08:05:01: Configuring 'Slaapkamer1_Verlichting'
Nov 03 08:05:03 domoticapi1 npm[28003]: Zigbee2MQTT:error 2020-11-03 08:05:03: Failed to configure 'Slaapkamer1_Verlichting', attempt 3 (Typorting 0xbc33acfffe6563a7/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"bleResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCodeequenceNumber":null}) failed (payload is not iterable)
Nov 03 08:05:03 domoticapi1 npm[28003]:     at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/end
Nov 03 08:05:03 domoticapi1 npm[28003]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/end
Nov 03 08:05:03 domoticapi1 npm[28003]:     at Generator.next (<anonymous>)
Nov 03 08:05:03 domoticapi1 npm[28003]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24

morphelinho, 2020-11-03:
In the meantime I switched to slae.sh CC2652RB stick but the problem stays the same, still nearly the same message (running on 1.15.0):
>zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":6,"statusCode":130}},"address":42475,"endpoint":1,"linkquality":54,"groupID":0}' +238ms

>error 2020-11-03 08:28:02: Failed to configure 'MySwitch_1', attempt 1 (Error: ConfigureReporting MySwitch_1/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 42475 - 1 - 3 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))

Nevertheless the device is working so i can switch the ON/OFF state as expected.

ChrisHae, 2020-11-04:
@morphelinho So it is the same error with CC...Stick and z-stack adapter (Error Code 130) So it seems the device simply isn't configurable. But it is working so it shouldn't be a problem.

@Imaginous
Could you enable zigbee-herdsman debug logging https://www.zigbee2mqtt.io/information/debug.html
and post the output from that configuring step?

It could be the same issue that the device itself is not configureable. Is it working though?

Imaginous, 2020-11-04:
@ChrisHae The device is working. I only have one problem with it, if I send a brightness value of 1 to 15 the lamp itself is off (state is on). From value 16 to 254 it works normally. But I think that's a problem with the firmware of the device. There is no way to calibrate the dimming mode. At least I don't know how, I don't have the original bridge or app.

The requested log:
```
zigbee-herdsman:controller:endpoint ConfigureReporting 0xbc33acfffe6563a7/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable) +0ms
Zigbee2MQTT:error 2020-11-04 14:08:50: Failed to configure 'Slaapkamer1_Verlichting', attempt 1 (TypeError: ConfigureReporting 0xbc33acfffe6563a7/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (payload is not iterable)
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:171:28)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:386:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

morphelinho, 2020-11-15:
My issue seems to be fixed with 1.16.0, maybe wrt #4598

closed 2020-11-15 by morphelinho