Koenkk/zigbee2mqtt #4512: Records for unexisting devices in database.db

Hacker-CB, 2020-10-01:
After half year using z2m I found that `database.db` has 2 records for items which does not have corresponding devices in `configuration.yaml`:

```json
{"id":168,"ieeeAddr":"0xd0cf5efffe6f5da9","nwkAddr":47993,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1589833315846}
{"id":18,"ieeeAddr":"0xd0cf5efffe6f5da9","nwkAddr":39729,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1594289247676}
```
It's impossible to delete them from the frontend.
So, may be it will be better to delete such items form the db?

Hacker-CB, 2020-10-01:
Both has same `ieeeAddr` and there is no such addr in `coordinator_backup.json`

Koenkk, 2020-10-02:
Fixed, for the current entities remove them manually from the `database.db`. After the fix this will not happen anymore.

Hacker-CB, 2020-10-02:
Is I need to reorder all IDs after deleting? Or can just delete lines?

Koenkk, 2020-10-02:
Just delete the lines :smile:

Hacker-CB, 2020-10-02:
Ok, thanks

closed 2020-10-02 by Hacker-CB