Koenkk/zigbee2mqtt #4513: Can't start new stick

golddragon007, 2020-10-01:
## What happened
I've changed my CC2531 controller to a slaesh’s CC2652RB stick one. First I've flashed the CC2652RB, then I replaced the old stick to the new one, I started the zigbee, I saw a pan_id error. I've changed it, by incrementing one (from 6754 to 6755). After restart I've still getting the following:
```
Zigbee2MQTT:info  2020-10-01 14:50:09: Logging to console and directory: 'data/log/2020-10-01.14-50-09' filename: log.txt
Zigbee2MQTT:info  2020-10-01 14:50:10: Starting Zigbee2MQTT version 1.14.4 (commit #1.14.4)
Zigbee2MQTT:info  2020-10-01 14:50:10: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-10-01 14:51:26: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-01 14:51:26: Failed to start zigbee
Zigbee2MQTT:error 2020-10-01 14:51:26: Exiting...
Zigbee2MQTT:error 2020-10-01 14:51:26: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:171:23
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.4/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.14.4 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-01T12_51_26_198Z-debug.log
2020-10-01T14:51:28: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-10-01T14:51:28: PM2 log: App [npm:0] starting in -fork mode-
2020-10-01T14:51:28: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.4 start /zigbee2mqtt-1.14.4
> node index.js
```

## What did you expect to happen
Boot up and work.
## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.14.4.2 
Adapter hardware: CC2531 and slaesh’s CC2652RB stick
Adapter firmware version: CC2652RB_20200805 (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin)

golddragon007, 2020-10-01:
I've just updated to 1.15 both the stick and the software, incremented the pan_id, but still, it doesn't want to start... (same error)

Koenkk, 2020-10-02:
Could you provide the herdsman debug logging of this issue?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

golddragon007, 2020-10-02:
@Koenkk 
As I said, I don't know where do I need to put the `"zigbee_herdsman_debug": true` to enable it, it's not mentioned anywhere just configuration. To which? And on which level? advanced? root?

Trauma, 2020-10-03:
Some what the same issue with latest docker tag, reverting to 1.14.4 works.

```
Using '/app/data' as data directory

> zigbee2mqtt@1.14.4 start /app
> node index.js

Zigbee2MQTT:info  2020-10-03 16:59:00: Logging to console only'
Zigbee2MQTT:debug 2020-10-03 16:59:01: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2020-10-03 16:59:01: Starting Zigbee2MQTT version 1.14.4 (commit #654817a)
Zigbee2MQTT:info  2020-10-03 16:59:01: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-03 16:59:01: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[20],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2020-10-03 16:59:20: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-03 16:59:20: Failed to start zigbee
Zigbee2MQTT:error 2020-10-03 16:59:20: Exiting...
Zigbee2MQTT:error 2020-10-03 16:59:20: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:90:31)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.4 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.4 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-03T14_59_20_839Z-debug.log
```

golddragon007, 2020-10-03:
Here is the log, with a full revert I was able to reproduce it.
[mqtt.txt](https://github.com/Koenkk/zigbee2mqtt/files/5322034/mqtt.txt)

Koenkk, 2020-10-03:
@golddragon007  can you try staring on a different channel? (docs: https://www.zigbee2mqtt.io/information/configuration.html#configuration)

golddragon007, 2020-10-04:
Well, it's kind of late, I've found a way that seems it's work now, but I have still issues. So I did the following:
1. Create a full backup for HA.
2. Reflash ZigBee controller with the newest kernel
3. Remove antenna
4. Plugin and start z2m
5. After it's starting up all my ZigBee entities goes away from HA
6. Full restore of HA

Now I have a somewhat working controller, the strange stuff that only some of the devices connects to it the others refusing or as I see from the log the controller refusing it... Also, I got some errors and after maybe 12 hours it's breaking down totally with buffer full error. I try to collect this, until then please check the current reason why refusing devices to connect.
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/5323467/z2m.log)

Also for some reason for me, the UI doesn't work, I see always the "Loading, please wait." message.

TheMClaeys, 2020-10-05:
I receive the same error as Trauma a few posts above since a few versions. I have to unplug the stick, plug it back in and then it starts fine. I thought it was an issue with my stick, so I bought a new one, flashed it and it has the same issue.
The strange thing is that Home Assistant itself still sees the stick in hardware, so the stick is still connected even though Zigbee2MQTT is apparently unable to "ping it".

golddragon007, 2020-10-05:
And around a day of run here is the error:

```
2020-10-04T20:18:22.664Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":17}
2020-10-04T20:18:22.664Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
(node:243) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
(node:243) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
2020-10-04T20:18:22.667Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
```

'Full' log file: 
[z2m-buffer-full.log](https://github.com/Koenkk/zigbee2mqtt/files/5326228/z2m-buffer-full.log)

The problem that the Zigbee network doesn't work, nor before nor after the error, and now I have around 1 month already down my Zigbee network half of my stuff is Zigbee so it's starting to be really bad.

Koenkk, 2020-10-05:
@golddragon007 is your stick connected through a USB extension cable?

golddragon007, 2020-10-05:
@Koenkk currently not. There was a long time, approx half-year without extension cable with cc2531 when it worked simply well, with few battery devices. Now I have Niko sockets. These are routers as well and also speaks A LOT. That the cc2531 seemed can't handle it too well without too big delays sometimes and also I was not able to connect more device to the network. So I've changed to a more powerful one. I can connect it via extension cable, but it won't work differently I'm 100% sure about that...

Koenkk, 2020-10-05:
Before trying anything else, I really recommend connecting it via an extension cable and positioning it 20 cms away from the Pi. Especially Pi 4 generates a lot of interference. I've seen cases where the Pi generated so much interference that it prevented the stick from completely functioning.

golddragon007, 2020-10-05:
It doesn't need too much time to break down, here is the dmesg, you can see when did I add the extension cable:
```Shell
[  177.554918] IPv6: ADDRCONF(NETDEV_CHANGE): veth6c1e7c1: link becomes ready
[  179.282562] hassio: port 14(veth6c1e7c1) entered disabled state
[  179.304298] eth0: renamed from veth560290b
[  179.364264] hassio: port 14(veth6c1e7c1) entered blocking state
[  179.364273] hassio: port 14(veth6c1e7c1) entered forwarding state
[84951.816675] hassio: port 14(veth6c1e7c1) entered disabled state
[84951.820197] veth560290b: renamed from eth0
[84951.974592] hassio: port 14(veth6c1e7c1) entered disabled state
[84951.996589] device veth6c1e7c1 left promiscuous mode
[84951.996603] hassio: port 14(veth6c1e7c1) entered disabled state
[85016.079577] usb 1-1.3: USB disconnect, device number 3
[85016.080013] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[85016.080157] cp210x 1-1.3:1.0: device disconnected
[85366.032886] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[85366.178216] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[85366.178225] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[85366.178231] usb 1-1.3: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[85366.178237] usb 1-1.3: Manufacturer: Silicon Labs
[85366.178243] usb 1-1.3: SerialNumber: 00_12_4B_00_21_CC_02_11
[85366.185899] cp210x 1-1.3:1.0: cp210x converter detected
[85366.192136] usb 1-1.3: cp210x converter now attached to ttyUSB0
[85391.305299] hassio: port 14(veth55ce20d) entered blocking state
[85391.305308] hassio: port 14(veth55ce20d) entered disabled state
[85391.307149] device veth55ce20d entered promiscuous mode
[85391.343590] IPv6: ADDRCONF(NETDEV_CHANGE): vethd6afa32: link becomes ready
[85391.343788] IPv6: ADDRCONF(NETDEV_CHANGE): veth55ce20d: link becomes ready
[85391.343906] hassio: port 14(veth55ce20d) entered blocking state
[85391.343913] hassio: port 14(veth55ce20d) entered forwarding state
[85392.684417] hassio: port 14(veth55ce20d) entered disabled state
[85392.684975] eth0: renamed from vethd6afa32
[85392.734328] hassio: port 14(veth55ce20d) entered blocking state
[85392.734339] hassio: port 14(veth55ce20d) entered forwarding state
[85634.226191] hassio: port 14(veth55ce20d) entered disabled state
[85634.226497] vethd6afa32: renamed from eth0
[85634.369836] hassio: port 14(veth55ce20d) entered disabled state
[85634.405294] device veth55ce20d left promiscuous mode
[85634.405312] hassio: port 14(veth55ce20d) entered disabled state
[85635.004183] hassio: port 14(veth1a6e467) entered blocking state
[85635.004191] hassio: port 14(veth1a6e467) entered disabled state
[85635.004417] device veth1a6e467 entered promiscuous mode
[85635.004711] hassio: port 14(veth1a6e467) entered blocking state
[85635.004718] hassio: port 14(veth1a6e467) entered forwarding state
[85635.032759] IPv6: ADDRCONF(NETDEV_CHANGE): veth1a6e467: link becomes ready
[85636.376480] hassio: port 14(veth1a6e467) entered disabled state
[85636.380442] eth0: renamed from veth017a48e
[85636.428259] hassio: port 14(veth1a6e467) entered blocking state
[85636.428269] hassio: port 14(veth1a6e467) entered forwarding state
pi@raspberrypi:~ $ cat /proc/uptime
88264.30 289139.89
```
And the new log:
[z2m-buffer-full2.log](https://github.com/Koenkk/zigbee2mqtt/files/5328084/z2m-buffer-full2.log)

I really hope that all reconnection will solve this. If you have any other idea that you can find out from the logs.

Koenkk, 2020-10-05:
Can you try turning of the following device: `0x086bd7fffeee0278 (0x086bd7fffeee0278): 170-33505 - Niko Connected socket outlet (Router)` especially this once is causing a lot of trouble.

golddragon007, 2020-10-05:
Ok, I've found it and turned off that circuit with two other Niko router. We will see how it will go now.

golddragon007, 2020-10-06:
[z2m-buffer-full3.log](https://github.com/Koenkk/zigbee2mqtt/files/5331811/z2m-buffer-full3.log)
Well, it's still happening and nothing wants to connect to it...

Koenkk, 2020-10-06:
Can you try changing: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/fromZigbee.js#L803 to 

```js
electrical_measurement_power: {
        /**
         * When using this converter also add the following to the configure method of the device:
         * await readEletricalMeasurementConverterAttributes(endpoint);
         */
        cluster: 'haElectricalMeasurement',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
             if (msg.type === 'attributeReport') {
                msg.meta.frameControl.disableDefaultResponse = true;
            }
            const getFactor = (key) => {
                const multiplier = msg.endpoint.getClusterAttributeValue('haElectricalMeasurement', `${key}Multiplier`);
                const divisor = msg.endpoint.getClusterAttributeValue('haElectricalMeasurement', `${key}Divisor`);
                const factor = multiplier && divisor ? multiplier / divisor : 1;
                return factor;
            };

            const lookup = [
                {key: 'activePower', name: 'power', factor: 'acPower'},
                {key: 'activePowerPhB', name: 'power_phase_b', factor: 'acPower'},
                {key: 'activePowerPhC', name: 'power_phase_c', factor: 'acPower'},
                {key: 'rmsCurrent', name: 'current', factor: 'acCurrent'},
                {key: 'rmsCurrentPhB', name: 'current_phase_b', factor: 'acCurrent'},
                {key: 'rmsCurrentPhC', name: 'current_phase_c', factor: 'acCurrent'},
                {key: 'rmsVoltage', name: 'voltage', factor: 'acVoltage'},
                {key: 'rmsVoltagePhB', name: 'voltage_phase_b', factor: 'acVoltage'},
                {key: 'rmsVoltagePhC', name: 'voltage_phase_c', factor: 'acVoltage'},
            ];

            const payload = {};
            for (const entry of lookup) {
                if (msg.data.hasOwnProperty(entry.key)) {
                    const factor = getFactor(entry.factor);
                    const property = postfixWithEndpointName(entry.name, msg, model);
                    payload[property] = precisionRound(msg.data[entry.key] * factor, 2);
                }
            }
            return payload;
        },
    },
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify fromZigbee.js

golddragon007, 2020-10-07:
I've added, so far it seems it works, at least there are messages and it doesn't stuck/slow down like previously. However, still, the devices don't want to connect to it (the already paired one from the previous system).
Is it possible to add a last seen column to the UI device list? Which displays the last received or ack message time?

golddragon007, 2020-10-08:
I've factory reset all my devices, uninstalled z2m deleted the /share/zigbee2mqtt folder and reflashed the ZigBee coordinator.
HA restart.
Installed back z2m, and surprise! Can't start the controller. Same issue like in the first comment. I've tried to change pan_id (6754 -> 6755) and also the channel (11 -> 12), doesn't matter, it doesn't start the same error.
[panid-in-use.log](https://github.com/Koenkk/zigbee2mqtt/files/5350361/panid-in-use.log)

Koenkk, 2020-10-08:
- For last seen in frontend, enable the `last_seen` option: https://www.zigbee2mqtt.io/information/configuration.html#configuration
- Have you also tried e.g. channel 15? If that doesn't work I'm afraid we need to sniff the traffic to see what is going on, sniff the traffic on which you try to start Zigbee2MQTT. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

golddragon007, 2020-10-08:
@Koenkk It doesn't want to start with ch 15 either. It seems for me that there's no need to debug the Zigbee at all. The problem is in the software on the stick or z2m itself. Or I've got two faulty stick from @slaesh or I missing something that's not described anywhere.
I've just reflashed my cc2531, clean z2m and it started without any issue. I was even able to pair a device.

Koenkk, 2020-10-08:
@golddragon007 if I understand correctly both CC2652RB don't work? I still think sniffing makes sense to do, the CC2531 will not do the check which makes your CC2652RB fail on startup.

golddragon007, 2020-10-08:
@Koenkk Yes, I have two CC2652RB and both failing with the same error (see previous log files). Then I can try to sniff with the CC2531 when the CC2652RB boots up (at least tries).

slaesh, 2020-10-08:
> I also flashed the debug firmware that shows the LEDs on the board. The red one went slightly dim on the boot even if it failed with the error. Then somehow configure itself, next reboot it went the red off, the blue was continuously on in both cases. But I was not able to pair any device...

Sounds good.

Blue on = CPU is running
the Orange one = just toggles on every received byte. On/Off doesn't mean anything. just to show if there is communication at all. ;)

just troughing this two cents right now. Need to read everything else tomorrow.

golddragon007, 2020-10-08:
@Koenkk Here is an export for the Wireshark. The controller failed appr. 3 times under this time.
[ws.txt](https://github.com/Koenkk/zigbee2mqtt/files/5351042/ws.txt)
Now I think maybe the battery-powered Zigbee devices block the start of this?

Koenkk, 2020-10-08:
Can you share the `pcapng` file and your nework key?

golddragon007, 2020-10-08:
Amm where can I find the `pcapng`? Amm it seems the program didn't create it, because there was no path... meh. I will redo it today.

golddragon007, 2020-10-09:
Well, I tried now, but it seems that it doesn't really catch too many messages... To be more precise, nothing from the coordinator. It's just failing...
I can't see anything now, yesterday I've just reset all the battery-powered devices, now I can't see any message, except if I try to pair a device, you can see it here those requests:
[try-to-connect.zip](https://github.com/Koenkk/zigbee2mqtt/files/5356944/try-to-connect.zip)
The keys are defaults as the docs says.

Koenkk, 2020-10-09:
Are you sure that non of the "Beacon requests" is from the slaesh stick? 

When I start my slaesh stick (with the following fw: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC2652RB_20200925.zip), a "Beacon request" is send by the stick (and this is expected behaviour).

[image]

Btw; also make sure there is no `data/coordinator_backup.json` file.

golddragon007, 2020-10-09:
Sorry if it wasn't clear. Yesterday I've cleared all the used battery devices.
Now when I listened I've didn't catch any message from the slaesh stick.
To check somehow if the sniffing working, I've tried two devices pair, that's what you see in the file.
I've always stopped z2m and deleted the full shared directory of the z2m. And I need to note that some cases it went into coordinator mode (booted up), but still, it wasn't able to connect any device to it.

Koenkk, 2020-10-09:
@golddragon007 you can try out Home Assistant ZHA, except the CC2652RB firmware, the whole stack is different.

golddragon007, 2020-10-10:
@Koenkk I've tried now to add it to ZHA and I can't even add it. This is the error message: "Unable to connect to ZHA device."
Some logs:
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.api] Removing listener OneShotResponseListener(matching_commands=(SYS.Ping.Rsp(Capabilities=None), RPCError.CommandNotRecognized.Rsp(ErrorCode=None, RequestHeader=CommandHeader(id=0x01, subsystem=Subsystem.SYS, type=CommandType.SREQ))), future=<Future cancelled>)
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.api] Cleaning up empty listener list for header CommandHeader(id=0x00, subsystem=Subsystem.RPCError, type=CommandType.SRSP)
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.api] Cleaning up empty listener list for header CommandHeader(id=0x01, subsystem=Subsystem.SYS, type=CommandType.SRSP)
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.api] There are 0 callbacks and 0 one-shot listeners remaining
2020-10-10 14:13:24 WARNING (MainThread) [zigpy_znp.zigbee.application] Failed to probe ZNP radio with config {'path': '/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0', 'baudrate': 115200, 'flow_control': None}
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 467, in request
    response = await response_future
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/zigbee/application.py", line 116, in probe
    await znp.connect()
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 210, in connect
    ping_rsp = await self.request(c.SYS.Ping.Req())
  File "/usr/local/lib/python3.8/site-packages/zigpy_znp/api.py", line 467, in request
    response = await response_future
  File "/usr/local/lib/python3.8/site-packages/async_timeout/__init__.py", line 55, in __aexit__
    self._do_exit(exc_type)
  File "/usr/local/lib/python3.8/site-packages/async_timeout/__init__.py", line 92, in _do_exit
    raise asyncio.TimeoutError
asyncio.exceptions.TimeoutError
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.api] We were disconnected from /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0: None
2020-10-10 14:13:24 DEBUG (MainThread) [zigpy_znp.uart] Closing /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 serial port

2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.zigbee.application] Probing /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 at 115200 baud
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 serial port
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 at 115200 baud
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.api] Sending special byte to skip the bootloader
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xEF'
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.api] Sending request: SYS.Ping.Req()
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.api] Creating one-shot listener OneShotResponseListener(matching_commands=(SYS.Ping.Rsp(Capabilities=None), RPCError.CommandNotRecognized.Rsp(ErrorCode=None, RequestHeader=CommandHeader(id=0x01, subsystem=Subsystem.SYS, type=CommandType.SREQ))), future=<Future pending cb=[ZNP.wait_for_responses.<locals>.<lambda>() at /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:400, <TaskWakeupMethWrapper object at 0xa28c4538>()]>)
2020-10-10 14:13:19 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xFE\x00\x21\x01\x20'

2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.zigbee.application] Probing /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.uart] Connecting to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 at 115200 baud
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.uart] Opened /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 serial port
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.uart] Connected to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0 at 115200 baud
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.api] Sending special byte to skip the bootloader
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xEF'
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.api] Testing connection to /dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_09_E4-if00-port0
[... cut ...]
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.api] Creating one-shot listener OneShotResponseListener(matching_commands=(SYS.Ping.Rsp(Capabilities=None), RPCError.CommandNotRecognized.Rsp(ErrorCode=None, RequestHeader=CommandHeader(id=0x01, subsystem=Subsystem.SYS, type=CommandType.SREQ))), future=<Future pending cb=[ZNP.wait_for_responses.<locals>.<lambda>() at /usr/local/lib/python3.8/site-packages/zigpy_znp/api.py:400, <TaskWakeupMethWrapper object at 0xa44892c8>()]>)
2020-10-10 14:12:26 DEBUG (MainThread) [zigpy_znp.uart] Sending data: b'\xFE\x00\x21\x01\x20'

If you see here a different mac, that's because I've tried also with my second stick...

Koenkk, 2020-10-10:
@slaesh should the stick work with Home Assistant ZHA?

puddly, 2020-10-10:
It should as of 0.116.0, which was released three days ago.

The above log is for an older release that did not toggle the DTR and RTS pins. I don't own the hardware but replicating the pin toggling you used in zigbee-herdsman made it work for another user (https://github.com/zha-ng/zigpy-znp/issues/35, https://github.com/zha-ng/zigpy-znp/pull/36).

golddragon007, 2020-10-10:
Yes, so now I've updated the HA and I was able to add the coordinator, now we look into why I can't pair a device to it.

golddragon007, 2020-10-10:
Same issue with zha like with z2m. It detects the coordinator, but can't start the network.

puddly, 2020-10-10:
Solved by using a powered hub on the Raspberry Pi 4 that was also booting off of an m.2 SSD.

Interestingly, @golddragon007 said that a CC2531 was working just fine plugged into the same Pi so could it possibly be a power issue? The stick seems to become unresponsive right when the network is being formed, which is when Z-Stack should be performing a scan to find existing networks on the provided channel and may be momentarily drawing a lot of current.

golddragon007, 2020-10-10:
@Koenkk it seems you were 100% right. For some reason for this stick, it's not enough the power for the CC2652RB that the Pi can provide, but it's enough for the cc2531. My USB meter is on the way, once it will arrive I will do further analysis on this. With a USB hub that powered separately, it works.

```Shell
# python 3.7+ requried
# With virtual env
$ pip install virtualenv
$ virtualenv -p python3
$ git clone https://github.com/zha-ng/zigpy-znp/
$ virtualenv -p python3 venv
$ source venv/bin/activate
(venv) $ pip install -e ./zigpy-znp/
(venv) $ python -m zigpy_znp.tools.nvram_reset -vvv -c /dev/serial/by-id/...
(venv) $ python -m zigpy_znp.tools.energy_scan -vvv /dev/serial/by-id/...

# Without virtual env
git clone https://github.com/zha-ng/zigpy-znp/
pip install -e ./zigpy-znp/
# For reset network.
python -m zigpy_znp.tools.nvram_reset -vvv -c /dev/serial/by-id/...
# For scan networks.
python -m zigpy_znp.tools.energy_scan -vvv /dev/serial/by-id/...
```

Koenkk, 2020-10-11:
@golddragon007 great to hear you found the root cause. I've added this info to the FAQ.

golddragon007, 2020-10-16:
I've done the measures. In the end, it seems the cc2531 eats more than the cc2652rb, but also more stable for the voltage drop.
You can find the full here: [Usb-measures.zip](https://github.com/Koenkk/zigbee2mqtt/files/5394266/Usb-measures.zip)
You need to download from here (https://www.avhzy.com/html/product-detail/ct3) the PC software (only windows) if you want to check directly the measures, otherwise, there are BMP pictures inside the folder.

closed 2020-11-16 by golddragon007