Koenkk/zigbee2mqtt #4514: Sonoff SNZB-02 Device ID 66666

alexinthis, 2020-10-01:
I've just added my first device and it was not detected properly. 

https://pastebin.com/t10vWRZS 

I have since paired a Hue Switch and that works fine.
The Sonoff SNZB-02 Temperature and Humidity sensor has a modelID of 66666 which I assume is incorrect. Do I just have a faulty device or is this something I can fix? 

database.db output: 
https://pastebin.com/A6YTZGHQ

Zigbee2MQTT version: 1.14.4 
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200805

Koenkk, 2020-10-02:
@alexinthis could you share the `data/database.db` entry of this device?

julioyg, 2020-10-02:
same here, this is the entry 
```
{
  "id": 3,
  "type": "EndDevice",
  "ieeeAddr": "0x00124b00222e9824",
  "nwkAddr": 38658,
  "manufId": 0,
  "manufName": "eWeLink",
  "powerSource": "Battery",
  "modelId": "66666",
  "epList": [
    1
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 770,
      "inClusterList": [
        0,
        3,
        1026,
        1029,
        1
      ],
      "outClusterList": [
        3
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "66666",
            "manufacturerName": "eWeLink",
            "powerSource": 3,
            "zclVersion": 1,
            "appVersion": 4,
            "hwVersion": 1,
            "dateCode": "20200718"
          }
        }
      },
      "binds": [],
      "meta": {}
    }
  },
  "appVersion": 4,
  "hwVersion": 1,
  "dateCode": "20200718",
  "zclVersion": 1,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1601674267791
}
```

alexinthis, 2020-10-03:
> @alexinthis could you share the `data/database.db` entry of this device?

Hey, it's linked in my first post here: https://pastebin.com/A6YTZGHQ
Did you need more info? 

Thanks.

Koenkk, 2020-10-03:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

julioyg, 2020-10-04:
yay! woking like a charm now.

Thanks a lot :)

closed 2020-10-05 by Koenkk