Koenkk/zigbee2mqtt #4515: Support for new hue ambiance candle e14

JonOve, 2020-10-01:
Hello,

I am running Zigbee2MQTT add on version 1.14.4.2 for Home Assistant on a Synology NAS.
Spend some hours to find out how to add support for "not supported devices" but up to now without succes unfortunately.
I am not able to find or create a file like devices.js or fromZigbee.js available to adjust.

Now I just received the latest Hue ambiance candle e14 and added it to Zigbee2MQTT, however the log says it is not supported.

info  2020-09-30 13:16:58: Device '0x00178801090d4a88' joined
info  2020-09-30 13:16:58: Starting interview of '0x00178801090d4a88'
info  2020-09-30 13:16:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00178801090d4a88"},"type":"device_connected"}'
info  2020-09-30 13:16:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00178801090d4a88"},"type":"pairing"}'
info  2020-09-30 13:17:01: Successfully interviewed '0x00178801090d4a88', device has successfully been paired
warn  2020-09-30 13:17:01: Device '0x00178801090d4a88' with Zigbee model 'LTE002' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
info  2020-09-30 13:17:01: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x00178801090d4a88","supported":false},"type":"pairing"}'

Can you please support me in this matter?

Koenkk, 2020-10-02:
What features does this bulb support, brighthness, color_temp and color (green, red, e.g.)?

JonOve, 2020-10-02:
Hello Koen,

Brightness, color_temp, on/off. No RGB. In fact equal to the 8718696695203 but then including Bluetooth.

Thanks a lot for your support!!

Koenkk, 2020-10-03:
Added!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-03 by Koenkk