Koenkk/zigbee2mqtt #4519: installation issues

Tweakerd, 2020-10-01:
# Bug Report
## What happened
Can't get zigbee2mqtt working on whatever device I'll try.

## What did you expect to happen
A working system 

## How to reproduce it (minimal and precise)
Hi, 
I'm running Domoticz for queite a time now, and everything is working, but as the amount of devices keeps growing, the wifi isn't getting easy to maintain anymore so I bought a Sniffer, debugger and some time.

I followd the instructions site but ran in exactly the same errors over and over again with the result .

I tried to start zigbee2mqtt on:
Synology DS916+ using Docker => result Docker keeps shutting down (loop on autostart on off)
Thin client running debian Buster (10)  (AMD x64 CPU)
Raspberry pi 2B+ running Raspian Buster (10) and Stretch (9)
A windows 10 PC Intel CPU x64 (Pentium)

the strange thing is that on the raspi, (ARMv6) and get repistory for Node (V12.9.0 or whathever) that I'll download after install, and check the version, it is NOT V12.x but V8.x
NPM is not installed then.
I'll then apt- install npm end result is v 5.x.

Also even om my W 10 system, no working system.

I'm getting desperit :-(

In docker I also tried older versions  of zigbee@mqtt with same result (Loop rebooting Docker)

Who can help me or what is going wrong?

## Debug Info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version:
CC DEBUGGER\CC2531_DEFAULT_20190608\CC2531ZNP-Prod.hex

JonOve, 2020-10-03:
I am running Zigbee2MQTT on a Synology Nas and it works great. Maybe you have already checked, but Bearded Tinker is a great support for Synology Nas. In this video he installs HA and Z2mqtt. https://youtu.be/QdBYUbj0B5Q

Tweakerd, 2020-10-03:
Txs, gonna check it out.Let you know if it worked.Verzonden vanaf Samsung-tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Koenkk/zigbee2mqtt/issues/4519#issuecomment-703056493",
"url": "https://github.com/Koenkk/zigbee2mqtt/issues/4519#issuecomment-703056493",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Tweakerd, 2020-10-11:
Thanks all, 

I watched the video's but cant come anny further. 
First of all i'm not running HA, cause I like Domotics more and it is running fine with me. (Aso because is have alread around 30 devices working fine on it) Lot of work to transfer to HA.
Th only thing I want to do is to add Zigbee to it.
As mentioned I didn't find any tutorial to let my flashed Texas Instruments CC2531stick get to work.
Not on a Rpi2B+, Synology nas running Docker (DS916+, not installable on a DS214+ due to hardware limitations) and on a Thin Client running debian-10.4.0-amd64-netinst. Yes for sure I folowed the tutorials.

My Domoticz is running on a Rpi3B+ Running on a SSD. Atached a directly is my Energy meter an through Modbus connection a power meter for my Solar pannels.
It's completely stand alone and dedicated.

My MQTT broker is running on my Synology DS916+
I would like to add Zigbee 2 it to unload trafic on my WLan-Routers and extend my automation and make it a little more relyable.

On all systems it looks like there are some files missing in the script..
Perhaps I shall try again and make some pictures/screenshots

Tweakerd, 2020-10-12:
hereby some logs I'm getting.
Again, it's a clean Debian 10 install, nothing else installed, only thing is that I had to manualy install "curl" for some reason.

Perhaps some more info.. there was a update my domoticz running Rpi3+ on Debian Stretch. After update of domoticz everything crashed. I had to reinstall to Debian 10 en put config domoticz back.
So there is al lot changed  between Debian Stretch and Buster.
Can it be that there are some things going wrong in the scripts Zigbee2mqtt?

[output 2.txt](https://github.com/Koenkk/zigbee2mqtt/files/5365390/output.2.txt)
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/5365391/debug.txt)
[output.txt](https://github.com/Koenkk/zigbee2mqtt/files/5365392/output.txt)

Koenkk, 2020-10-12:
Try `cd /opt/zigbee2mqtt && npm start`.

Tweakerd, 2020-10-12:
Koen Tnx for the sugestion.
Sadly the result is the same.

I think I red somewhere that there is a difference in the way Debian is using G++ and GCC (something with different way of working.. 
Perhaps a lead?

Koenkk, 2020-10-13:
Can you provide logs?

Tweakerd, 2020-10-13:
Yes, latef on the day I wil send the logs..Regards,Verzonden vanaf Samsung-tablet.

Tweakerd, 2020-10-13:
Hereby the latest install folowing the manual.
Now tried on Ubuntu, 
No succes still..

also the install log and logfile,

regards, 

[INSTALL ZIGBEE UBUNTU.txt](https://github.com/Koenkk/zigbee2mqtt/files/5370736/INSTALL.ZIGBEE.UBUNTU.txt)
[LOGFILE UBUNTU.txt](https://github.com/Koenkk/zigbee2mqtt/files/5370737/LOGFILE.UBUNTU.txt)

Koenkk, 2020-10-13:
You didn't execute `npm ci`, last command of https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#2-installing

Tweakerd, 2020-10-13:
You're right.. 
In previous installations , I did run the command.
When my drive is coppied, I'll try again.
You'll see the logs comming.. 👍

Tweakerd, 2020-10-13:
[Started and same error.txt](https://github.com/Koenkk/zigbee2mqtt/files/5371203/Started.and.same.error.txt)
[Yaml changed.txt](https://github.com/Koenkk/zigbee2mqtt/files/5371204/Yaml.changed.txt)

to bad.. this is what I get over and over on whatever system..

Koenkk, 2020-10-13:
Change `server: '192.168.2.16:1883'` to `server: 'mqtt://192.168.2.16:1883'` in your `configuration.yaml`. (https://www.zigbee2mqtt.io/information/configuration.html#configuration)

Tweakerd, 2020-10-13:
Hi, 

See latest logs, tried a few things, still no succes altough some things changed..
[latest YAML file.txt](https://github.com/Koenkk/zigbee2mqtt/files/5373691/latest.YAML.file.txt)
[latest log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5373692/latest.log.txt)
[latest npm ci and start.txt](https://github.com/Koenkk/zigbee2mqtt/files/5373693/latest.npm.ci.and.start.txt)

Koenkk, 2020-10-14:
`Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/data/log/2020-10-13.21-24-56'` the user you are running Zigbee2MQTT as doensn't have permission to the `/opt/zigbee2mqtt` directory.

closed 2020-11-21 by github-actions[bot]