Koenkk/zigbee2mqtt #4521: Add Prometheus Support

jcarvalho, 2020-10-01:
The introduction of the Frontend in 1.15.0 brought web support to the project, which would be a great opportunity to expose a [Prometheus Metrics](https://github.com/prometheus/docs/blob/master/content/docs/instrumenting/exposition_formats.md) endpoint, containing information on the current state of the various registered devices (state, LQI, battery, etc).

If there's interest, I would be glad to work on this 🙂 .

## Describe alternatives you've considered
An alternative to this endpoint would be to integrate with another software (such as Home Assistant), and scrape the Prometheus metrics from there (this is the approach I'm currently taking).

Koenkk, 2020-10-02:
If there is interest (users that want this please put a thumbs up on the OP) I'm fine with adding this. Implementation wise, such functionality should fit well in an additional extension (https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/extension.js).

sjorge, 2020-10-03:
Maybe a more generic metrics backend that can submit to prometheus, graphite, and influxdb? Assuming all three are rather close aside the output format.

nekeress, 2020-10-04:
There is an already existing solution for MQTT which supposedly works with mosquitto, did anyone try it? "zigbe2mqtt" is actually mentioned on the page.
It would be nice to have a Grafana dashboard as well though.
https://github.com/kpetremann/mqtt-exporter
P.S.: I will try to install and test it, and post an update.

nekeress, 2020-10-09:
I couldn't make it work as my MQTT server has anonymous access disabled and I don't want to change that. Played around with it for a while but I only saw connection errors in the MQTT log, and I'm too busy right now to test it more. If anyone had success with it. let me know.

jcarvalho, 2020-11-16:
Should we reopen this to gather more feedback, or leave it as is?

AverageMarcus, 2022-01-12:
I was actually hoping for metrics around the state of Zigbee2MQTT itself and the network as a whole, rather than the specific device values.

For example:
* metrics around the number of failed vs successful commands sent to devices
* metrics around the average, min and max last seen time
* simple counter for number of devices in the network
* simple `up` metric for alerting purposes
* a status metric with the current version of app and controller (for use with other metrics to compare metrics after upgrade)

mweinelt, 2022-03-01:
I'd very much like to watch lqi over time to see if the channel is degrading, would've been helpful with #11271.

onedr0p, 2022-08-24:
I would be interested in this for LQI and also getting notified thru Alert Manager when a device goes offline. There seems to be a pretty popular prom library for node that might make integrating this easy: https://github.com/siimon/prom-client

@Koenkk any chance this issue can be opened back up? Thanks!

runningman84, 2022-08-25:
I think you can do all of these things using the mqtt exporter. Here is my flux config:

```
---
apiVersion: helm.toolkit.fluxcd.io/v2beta1
kind: HelmRelease
metadata:
  name: zigbee2mqtt-exporter
  namespace: monitoring
spec:
  interval: 60m
  chart:
    spec:
      # renovate: registryUrl=https://k8s-at-home.com/charts/
      chart: mqtt-exporter
      version: 1.1.2
      sourceRef:
        kind: HelmRepository
        name: k8s-at-home-charts
        namespace: flux-system
      interval: 60m
  values:
    image:
      repository: kpetrem/mqtt-exporter
      tag: latest
      pullPolicy: IfNotPresent
    env:
      # -- Set the container timezone
      TZ: UTC
      # -- Logging level (default: INFO)
      LOG_LEVEL: INFO
      # -- Comma-separated lists of topics to ignore. Accepts wildcards. (default: None)
      MQTT_IGNORED_TOPICS:
      # -- IP or hostname of MQTT broker (default: 127.0.0.1)
      MQTT_ADDRESS: mosquitto.hass.svc.cluster.local
      # -- TCP port of MQTT broker (default: 1883)
      MQTT_PORT: 1883
      # -- Topic path to subscribe to (default: #)
      MQTT_TOPIC: "zigbee2mqtt/#"
      # -- Keep alive interval to maintain connection with MQTT broker (default: 60)
      MQTT_KEEPALIVE: 60
      # -- Username which should be used to authenticate against the MQTT broker (default: None)
      MQTT_USERNAME:
      # -- Password which should be used to authenticate against the MQTT broker (default: None)
      MQTT_PASSWORD:
      # -- HTTP server PORT to expose Prometheus metrics (default: 9000)
      PROMETHEUS_PORT: 9000
      # -- Prefix added to the metric name, example: mqtt_temperature (default: mqtt_)
      PROMETHEUS_PREFIX: mqtt_
      # --  Define the Prometheus label for the topic, example temperature{topic="device1"} (default: topic)
      TOPIC_LABEL: topic
      # --  Normalize sensor name for device availability metric added by Zigbee2MQTT (default: False)
      ZIGBEE2MQTT_AVAILABILITY: "True"
    metrics:
      # -- Enable and configure a Prometheus serviceMonitor for the chart under this key.
      # @default -- See values.yaml
      enabled: true
    resources:
      limits:
        memory: 64Mi
      requests:
        cpu: 50m
        memory: 64Mi
```

and these are my rules:
```
apiVersion: monitoring.coreos.com/v1
kind: PrometheusRule
metadata:
  labels:
    app: kube-prometheus-stack
    release: prometheus-operator
  name: zigbee2mqtt.rules
  namespace: monitoring
spec:
  groups:
  - name: zigbee2mqtt.rules
    rules:
    - alert: MqttExporterAbsent
      expr: absent(up{job=~".*mqtt-exporter.*"} == 1)
      for: 15m
      labels:
        severity: critical
      annotations:
        summary: The mqtt exporter job is absent
        description: The mqtt exporter job is absent
    - alert: Zigbee2MqttUnavailable
      expr: mqtt_zigbee_availability == 0
      for: 30m
      labels:
        severity: critical
      annotations:
        summary: The zigbee device connection is lost
        description: connection on topic {{$labels.topic}} is down
    - alert: Zigbee2MqttLinkqualityLow
      expr: mqtt_linkquality < 15
      for: 60m
      labels:
        severity: warning
      annotations:
        summary: The zigbee device link quality is low
        description: link quality on topic {{$labels.topic}} is at {{$value}}
    - alert: Zigbee2MqttBatteryLow
      expr: mqtt_battery < 40
      for: 60m
      labels:
        severity: warning
      annotations:
        summary: The zigbee device battery level is low
        description: battery level on topic {{$labels.topic}} is at {{$value}}%
    - alert: Zigbee2MqttTemperatureHigh
      expr: mqtt_temperature > 50
      labels:
        severity: warning
      annotations:
        summary: Temperature high (>50c)
        description: temperature on topic {{$labels.topic}} is {{$value}}c
    - alert: Zigbee2MqttTemperatureLow
      expr: mqtt_temperature < 10
      labels:
        severity: warning
      annotations:
        summary: Temperature low (<10c)
        description: temperature on topic {{$labels.topic}} is {{$value}}c
    - alert: Zigbee2MqttSmokeDetectorAlert
      expr: mqtt_smoke > 0
      labels:
        severity: critical
      annotations:
        summary: A smoke detector is in state alarm
        description: smoke detector on topic {{$labels.topic}} is reporting smoke
    - alert: Zigbee2MqttWaterLeakAlert
      expr: mqtt_water_leak > 0
      labels:
        severity: critical
      annotations:
        summary: A water leak detector is in state alarm
        description: water leal detector on topic {{$labels.topic}} is reporting water
```

onedr0p, 2022-08-25:
I discovered this too @runningman84 but I would be very happy to have this built into the application instead of relying on a separate application to handle it. It's a good workaround for sure but I'm running into issues using it. https://github.com/kpetremann/mqtt-exporter/issues/32

sjorge, 2022-08-25:
Not prometheus but I have a half finished extension to send the data to influxdb, but It's sort of stuck behind https://github.com/Koenkk/zigbee2mqtt/issues/13025 for the moment as I also want to use that to push additional config to sets of devices based on model. (e.g. store humity + temperature for sensors of type x, store position for devices of type y ...)

For now I also fell back to custom written python app that checks the device model and maps that to 'per model' settings I store in a config file for the python app.

AngryMandragora, 2022-09-06:
As a workaround to those who use Home Assistant with Zigbee2mqtt like me and want to have their data in Prometheus, here is a guide that I used to set up the builtin HA Prometheus exporter:
https://josh-v.com/home_assistant/homeassistant-prometheus/

pastukhov, 2022-10-27:
There is 2 exporters for zwave2mqtt

- https://github.com/varet80/zj2m-prom-exporter
- https://github.com/kvaster/zwavejs-prom

These can be use as examples

onedr0p, 2022-11-27:
Still interested in this.

elad-bar, 2023-02-19:
Also interested in that

tschieggm, 2023-03-02:
I am also interested in this.

junosuarez, 2023-03-03:
Started working on this in #16905 - feedback welcome from users on what kinds of metrics to expose.

onedr0p, 2023-04-03:
Not stale, @junosuarez any update?

junosuarez, 2023-04-05:
Still waiting for feedback on the PR. In the mean time, I've also raised this with https://github.com/smlight-dev/slzb-06-firmware/issues/21 for metrics on the SLZB-06 coordinator, if anyone is interested.

onedr0p, 2023-04-05:
Cool, I own the same coordinator! I just read over that issue and on that last comment the metrics don't appear to be in the right format...

onedr0p, 2023-05-06:
Not stale.

onedr0p, 2023-06-06:
Not stale.

moritz31, 2023-06-26:
any update on this ?

sjorge, 2023-06-26:
Basically https://github.com/Koenkk/zigbee2mqtt/pull/16905#issuecomment-1556190230, the PR can be opened against the extension repo where it can skip all the testing requirements and it can be iterated over quicker.

onedr0p, 2023-07-27:
@Koenkk can you transfer this issue to https://github.com/Koenkk/zigbee2mqtt-user-extensions ?

onedr0p, 2023-08-27:
Issue is not transferred.

onedr0p, 2023-09-27:
Issue still not transferred.