Koenkk/zigbee2mqtt #4523: Cannot find devices.js file

sloopee, 2020-10-02:
Hi all, I'm brand new to all of this and have literally no programming background. I'm trying to add support for the aqara D1 3 gang with neutral switch. The switch is discovered by mqtt discovery (I think?) but doesn't support power measurement, it currently shows link quality and the on/off options for all 3 gangs. I've seen someone else manage to do it in the following thread (https://github.com/Koenkk/zigbee-herdsman-converters/issues/1274) by manually adding the "devices.js" file however I just can't seem to find devices.js file. I've tried hard to follow the following link but I'm stuck where it says edit the devices.js file. https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I'm currently running home assistant off a synology NAS and have zigbee2mqtt 1.15.0 installed with zigbee2mqttassistant 0.3.157 as well. I'm using the HA mqtt broker add on 5.1.  

The closest I seem to have gotten is to HomeAssistant > hass.io > share > zigbee2mqtt folder and I see the file devices.yaml, state.json, etc but not devices.js.

I've searched all over youtube and google for help and have been stuck for 3 days so any help is much appreciated!

Koenkk, 2020-10-02:
Duplicate of https://github.com/Koenkk/zigbee-herdsman-converters/issues/1616

closed 2020-10-02 by Koenkk

Gelisob, 2020-10-23:
I'm not sure this is a duplicate @Koenkk , the issue here is not the aquara stuff, the issue here is, that user cant find devices.js file within their system, to perform new device support, like the how to guides them to.
I have the same issue, I found some "whatever who cares" motion sensors, that pair into unsupported status with guidance in the log that suggests me to follow "How to support new devices" page.
And I too run into the problem, that there's no devices.js to be found.
from other threads i see path's to /opt and /app being referred, but /opt is empty for me and /app doesnt exist.
Is there a dependency that we are missing by some error or how to page is out of date?

hass.io direct to ssd image install, no docker, version 0.116.2
mosquitto broker version 5.1
zigbee2mqtt version 1.15.0