Koenkk/zigbee2mqtt #4524: Spurious MQTT publish on set/position

jdtsmith, 2020-10-02:
## What happened

Latest main branch version of z2m sends a spurious MQTT message upon receiving a `/set/position` command for Fyrtur blinds:

```
Zigbee2MQTT:debug 2020-10-01 23:28:08: Received MQTT message on 'zigbee2mqtt/ikea_blind_famroom_30in_S/set/position' with data '100'
Zigbee2MQTT:debug 2020-10-01 23:28:08: Publishing 'set' 'position' to 'ikea_blind_famroom_30in_S'
Zigbee2MQTT:info  2020-10-01 23:28:08: MQTT publish: topic 'zigbee2mqtt/ikea_blind_famroom_30in_S', payload '{"battery":69,"linkquality":63,"position":100,"update_available":false}'
Zigbee2MQTT:debug 2020-10-01 23:28:10: Received Zigbee message from 'ikea_blind_famroom_30in_S', type 'attributeReport', cluster 'closuresWindowCovering', data '{"currentPositionLiftPercentage":99}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-01 23:28:10: MQTT publish: topic 'zigbee2mqtt/ikea_blind_famroom_30in_S', payload '{"battery":69,"linkquality":65,"position":1,"update_available":false}'
```

You can see how the `"position":100` is published, despite this being the _target_ position, not the actual current position. The actual positions updates start rolling in shortly thereafter (starting here with position=1).

## What did you expect to happen

No update to the position reporting topic using the recently _set_ position.  Perhaps another topic should be used?

## How to reproduce it (minimal and precise)

Command a blind to a position and observe the MQTT output at the nominal position reporting MQTT topic. 

## Debug info
Zigbee2MQTT version: zigbee2mqtt@1.15.0
Adapter hardware: CC2531
Adapter firmware version: zStack12

Koenkk, 2020-10-02:
This is something quite fundamentally (and not something that has been changed recently). When Zigbee2MQTT sends a Zigbee command, and that command is successfully received by the device, Zigbee2MQTT assumes the device changed to that state so it publishes this as the new state. This is right in almost all cases.

An alternative solution here would be to immediately poll the device for the actual state. But this has two down sides:
- Doubles the Zigbee traffic (which will especially hurt the not powerful CC2531 coordinator)
-  There will be a bigger delay between setting and getting the confirmation published (as it also needs to read).

jdtsmith, 2020-10-02:
Interesting, not sure why I didn't notice it before.  I wonder if this can be optionally disabled if one does not need "immediate confirmation"?  I'm sure there are other devices for which setting a given state takes time to achieve (garage doors, etc.).  I can probably work around it by discarding the first position update that matches the target state.

jdtsmith, 2020-10-02:
Alternatively, perhaps there is a way to indicate that this is an "unconfirmed" value in the MQTT message itself.

jdtsmith, 2020-10-04:
> Alternatively, perhaps there is a way to indicate that this is an "unconfirmed" value in the MQTT message itself.

Just an additional note: I spent some time trying to recognize and squash spurious positional updates from Z2M, without luck.  The issue is (at least for Home app as a front end controller), is when sliding the position of the blinds using this or similar interface:

[image]

In this case the controller invokes `/set/position` 3 or more in quick succession.  All of these newly set targets get erroneously reported by Z2M as the current position some time after they were set.  In the meantime, *real* position updates are coming in.  This drives the reported state of the blind crazy, at it flips between opening/closing/some percent open/etc.  

Here's an example, starting at a good confirmed position of 67:

```
currentPosition=67
Setting targetPosition->63
Setting targetPosition->57
Setting targetPosition->51
currentPosition=63 # SPURIOUS
Setting targetPosition->50
currentPosition=66
currentPosition=63
currentPosition=57 #SPURIOUS
currentPosition=62
currentPosition=59
currentPosition=51 #SPURIOUS
currentPosition=56
currentPosition=53
currentPosition=50 #SPURIOUS (but harmless)
currentPosition=50
currentPosition=50
currentPosition=50
```

I've looked at saving a "recent targets set" list, checking for changes in the direction of motion, and even measuring the apparent rate of progress (percent open per second) and flagging that if it's too high.  But because real and spurious positional updates are intermixed in a random order, it's very challenging to reliably detect them.  Since Z2M itself knows which events were "presumptions" of assumed state, I do believe adding a flag to the message reporting to that effect is the only reliable solution, something like:

```
{"battery":69,"linkquality":60,"position":63,"assumed":true,"update_available":false}
```

Thoughts?

Koenkk, 2020-10-05:
To make sure the issue is indeed caused by the "optimistic state", can you try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/e6912a1a89c8c1d2094d3525cf588ad390ff90df/converters/toZigbee.js#L292 to `return {};`?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify toZigbee.js

jdtsmith, 2020-10-05:
> `return {};`?

This does indeed entirely fix the issue, as the "optimistic state" message is no longer sent.  I have also now implemented rate control on the homebridge-mqttthing side to avoid sending rapid-fire target position requests, which really confuses things.  This helps, but doesn't solve the confused ordering of positions issue.  But either with or without rate limiting, omitting the "optimistic state" updates fixes this problem and normalizes feedback on the opening progress.  Thanks!

Koenkk, 2020-10-06:
In https://github.com/Koenkk/zigbee2mqtt/pull/4572/files I've added the `optimistic` option, if you take over the changes to `publish.sh` it should work when setting the following in your `configuration.yaml`.

```yaml
devices:
  0x12345678:
    friendly_name: my_curtain
    optimistic: false
```

jdtsmith, 2020-10-06:
Great, thanks.  I checked out and installed this PR and it works as advertised with the `optimistic: false` settings on the blinds.  I'll probably still continue to rate limit the target position setting just to keep traffic down.

Koenkk, 2020-10-06:
To limit the traffic change https://github.com/Koenkk/zigbee-herdsman-converters/blob/77efc3a4f47f22ebd3f6d0e1016c4ce34b8c770c/devices.js#L92 to `const payload = configureReportingPayload('currentPositionLiftPercentage', 3, repInterval.MAX, 1, overrides);` and do a configure after that https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Let me know if this limits the update, then we can make this the default.

closed 2020-10-06 by Koenkk