Koenkk/zigbee2mqtt #4526: Sercomm SZ-ESW01-AU fails to pair with CC2530 + CC2591, pairs ok with CC2531

garywatts, 2020-10-02:
## What happened
I tried to pair a Sercomm SZ-ESW01-AU Power Plug with my setup of CC2530 + CC2591 and it would not show up, so I took to friend's house and tested his similar setup but with a CC2531 and it paired ok. Tested again on my setup with a CC2531 and it pairs OK with that stick but not the CC2530 + CC2591. Once paired via the CC2531 and swapping back to the CC2530 + CC2591 I can control it OK, but of course it is not showing as connected on the map. (The CC2530 + CC2591 previously paired my 13 Xiaomi devices without issue) 

## What did you expect to happen

Sercomm SZ-ESW01-AU should pair. 

Note the official pairing process is unplug the device, hold pair button for 4sec, plug it back in while holding button and keep holding it for 2 sec. You know it in pairing mode when the red LED blinks 3 times quickly and then pause followed by 3 blinks again and so on. 

## How to reproduce it (minimal and precise)
Attempting to pair the device results in no pairing or any feedback in the default log. Debug log mentions "log 'zcl' data is from unknown device with address '1443', skipping.."

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5316338/log.txt)

## Debug info
Zigbee2MQTT version: tested with both 1.15.0 and edge
Adapter hardware: CC2530 + CC2591
Adapter firmware version: Z-Stack_Home_1.2 default

Koenkk, 2020-10-02:
Check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair , if that doesn't help try to reflash the CC2530.

garywatts, 2020-10-03:
Thanks, I ended up had to reflash the CC2530 and it connected ok after a few goes. The device is a finicky thing to pair and mine also has a fault that causes it turns off randomly. I would not recommend purchase of a Sercomm SZ-ESW01-AU.

closed 2020-10-03 by garywatts